MichaIng/DietPi #5482: Issue Updating

doobersn, 2022-05-09:
Hi, I am using a pi zero w and trying to update dietpi. I got this error:

```
Downloading update archive                                                                                           │ 
         │  - Command: curl -sSfLO https://github.com/MichaIng/DietPi/archive/master.tar.gz                                     │ 
         │  - Exit code: 127                                                                                                    │ 
         │  - DietPi version: v8.3.1 (MichaIng/master) | HW_MODEL: 1 | HW_ARCH: 1 | DISTRO: 5                                   │ 
         │  - Image creator: DietPi Core Team                                                                                   │ 
         │  - Pre-image: Raspbian Lite                                                                                          │ 
         │  - Error log:                                                                                                        │ 
         │ curl: error while loading shared libraries: libz.sO*1: cannot open shared object file: No such file or directory  
```

Any idea what I should do? Thanks!

MichaIng, 2022-05-09:
Many thanks for your report.

Please try to reinstall the library and curl:
```sh
apt install --reinstall zlib1g libcurl4 curl
```

doobersn, 2022-05-09:
> Many thanks for your report.
> 
> Please try to reinstall the library and curl:
> 
> ```shell
> apt install --reinstall zlib1g libcurl4 curl
> ```

Excellent, that seems to have solved it, thank you very much! 🙂

Joulinar, 2022-05-09:
keep an eye of your system. There might be more libraries that got damaged or corruption. I'm going to close this issue for now. Feel free to reopen if needed.

closed 2022-05-09 by Joulinar

MichaIng, 2022-05-10:
Jep, have an eye on `dmesg -l emerg,alert,crit,err` for filesystem errors and run an fsck on reboot:
```sh
> /forcefsck
reboot
# to check logs after reboot
journalctl -t systemd-fsck
```