MichaIng/DietPi #5486: Unable to update due to DNS resolution failed 🤔

desegura, 2022-05-11:
#### Details:
- Date           | Wed May 11 16:07:50 CEST 2022
- DietPi version | v8.3.1 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | from scratch
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | `Linux PI4 5.15.32-v8+ #1538 SMP PREEMPT Thu Mar 31 19:40:39 BST 2022 aarch64 GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=0)
- Command        | `ping -nc 1 -W 10 dns9.quad9.net`
- Exit code      | 1
- Software title | DietPi-Update
#### Steps to reproduce:
1. Start rpi
2. Run sudo dietpi-update
#### Expected behaviour:
- I expected a successful update message
#### Actual behaviour:
- Error about *.quad9.net cannot be resolved
#### Extra details:
- I've disabled pi-hole, restarted router, blow connector, etc.
#### Additional logs:
```
PING dns9.quad9.net(2620:fe::9) 56 data bytes

--- dns9.quad9.net ping statistics ---
1 packets transmitted, 0 received, 100% packet loss, time 0ms
```
---------------------------------------------------------------------
[FAILED] DietPi-Update | Unable to continue, DietPi-Update will now terminate.

Joulinar, 2022-05-11:
Pls could you try to disable IPv6 on `dietpi-config` network settings

desegura, 2022-05-15:
@Joulinar fixed disabling IPv6
thanks!

Joulinar, 2022-05-15:
ok I'm going to close this one. Feel free to reopen if needed.

closed 2022-05-15 by Joulinar