MichaIng/DietPi #5488: qBittorrent will randomly crash after a non-specific amount of time

Praxsidian, 2022-05-13:
But report ref code: 5e7cd596-c49d-4849-934a-f7d3667b3961

This has been an ongoing issue for the past few days. I am not sure of the exact time frame in which qBittorrent will crash, but it seems to be within a couple of hours after starting the service back up. Looking through the forums, I've seen other issues describing what I was seeing, and one of the suggestions was to increase the swap file. I've since increased my swap file to 5 GB. I also noticed my Raspberry Pi was getting hot, so I changed some performance settings to try to keep my temps low. Furthermore, I lowered the max TDP to 55 before throttling, changed it to the recommended server setting for the CPU throttle of 85%, and changed the CPU governor to conservative. I'm using jackett, sonarr, and radarr to pull my files for movies and shows, so I do have a couple other services running, but none of those have crashed. I also manage the Pi headless over SSH, and I've turned off the LAN since I'm using Wi-Fi only at the moment.

Joulinar, 2022-05-13:
Some more information on your system would be helpful, that's why we have the questionnaire on the issue template.

#### Required 
- DietPi version | `cat /boot/dietpi/.version`
- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
- Kernel version | `uname -a`
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
- Power supply used | (EG: 5V 1A RAVpower)
- SDcard used | (EG: SanDisk ultra)

Did you checked the service log once Qbit has been crashed? 

```
journalctl -u qbittorrent
```

Praxsidian, 2022-05-13:
- DietPi version | cat /boot/dietpi/.version
```
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=4
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
```
- Distro version | echo $G_DISTRO_NAME $G_RASPBIAN
`bullseye 0`
- Kernel version | uname -a
`Linux meatbone 5.15.32-v8+ #1538 SMP PREEMPT Thu Mar 31 19:40:39 BST 2022 aarch64 GNU/Linux`
- SBC model | echo $G_HW_MODEL_NAME or (EG: RPi3)
`RPi 3 Model B (aarch64)`
- Power supply used | (EG: 5V 1A RAVpower)
`[CanaKit 5V 2.5A Raspberry Pi 3 B+ Power Supply/Adapter](https://www.amazon.com/gp/product/B00MARDJZ4/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1)`
- SDcard used | (EG: SanDisk ultra)
`Samsung 16 GB micro SD card (I've had this for years, I don't remember what model, I believe it came with my Samsung note 5)`

Praxsidian, 2022-05-13:
```
-- Journal begins at Wed 2022-05-11 21:40:00 CDT, ends at Fri 2022-05-13 09:15:24 CDT. --
May 11 21:40:12 meatbone systemd[1]: Started qBittorrent (DietPi).
May 12 03:11:33 meatbone qbittorrent-nox[546]: *************************************************************
May 12 03:11:33 meatbone qbittorrent-nox[546]: Please file a bug report at http://bug.qbittorrent.org and provide the following information:
May 12 03:11:33 meatbone qbittorrent-nox[546]: qBittorrent version: v4.2.5
May 12 03:11:33 meatbone qbittorrent-nox[546]: Caught signal: SIGSEGV
May 12 03:11:33 meatbone qbittorrent-nox[546]: Stack trace:
May 12 03:11:34 meatbone qbittorrent-nox[546]:   /lib/aarch64-linux-gnu/libtorrent-rasterbar.so.10 : ()+0x159318  [0x7fb10b5318]
May 12 03:11:34 meatbone qbittorrent-nox[546]:   /lib/aarch64-linux-gnu/libtorrent-rasterbar.so.10 : ()+0x23cbcc  [0x7fb1198bcc]
May 12 03:11:34 meatbone qbittorrent-nox[546]:   /lib/aarch64-linux-gnu/libtorrent-rasterbar.so.10 : ()+0x1683a4  [0x7fb10c43a4]
May 12 03:11:34 meatbone qbittorrent-nox[546]:   /lib/aarch64-linux-gnu/libtorrent-rasterbar.so.10 : ()+0x263984  [0x7fb11bf984]
May 12 03:11:34 meatbone qbittorrent-nox[546]:   /lib/aarch64-linux-gnu/libtorrent-rasterbar.so.10 : ()+0x1f6230  [0x7fb1152230]
May 12 03:11:34 meatbone qbittorrent-nox[546]:   /lib/aarch64-linux-gnu/libtorrent-rasterbar.so.10 : ()+0x201704  [0x7fb115d704]
May 12 03:11:34 meatbone qbittorrent-nox[546]:   /lib/aarch64-linux-gnu/libtorrent-rasterbar.so.10 : ()+0x1dfa5c  [0x7fb113ba5c]
May 12 03:11:34 meatbone qbittorrent-nox[546]:   /lib/aarch64-linux-gnu/libtorrent-rasterbar.so.10 : ()+0x1dfe00  [0x7fb113be00]
May 12 03:11:34 meatbone qbittorrent-nox[546]:   /lib/aarch64-linux-gnu/libtorrent-rasterbar.so.10 : ()+0xa7530  [0x7fb1003530]
May 12 03:11:34 meatbone qbittorrent-nox[546]:   /lib/aarch64-linux-gnu/libtorrent-rasterbar.so.10 : ()+0x1b022c  [0x7fb110c22c]
May 12 03:11:34 meatbone qbittorrent-nox[546]:   /lib/aarch64-linux-gnu/libtorrent-rasterbar.so.10 : ()+0x1ae0d8  [0x7fb110a0d8]
May 12 03:11:34 meatbone qbittorrent-nox[546]:   /lib/aarch64-linux-gnu/libstdc++.so.6 : ()+0xc9cac  [0x7fb06bacac]
May 12 03:11:34 meatbone qbittorrent-nox[546]:   /lib/aarch64-linux-gnu/libpthread.so.0 : ()+0x8628  [0x7fb0f33628]
May 12 03:11:34 meatbone qbittorrent-nox[546]:   /lib/aarch64-linux-gnu/libc.so.6 : ()+0xd601c  [0x7fb048201c]
May 12 03:11:34 meatbone systemd[1]: qbittorrent.service: Main process exited, code=killed, status=11/SEGV
May 12 03:11:34 meatbone systemd[1]: qbittorrent.service: Failed with result 'signal'.
May 12 03:11:34 meatbone systemd[1]: qbittorrent.service: Consumed 44min 45.605s CPU time.
May 12 07:10:14 meatbone systemd[1]: Started qBittorrent (DietPi).
May 12 15:16:50 meatbone qbittorrent-nox[15796]: *************************************************************
May 12 15:16:50 meatbone qbittorrent-nox[15796]: Please file a bug report at http://bug.qbittorrent.org and provide the following information:
May 12 15:16:50 meatbone qbittorrent-nox[15796]: qBittorrent version: v4.2.5
May 12 15:16:50 meatbone qbittorrent-nox[15796]: Caught signal: SIGSEGV
May 12 15:16:50 meatbone qbittorrent-nox[15796]: Stack trace:
May 12 15:16:51 meatbone qbittorrent-nox[15796]:   /lib/aarch64-linux-gnu/libtorrent-rasterbar.so.10 : ()+0x159318  [0x7f9ccfb318]
May 12 15:16:51 meatbone qbittorrent-nox[15796]:   /lib/aarch64-linux-gnu/libtorrent-rasterbar.so.10 : ()+0x23cbcc  [0x7f9cddebcc]
May 12 15:16:51 meatbone qbittorrent-nox[15796]:   /lib/aarch64-linux-gnu/libtorrent-rasterbar.so.10 : ()+0x1683a4  [0x7f9cd0a3a4]
May 12 15:16:51 meatbone qbittorrent-nox[15796]:   /lib/aarch64-linux-gnu/libtorrent-rasterbar.so.10 : ()+0x263984  [0x7f9ce05984]
May 12 15:16:51 meatbone qbittorrent-nox[15796]:   /lib/aarch64-linux-gnu/libtorrent-rasterbar.so.10 : ()+0x1f6230  [0x7f9cd98230]
May 12 15:16:51 meatbone qbittorrent-nox[15796]:   /lib/aarch64-linux-gnu/libtorrent-rasterbar.so.10 : ()+0x201704  [0x7f9cda3704]
May 12 15:16:51 meatbone qbittorrent-nox[15796]:   /lib/aarch64-linux-gnu/libtorrent-rasterbar.so.10 : ()+0x1dfa5c  [0x7f9cd81a5c]
May 12 15:16:51 meatbone qbittorrent-nox[15796]:   /lib/aarch64-linux-gnu/libtorrent-rasterbar.so.10 : ()+0x1dfe00  [0x7f9cd81e00]
May 12 15:16:51 meatbone qbittorrent-nox[15796]:   /lib/aarch64-linux-gnu/libtorrent-rasterbar.so.10 : ()+0xa7530  [0x7f9cc49530]
May 12 15:16:51 meatbone qbittorrent-nox[15796]:   /lib/aarch64-linux-gnu/libtorrent-rasterbar.so.10 : ()+0x1b022c  [0x7f9cd5222c]
May 12 15:16:51 meatbone qbittorrent-nox[15796]:   /lib/aarch64-linux-gnu/libtorrent-rasterbar.so.10 : ()+0x1ae0d8  [0x7f9cd500d8]
May 12 15:16:51 meatbone qbittorrent-nox[15796]:   /lib/aarch64-linux-gnu/libstdc++.so.6 : ()+0xc9cac  [0x7f9c300cac]
May 12 15:16:51 meatbone qbittorrent-nox[15796]:   /lib/aarch64-linux-gnu/libpthread.so.0 : ()+0x8628  [0x7f9cb79628]
May 12 15:16:51 meatbone qbittorrent-nox[15796]:   /lib/aarch64-linux-gnu/libc.so.6 : ()+0xd601c  [0x7f9c0c801c]
May 12 15:16:51 meatbone systemd[1]: qbittorrent.service: Main process exited, code=killed, status=11/SEGV
May 12 15:16:51 meatbone systemd[1]: qbittorrent.service: Failed with result 'signal'.
May 12 15:16:51 meatbone systemd[1]: qbittorrent.service: Consumed 1h 9min 58.186s CPU time.
May 12 17:51:29 meatbone systemd[1]: Started qBittorrent (DietPi).
May 12 18:00:03 meatbone systemd[1]: Stopping qBittorrent (DietPi)...
May 12 18:00:03 meatbone qbittorrent-nox[30013]: Catching signal: SIGTERM
[... cut ...]
May 13 04:19:25 meatbone qbittorrent-nox[45403]:   /lib/aarch64-linux-gnu/libpthread.so.0 : ()+0x8628  [0x7fa8469628]
May 13 04:19:25 meatbone qbittorrent-nox[45403]:   /lib/aarch64-linux-gnu/libc.so.6 : ()+0xd601c  [0x7fa79b801c]
May 13 04:19:25 meatbone systemd[1]: qbittorrent.service: Main process exited, code=killed, status=11/SEGV
May 13 04:19:25 meatbone systemd[1]: qbittorrent.service: Failed with result 'signal'.
May 13 04:19:25 meatbone systemd[1]: qbittorrent.service: Consumed 3min 50.388s CPU time.
May 13 08:45:50 meatbone systemd[1]: Started qBittorrent (DietPi).
```

MichaIng, 2022-05-14:
Please try to reinstall the `libtorrent-rasterbar10` pcakge, probably the library is corrupted:
```sh
apt install --reinstall libtorrent-rasterbar10
```
RAM usage is fine during downloads?

Praxsidian, 2022-05-14:
I spent about 30 minutes with htop open to see ram usage, as well as intermittently over the entire day, and I never saw the ram usage climb above 700 MB's of use. I do see that swap is being used very lightly. The most I saw was about 10 MB, but other than that, I didn't directly see ram usage hit 1GB ever.

Praxsidian, 2022-05-15:
```
May 15 00:25:20 meatbone qbittorrent-nox[21097]: *************************************************************
May 15 00:25:20 meatbone qbittorrent-nox[21097]: Please file a bug report at http://bug.qbittorrent.org and provide the following information:
May 15 00:25:20 meatbone qbittorrent-nox[21097]: qBittorrent version: v4.2.5
May 15 00:25:20 meatbone qbittorrent-nox[21097]: Caught signal: SIGSEGV
May 15 00:25:20 meatbone qbittorrent-nox[21097]: Stack trace:
May 15 00:25:20 meatbone qbittorrent-nox[21097]:   /lib/aarch64-linux-gnu/libtorrent-rasterbar.so.10 : ()+0x159318  [0x7f964db318]
May 15 00:25:20 meatbone qbittorrent-nox[21097]:   /lib/aarch64-linux-gnu/libtorrent-rasterbar.so.10 : ()+0x23cbcc  [0x7f965bebcc]
May 15 00:25:20 meatbone qbittorrent-nox[21097]:   /lib/aarch64-linux-gnu/libtorrent-rasterbar.so.10 : ()+0xb2904  [0x7f96434904]
May 15 00:25:20 meatbone qbittorrent-nox[21097]:   /lib/aarch64-linux-gnu/libtorrent-rasterbar.so.10 : ()+0xb9a70  [0x7f9643ba70]
May 15 00:25:20 meatbone qbittorrent-nox[21097]:   /lib/aarch64-linux-gnu/libtorrent-rasterbar.so.10 : ()+0xba94c  [0x7f9643c94c]
May 15 00:25:20 meatbone qbittorrent-nox[21097]:   /lib/aarch64-linux-gnu/libtorrent-rasterbar.so.10 : ()+0x16d74c  [0x7f964ef74c]
May 15 00:25:20 meatbone qbittorrent-nox[21097]:   /lib/aarch64-linux-gnu/libtorrent-rasterbar.so.10 : ()+0x16e4d8  [0x7f964f04d8]
May 15 00:25:20 meatbone qbittorrent-nox[21097]:   /lib/aarch64-linux-gnu/libtorrent-rasterbar.so.10 : ()+0x2ddad0  [0x7f9665fad0]
May 15 00:25:20 meatbone qbittorrent-nox[21097]:   /lib/aarch64-linux-gnu/libtorrent-rasterbar.so.10 : ()+0xa7530  [0x7f96429530]
May 15 00:25:20 meatbone qbittorrent-nox[21097]:   /lib/aarch64-linux-gnu/libtorrent-rasterbar.so.10 : ()+0x1b022c  [0x7f9653222c]
May 15 00:25:20 meatbone qbittorrent-nox[21097]:   /lib/aarch64-linux-gnu/libtorrent-rasterbar.so.10 : ()+0x1ae0d8  [0x7f965300d8]
May 15 00:25:20 meatbone qbittorrent-nox[21097]:   /lib/aarch64-linux-gnu/libstdc++.so.6 : ()+0xc9cac  [0x7f95ae0cac]
May 15 00:25:20 meatbone qbittorrent-nox[21097]:   /lib/aarch64-linux-gnu/libpthread.so.0 : ()+0x8628  [0x7f96359628]
May 15 00:25:20 meatbone qbittorrent-nox[21097]:   /lib/aarch64-linux-gnu/libc.so.6 : ()+0xd601c  [0x7f958a801c]
May 15 00:25:20 meatbone systemd[1]: qbittorrent.service: Main process exited, code=killed, status=11/SEGV
May 15 00:25:20 meatbone systemd[1]: qbittorrent.service: Failed with result 'signal'.
May 15 00:25:20 meatbone systemd[1]: qbittorrent.service: Consumed 1min 55.552s CPU time.
May 15 00:25:25 meatbone systemd[1]: qbittorrent.service: Scheduled restart job, restart counter is at 33.
May 15 00:25:25 meatbone systemd[1]: Stopped qBittorrent (DietPi).
May 15 00:25:25 meatbone systemd[1]: qbittorrent.service: Consumed 1min 55.552s CPU time.
May 15 00:25:25 meatbone systemd[1]: Started qBittorrent (DietPi).
```

I have systemd configured to restart the qBittorrent service after 5 seconds, and it appears that qBittorrent fails numerous times each day. According to the counter, it appears to have crashed 32 times yesterday. I did try the fix you suggested around 12 hours ago, when I replied to you, by reinstalling libtorrent-rasterbar10. However, as you can see, whatever is triggering the crash has not stopped.

MichaIng, 2022-05-27:
Let's try to reinstall all related packages. Generally segmentation faults are often caused by corrupted files:
```sh
apt install --reinstall qbittorrent-nox libstdc++6 libc6 libgcc-s1 libicu67 libatomic1
```
Are you able to derive a specific task/action/job qBittorrent does when the crash happens? Probably also the log files contain further info: `/var/log/qbittorrent`

I fail to find some log level/debug setting, sadly 🤔.

Praxsidian, 2022-06-11:
Hey Michalng,

I've reinstalled those packages and this has not resulted in any stability, I'm still experiencing regular crashing. I did not find anything in the qBitTorrent log file, it was empty. When I first opened this issue, I had already stopped using qBitTorrent on my Raspberry Pi, so I'm honestly not sure at this point what is triggering the crash. I have noticed that my Raspberry Pi starts to warm up quite a bit when I was running a torrent or seeding, could it be heat related?

Here is the latest journal log that I pulled. As you can see, it's crashing at an irregular rate. Sometimes it will crash within a few minutes, sometimes it will crash hours apart.

[image]

MichaIng, 2022-06-25:
Okay, thanks for the feedback. I'll mark this as closed then, feel free to reopen if the issue reappears.

closed 2022-06-25 by MichaIng