MichaIng/DietPi #5491: [FAILED] DietPi-Software | ./install.sh --disable-install-webserver

dr4ugr, 2022-05-13:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
```
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=4
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

```
- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
`bullseye 1`
- Kernel version | `uname -a`
`Linux dietpi 5.15.32-v7+ #1538 SMP Thu Mar 31 19:38:48 BST 2022 armv7l GNU/Linux`

- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
`RPi 2 Model B (armv7l)`
- Power supply used | (EG: 5V 1A RAVpower)
`5V 2.5A OEM`
- SD card used | (EG: SanDisk ultra)
`SanDisk Ultra`
#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
`PiHole with UnBound`
- Was the software title installed freshly or updated/migrated?
`New/fresh install`
- Can this issue be replicated on a fresh installation of DietPi?
`Unknown`
 `907886a5-087c-4aa9-b233-482aceb92beb`
- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. Follow `https://www.reddit.com/r/pihole/comments/ap69s0/guide_security_hardened_dietpi_os_for_a_great/`
2. As root, open the dietpi-software desktop icon. Browse software and install pihole and unbound
3. Wait for software to install; around 10 minutes in, see this error
```
[ INFO ] DietPi-Software | ./install.sh --disable-install-webserver, please wait...
/boot/dietpi/func/dietpi-globals: line 850: ./install.sh: Permission denied
[FAILED] DietPi-Software | ./install.sh --disable-install-webserver
```
4. List permissions `root@hostname:/tmp/DietPi-Software# ls -l`
`-rwxr-xr-x 1 root root 119278 May 13 18:22 install.sh`
5. Change permissions to `777` 
`root@hostname:/tmp/DietPi-Software# chmod install.sh`
6. `exit` and attempt to continue installation
7. Note it still fails
8. Chmod `install.sh` back to `755`

#### Expected behaviour

- PiHole and UnBound should install without error. I elected to install the last webserver option (name escapes me)

#### Actual behaviour

- Fails to run `./install.sh` due to permission issues even though I'm running as root

#### Extra details

```
[ INFO ] DietPi-Software | ./install.sh --disable-install-webserver, please wait...
/boot/dietpi/func/dietpi-globals: line 850: ./install.sh: Permission denied
[FAILED] DietPi-Software | ./install.sh --disable-install-webserver
```

Joulinar, 2022-05-14:
Hi,

basically the guide you followed introduced the issue as you set `noexec` permission on temp folder. Same as this one. https://github.com/MichaIng/DietPi/issues/5478#issuecomment-1122759768

You would need to revert this part back to DietPi standard

dr4ugr, 2022-05-14:
okay, yep that was my issue. Thanks (again).

closed 2022-05-14 by dr4ugr

MichaIng, 2022-05-14:
The guide is 3 years old, why do suddenly multiple users run into this issue following that guide? 😄
Was it somehow pushed/bumped?

Joulinar, 2022-05-14:
I left a comment on the guide located inside DietPi community. The one inside PiHole community is already archived. https://www.reddit.com/r/dietpi/comments/ap658g/guide_security_hardened_dietpi_system/

dr4ugr, 2022-05-14:
> The guide is 3 years old, why do suddenly multiple users run into this issue following that guide? 😄
> Was it somehow pushed/bumped? In case I'll ask the author to remove the `noexec` on `/tmp` to stop causing us unnecessary support efforts.

When I started down the path of deploying pihole, I wanted to ensure I didn’t leave any portion of my pi vulnerable to attack. I went with DietPi for its lightweight-ness. But wanted to know if there were ways to further harden.

Moral of the story, however, `noexec` on `tmp` doesn’t add any intrinsic security value and can actually break apps/ programs that compile there!

Joulinar, 2022-05-14:
question is if your system is internet facing? Means do you have some ports open/forward and your system is reachable from the web? If not, some settings don't have a real effect. As well using `chrony` is some kind of not needed, as DietPi has a build in time synchronization process already.

dr4ugr, 2022-05-14:
> question is if your system is internet facing? Means do you have some ports open/forward and your system is reachable from the web? If not, some settings don't have a real effect. As well using `chrony` is some kind of not needed, as DietPi has a build in time synchronization process already.

Yes, `chrony` is a step I skip. It’s not Internet facing yet. I do intend to install tailscale or other VPN…I should probably wait to do “harden” until and if I do so.