MichaIng/DietPi #5497: Issue with Dietpi when starting up

ITCal, 2022-05-17:
#### Details:
- Date           | Tue May 17 01:23:28 BST 2022
- DietPi version | v8.4.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | from scratch
- Hardware       | RPi 4 Model B (armv7l) (ID=4)
- Kernel version | `Linux DietPi 5.15.32-v7l+ #1538 SMP Thu Mar 31 19:39:41 BST 2022 armv7l GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=1)
- Command        | `curl --key  --connect-timeout 8 --retry 1 --retry-delay 4 -sSvT 3a278f66-3cb9-45ae-bce0-ac9d45b4c334.7z sftp://dietpi-survey:upload2dietpi@ssh.dietpi.com:22/bugreport/`
- Exit code      | 28
- Software title | DietPi-BugReport
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
* Resolving timed out after 8000 milliseconds
* Closing connection 0
curl: (28) Resolving timed out after 8000 milliseconds
* Resolving timed out after 8000 milliseconds
* Closing connection 1
curl: (28) Resolving timed out after 8000 milliseconds
```

Joulinar, 2022-05-17:
Please could you share the error you are facing during startup? Because you just shared a DNS error while creating a bug report.

ITCal, 2022-05-18:
[FAILED] DietPi-Software | APT update
 - Command: apt-get -y update
The updates continue to fail

Joulinar, 2022-05-18:
can you share the following output fully

```sh
apt update && apt upgrade
```

ITCal, 2022-05-19:
Hit:1 https://archive.raspberrypi.org/debian bullseye InRelease
Get:2 http://raspbian.raspberrypi.org/raspbian bullseye InRelease [15.0 kB]
Get:3 http://raspbian.raspberrypi.org/raspbian bullseye/main armhf Packages [13.2 MB]
Fetched 13.2 MB in 29s (459 kB/s)
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
2 packages can be upgraded. Run 'apt list --upgradable' to see them.
-bash: upgrade: command not found
root@DietPi:~# Hit:1 https://archive.raspberrypi.org/debian bullseye InRelease
Get:2 http://raspbian.raspberrypi.org/raspbian bullseye InRelease [15.0 kB]
Get:3 http://raspbian.raspberrypi.org/raspbian bullseye/main armhf Packages [13.2 MB]
Fetched 13.2 MB in 29s (459 kB/s)
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
2 packages can be upgraded. Run 'apt list --upgradable' to see them.
-bash: upgrade: command not found

Joulinar, 2022-05-19:
are you sure you copied the command correctly? And `apt upgrade` is not working for you?

Joulinar, 2022-05-19:
the command you execute is incorrect. You need to copy it as I have stated above The command you used
`apt update && upgrade` 
vs 
`apt update && apt `upgrade``

You are missing the `apt` after `&&` and before `upgrade`

ITCal, 2022-05-20:
***@***.***:~# apt update && apt upgrade
Get:1 https://archive.raspberrypi.org/debian bullseye InRelease [23.7 kB]
Get:2 https://archive.raspberrypi.org/debian bullseye/main armhf Packages
[283 kB]
Err:3 http://raspbian.raspberrypi.org/raspbian bullseye InRelease
  Temporary failure resolving 'raspbian.raspberrypi.org'
Fetched 306 kB in 16s (18.7 kB/s)
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
2 packages can be upgraded. Run 'apt list --upgradable' to see them.
W: Failed to fetch
http://raspbian.raspberrypi.org/raspbian/dists/bullseye/InRelease
 Temporary failure resolving 'raspbian.raspberrypi.org'
W: Some index files failed to download. They have been ignored, or old ones
used instead.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Calculating upgrade... Done
The following packages will be upgraded:
  libssl1.1 openssl
2 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Need to get 2108 kB of archives.
After this operation, 36.9 kB of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:1 https://archive.raspberrypi.org/debian bullseye/main armhf libssl1.1
armhf 1.1.1n-0+deb11u2+rpt1 [1292 kB]
Get:2 https://archive.raspberrypi.org/debian bullseye/main armhf openssl
armhf 1.1.1n-0+deb11u2+rpt1 [816 kB]
Fetched 2108 kB in 10s (209 kB/s)
debconf: delaying package configuration, since apt-utils is not installed
(Reading database ... 22155 files and directories currently installed.)
Preparing to unpack .../libssl1.1_1.1.1n-0+deb11u2+rpt1_armhf.deb ...
Unpacking libssl1.1:armhf (1.1.1n-0+deb11u2+rpt1) over (1.1.1n-0+deb11u1)
...
Setting up libssl1.1:armhf (1.1.1n-0+deb11u2+rpt1) ...
(Reading database ... 22155 files and directories currently installed.)
Preparing to unpack .../openssl_1.1.1n-0+deb11u2+rpt1_armhf.deb ...
Unpacking openssl (1.1.1n-0+deb11u2+rpt1) over (1.1.1n-0+deb11u1) ...
Setting up openssl (1.1.1n-0+deb11u2+rpt1) ...
Processing triggers for libc-bin (2.31-13+rpt2+rpi1+deb11u2) ...
***@***.***:~#

Joulinar, 2022-05-20:
looks fine now. Try again as there was a temp issue to resolve hoste name `raspbian.raspberrypi.org`

ITCal, 2022-05-20:
Still having the temp issues

Joulinar, 2022-05-20:
do you use any adblocker software? Or a special DNS server?

ITCal, 2022-05-20:
NO

Joulinar, 2022-05-20:
cat are the DNS server used

`cat /etc/resolv.conf`

Joulinar, 2022-05-20:
ok one of them is your router correct? But what is the one ending `.149` ? Are you using DHCP or STATIC ip?

ITCal, 2022-05-20:
I am unable to identify what is .149 looked at my router and nothing with that last octet came up.

ITCal, 2022-05-20:
I am using DHCP

ITCal, 2022-05-20:
ok .149 is set as primary DNS

Joulinar, 2022-05-20:
but if you are not sure what `.149` is, you should not set it as primary DNS server within DHCP configuration

ITCal, 2022-05-20:
I removed .149 from the config file.

ITCal, 2022-05-20:
root@DietPi:~# apt update && apt upgrade
Hit:1 https://archive.raspberrypi.org/debian bullseye InRelease
Err:2 http://raspbian.raspberrypi.org/raspbian bullseye InRelease
  Temporary failure resolving 'raspbian.raspberrypi.org'
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
All packages are up to date.
W: Failed to fetch http://raspbian.raspberrypi.org/raspbian/dists/bullseye/InRelease  Temporary failure resolving 'raspbian.raspberrypi.org'
W: Some index files failed to download. They have been ignored, or old ones used instead.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Calculating upgrade... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

Joulinar, 2022-05-20:
did you adjusted your DHCP config on the router?? Try to reboot and ensure that `/etc/resolv.conf` is correctly now.

MichaIng, 2022-07-23:
Marking as closed. Feel free to reopen if required.

closed 2022-07-23 by MichaIng