MichaIng/DietPi #5498: Sabnzbd reinstall issue

jaguar489, 2022-05-17:
I am running v 8.42 on a model 4 4GB, I was having issues with sabnzbd loading so I went to reinstall and that didn't work so I did an uninstall and then fresh reinstall and got the error shown in this bug report reference code : ffb05845-cb4e-4f7f-9a9b-578bb4e07177
Thanks,
Nathan

Joulinar, 2022-05-17:
Would it be possible to share the error on GitHub as well? Not everyone has access to the bug report. As well you could have filled the issue template. It has some sense and is not there for fun. 😉

jaguar489, 2022-05-17:
Sorry I somehow created an issue in a way that missed the issue template, comment has been updated

jaguar489, 2022-05-17:
It looks like there could possibly be some corruption involved since I also seem to get this error when attempting to install programs through apt-get install

dpkg: unrecoverable fatal error, aborting:
 files list file for package 'webmin' contains empty filename
E: Sub-process /usr/bin/dpkg returned an error code (2)

Can this be fixed or would I be better off reflashing the card?

Joulinar, 2022-05-17:
you could try following

```
> /forcefsck
reboot
# then after reboot
journalctl -t systemd-fsck
```

this should force a file system check on reboot.

jaguar489, 2022-05-17:
Here are the results from running that

oot@CutiePi:~# journalctl -t systemd-fsck
-- Journal begins at Tue 2022-05-17 18:31:28 EDT, ends at Tue 2022-05-17 18:32:00 EDT. --
May 17 18:31:28 CutiePi systemd-fsck[139]: Please pass 'fsck.mode=force' on the kernel command line rather than creating /forcefsck on the root file system.
May 17 18:31:28 CutiePi systemd-fsck[161]: e2fsck 1.46.2 (28-Feb-2021)
May 17 18:31:28 CutiePi systemd-fsck[161]: Pass 1: Checking inodes, blocks, and sizes
May 17 18:31:31 CutiePi systemd-fsck[161]: Pass 2: Checking directory structure
May 17 18:31:36 CutiePi systemd-fsck[161]: Pass 3: Checking directory connectivity
May 17 18:31:36 CutiePi systemd-fsck[161]: Pass 4: Checking reference counts
May 17 18:31:36 CutiePi systemd-fsck[161]: Pass 5: Checking group summary information
May 17 18:31:36 CutiePi systemd-fsck[161]: /dev/mmcblk0p2: 348359/1942080 files (0.2% non-contiguous), 3699575/7797760 blocks
May 17 18:31:38 CutiePi systemd-fsck[264]: Please pass 'fsck.mode=force' on the kernel command line rather than creating /forcefsck on the root file system.
May 17 18:31:38 CutiePi systemd-fsck[271]: fsck.fat 4.2 (2021-01-31)
May 17 18:31:38 CutiePi systemd-fsck[271]: /dev/mmcblk0p1: 352 files, 64666/258077 clusters

MichaIng, 2022-05-18:
Is it possible to reinstall webmin?
```sh
apt install --reinstall webmin
```

closed 2022-05-18 by jaguar489

Joulinar, 2022-05-18:
probably the best to start with a new image to bypass the corruption.

MichaIng, 2022-05-19:
If it was only the corrupted webmin package file list, it would have been possible to remove it, including the status entry, and reinstall the package as if it was not installed before. But probably other files were affected as well.