MichaIng/DietPi #5501: dietpi VPN failed to launch using ProtonVPN+ credentials

Wilford888, 2022-05-18:
#### Details:
- Date           | Wed May 18 01:29:36 EDT 2022
- Bug report     | 778c1c40-07c3-451d-85f5-61a470c7380b
- DietPi version | v8.4.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Armbian
- Hardware       | ROCKPro64 (aarch64) (ID=42)
- Kernel version | `Linux DietPi 5.15.25-rockchip64 #22.02.1 SMP PREEMPT Sun Feb 27 09:05:47 UTC 2022 aarch64 GNU/Linux`
- Distro         | bullseye (ID=6)
- Command        | `eval envsubst < /etc/openvpn/protonvpn/template.ovpn > /etc/openvpn/client.ovpn`
- Exit code      | 127
- Software title | DietPi-VPN
#### Steps to reproduce:

1. launch diet-pi vpn
2. configure for use with paid ProtonVPN account
3. save settings and start
#### Expected behaviour:

- VPN should launch
#### Actual behaviour:

- Got an error message, which is shown below line 858 ennsubst:  command not found
#### Extra details:

- ...
#### Additional logs:
```
/boot/dietpi/func/dietpi-globals: line 858: envsubst: command not found
```

Joulinar, 2022-05-18:
can you be a little more specific what you have entered exactly? Did you entered username + password? and selected a VPN server? Or did you specify any up or down script as well? Once hit by the error, can you scroll up a little bit on your SSH terminal? There you should see the commands executed. Could you share them as well pls.

MichaIng, 2022-05-19:
That command is part of the `gettext-base` package: https://packages.debian.org/bullseye/amd64/gettext-base/filelist
Indeed it is missing to be installed, in case. I'll add that, respectively replace the `envsubst` call with a simple `sed`. To solve in your case:
```sh
apt install gettext-base
```

MichaIng, 2022-05-21:
Fixed with https://github.com/MichaIng/DietPi/commit/146e4a1, using good old `sed` 😉.

closed 2022-05-27 by MichaIng