MichaIng/DietPi #5502: issue with  /etc/apt/sources.list.d/your-repo-name.list

KenMoreira, 2022-05-20:
#### Details:
- Date           | Fri May 20 02:54:12 BST 2022
- DietPi version | v7.5.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi CM 3+ (armv7l) (ID=3)
- Kernel version | `Linux KensBridge 4.19.73-v7+ #1266 SMP Tue Sep 17 20:52:18 BST 2019 armv7l GNU/Linux`
- Distro         | stretch (ID=4,RASPBIAN=1)
- Command        | `apt-get -q update`
- Exit code      | 100
- Software title | DietPi-Update
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
E: Type 'Building' is not known on line 1 in source list /etc/apt/sources.list.d/your-repo-name.list
E: The list of sources could not be read.

Joulinar, 2022-05-20:
Is this a 3rd party repository you added manually? Can you share the content?

`cat /etc/apt/sources.list.d/your-repo-name.list`

KenMoreira, 2022-05-20:
root@KensBridge:~# cat /etc/apt/sources.list.d/your-repo-name.list
cat: /etc/apt/sources.list.d/your-repo-name.list: No such file or directory
root@KensBridge:~#

so the issue i originally had on its own disappeared and now that file is gone? 
so i re ran dietpi-update and this NEW issue appears

[image]

Joulinar, 2022-05-20:
ok your system is quite old and you are running outdated Debian Stretch still. Therefore you have an issue with a PHP repository that we don't use anymore. 

You should be able to work around as follow:

apt install --reinstall ca-certificates
apt-key del '95BD4743'
curl -sSfL 'https://packages.sury.org/php/apt.gpg' -o /etc/apt/trusted.gpg.d/dietpi-php.gpg
/boot/dietpi/func/dietpi-set_software apt-cache clean
apt update

Another hint, you could have a look to our Stretch update guide once issues has been fixed
https://dietpi.com/docs/usage/#how-to-upgrade-to-buster

MichaIng, 2022-05-20:
The upmpdcli repository is faulty as well, and you added it for "buster" while you run "stretch". Check out the instructions [here](https://www.lesbonscomptes.com/upmpdcli/pages/downloads.html#debian) and update the list file and GPG key accordingly.

KenMoreira, 2022-05-20:
[image]

KenMoreira, 2022-05-20:
alot of the commands i try to run from help pages don't work

MichaIng, 2022-05-21:
Did you follow the link I posted above to update the upmpdcli repo list and key?

Joulinar, 2022-05-21:
maybe you can share the available keys as well

```
apt-key list
```

MichaIng, 2022-05-21:
> maybe you can share the available keys as well

It is not only the key which is invalid, the repository itself is wrong.

MichaIng, 2022-05-21:
... ah no actually the repo is right (not sure about the repo key), but the HTTPS certificate is not detected correctly. I wonder whether this is the OpenSSL issue with the obsolete/expired Let's Encrypt root certificate.

@KenMoreira 
Could you please try to upgrade all other packages manually first, see whether OpenSSL or libSSL are upgraded:
```sh
apt upgrade
```

KenMoreira, 2022-05-21:
[image]

KenMoreira, 2022-05-21:
[image]

[image]

Joulinar, 2022-05-21:
If possible try to avoid screen prints. You should be able to copy everything directly from SSH terminal. This would help us a lot and increase visibility.

KenMoreira, 2022-05-21:
root@KensBridge:~# apt update
Get:1 http://raspbian.raspberrypi.org/raspbian stretch InRelease [15.0 kB]
Hit:3 https://archive.raspberrypi.org/debian stretch InRelease
Get:4 http://raspbian.raspberrypi.org/raspbian stretch/main armhf Packages [11.7                                                                                                                                                              MB]
Ign:2 https://www.lesbonscomptes.com/upmpdcli/downloads/raspbian buster InReleas                                                                                                                                                             e
Err:5 https://www.lesbonscomptes.com/upmpdcli/downloads/raspbian buster Release
  server certificate verification failed. CAfile: /etc/ssl/certs/ca-certificates.crt CRLfile: none
Reading package lists... Done
N: Ignoring file 'your-repo-name.list.save.1' in directory '/etc/apt/sources.list.d/' as it has an invalid filename extension
E: The repository 'http://www.lesbonscomptes.com/upmpdcli/downloads/raspbian buster Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.

KenMoreira, 2022-05-21:
root@KensBridge:~# apt-key list
/etc/apt/trusted.gpg
--------------------
pub   rsa2048 2012-04-01 [SC]
      A0DA 38D0 D76E 8B5D 6388  7281 9165 938D 90FD DD2E
uid           [ unknown] Mike Thompson (Raspberry Pi Debian armhf ARMv6+VFP) <mpthompson@gmail.com>
sub   rsa2048 2012-04-01 [E]

pub   rsa2048 2012-06-17 [SC]
      CF8A 1AF5 02A2 AA2D 763B  AE7E 82B1 2992 7FA3 303E
uid           [ unknown] Raspberry Pi Archive Signing Key
sub   rsa2048 2012-06-17 [E]

pub   rsa4096 2016-02-21 [SC] [expires: 2022-12-06]
      F8E3 3472 5692 2A8A E767  605B 7808 CE96 D38B 9201
uid           [ unknown] Jean-Francois Dockes <jf@dockes.org>
uid           [ unknown] Jean-Francois Dockes <jfdockes@gmail.com>
sub   rsa4096 2016-02-21 [E] [expires: 2022-12-06]

KenMoreira, 2022-05-21:
root@KensBridge:~# apt upgrade
Reading package lists... Done
Building dependency tree
Reading state information... Done
Calculating upgrade... Done
The following packages will be upgraded:
  bind9-host cron curl gzip libavcodec57 libavfilter6 libavformat57 libavresample3 libavutil55 libbind9-140 libcups2 libcurl3 libcurl3-gnutls libdns-export162 libdns162 libelf1 libexpat1 libfaad2 libflac8 libfribidi0 libgmp10
  libgnutls30 libgssapi-krb5-2 libhogweed4 libicu57 libisc-export160 libisc160 libisccc140 libisccfg140 libk5crypto3 libkrb5-3 libkrb5support0 liblwres141 liblzma5 libnettle6 libnghttp2-14 libopencv-calib3d2.4v5 libopencv-core2.4v5
  libopencv-features2d2.4v5 libopencv-flann2.4v5 libopencv-highgui2.4-deb0 libopencv-imgproc2.4v5 libopencv-objdetect2.4v5 libopencv-video2.4v5 libopenexr22 libopenjp2-7 libpcap0.8 libpostproc54 libpython2.7 libpython2.7-minimal
  libpython2.7-stdlib libsamplerate0 libsasl2-2 libsasl2-modules-db libssh2-1 libssl-dev libssl1.0.2 libssl1.1 libswresample2 libswscale4 libtag1v5 libtag1v5-vanilla libtiff5 libvorbis0a libvorbisenc2 libvorbisfile3 libvpx4 lighttpd
  mariadb-client-10.1 mariadb-client-core-10.1 mariadb-common mariadb-server mariadb-server-10.1 mariadb-server-core-10.1 openssl python2.7 python2.7-minimal rsync tar tzdata xz-utils zlib1g zlib1g-dev
83 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Need to get 4,127 kB/56.4 MB of archives.
After this operation, 87.0 kB of additional disk space will be used.
N: Ignoring file 'your-repo-name.list.save.1' in directory '/etc/apt/sources.list.d/' as it has an invalid filename extension
Do you want to continue? [Y/n] y
Err:1 https://mirrors.switch.ca/raspbian/raspbian stretch/main armhf libssl-dev armhf 1.1.0l-1~deb9u6
  server certificate verification failed. CAfile: /etc/ssl/certs/ca-certificates.crt CRLfile: none
Err:2 https://mirrors.switch.ca/raspbian/raspbian stretch/main armhf libssl1.1 armhf 1.1.0l-1~deb9u6
  server certificate verification failed. CAfile: /etc/ssl/certs/ca-certificates.crt CRLfile: none
Err:3 https://mirrors.switch.ca/raspbian/raspbian stretch/main armhf libssl1.0.2 armhf 1.0.2u-1~deb9u7
  server certificate verification failed. CAfile: /etc/ssl/certs/ca-certificates.crt CRLfile: none
Err:4 https://mirrors.switch.ca/raspbian/raspbian stretch/main armhf openssl armhf 1.1.0l-1~deb9u6
  server certificate verification failed. CAfile: /etc/ssl/certs/ca-certificates.crt CRLfile: none
E: Failed to fetch https://mirrors.switch.ca/raspbian/raspbian/pool/main/o/openssl/libssl-dev_1.1.0l-1~deb9u6_armhf.deb  server certificate verification failed. CAfile: /etc/ssl/certs/ca-certificates.crt CRLfile: none
E: Failed to fetch https://mirrors.switch.ca/raspbian/raspbian/pool/main/o/openssl/libssl1.1_1.1.0l-1~deb9u6_armhf.deb  server certificate verification failed. CAfile: /etc/ssl/certs/ca-certificates.crt CRLfile: none
E: Failed to fetch https://mirrors.switch.ca/raspbian/raspbian/pool/main/o/openssl1.0/libssl1.0.2_1.0.2u-1~deb9u7_armhf.deb  server certificate verification failed. CAfile: /etc/ssl/certs/ca-certificates.crt CRLfile: none
E: Failed to fetch https://mirrors.switch.ca/raspbian/raspbian/pool/main/o/openssl/openssl_1.1.0l-1~deb9u6_armhf.deb  server certificate verification failed. CAfile: /etc/ssl/certs/ca-certificates.crt CRLfile: none
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?

MichaIng, 2022-05-21:
It is indeed the Let's Encrypt R3, respectively the unintentionally pulled obsolete certification chain pulled by the unpatched OpenSSL version, still appended by Certbot server-side (for backwards-compatibility reasons I did never understand...).

Let's see how to solve this best. Please try to switch from the Raspbian mirror you use to `raspbian.raspberrypi.org`:
```sh
sed -i 's|https://mirrors.switch.ca/raspbian/raspbian|http://raspbian.raspberrypi.org/raspbian|' /etc/apt/sources.list
apt update
apt upgrade
```
Then you should be able to switch back to the mirror, which has the benefit of being HTTPS:
```sh
sed -i 's|http://raspbian.raspberrypi.org/raspbian|https://mirrors.switch.ca/raspbian/raspbian' /etc/apt/sources.list
apt update
```

There seems to be an obsolete dummy/backup file you may want to remove (check it's content first, whether it is of use for you):
```sh
rm /etc/apt/sources.list.d/your-repo-name.list.save.1
```

KenMoreira, 2022-05-21:
sed -i 's|https://mirrors.switch.ca/raspbian/raspbian|http://raspbian.raspberrypi.org/raspbian|' /etc/apt/sources.list

has no affect
it just takes me to another 
root@ prompt

KenMoreira, 2022-05-21:
root@KensBridge:~# sed -i 's|https://mirrors.switch.ca/raspbian/raspbian|http://raspbian.raspberrypi.org/raspbian|' /etc/apt/sources.list
root@KensBridge:~# sed -i 's|http://raspbian.raspberrypi.org/raspbian|https://mirrors.switch.ca/raspbian/raspbian|' /etc/apt/sources.list
root@KensBridge:~#

Joulinar, 2022-05-21:
these commands are there to change configuration files. They will not execute anything. Therefore it is fine to just see the promt. Pls run following afterwards

```
apt update
```

KenMoreira, 2022-05-21:
root@KensBridge:~# sed -i 's|https://mirrors.switch.ca/raspbian/raspbian|http://                                                                                                                                                             raspbian.raspberrypi.org/raspbian|' /etc/apt/sources.list
root@KensBridge:~# apt update
Hit:1 http://raspbian.raspberrypi.org/raspbian stretch InRelease
Hit:3 https://archive.raspberrypi.org/debian stretch InRelease
Ign:2 https://www.lesbonscomptes.com/upmpdcli/downloads/raspbian buster InReleas                                                                                                                                                             e
Err:4 https://www.lesbonscomptes.com/upmpdcli/downloads/raspbian buster Release
  server certificate verification failed. CAfile: /etc/ssl/certs/ca-certificates                                                                                                                                                             .crt CRLfile: none
Reading package lists... Done
E: The repository 'http://www.lesbonscomptes.com/upmpdcli/downloads/raspbian buster Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
root@KensBridge:~#

KenMoreira, 2022-05-21:
root@KensBridge:~# apt upgrade
Reading package lists... Done
Building dependency tree
Reading state information... Done
Calculating upgrade... Done
The following packages will be upgraded:
  bind9-host cron curl gzip libavcodec57 libavfilter6 libavformat57 libavresample3 libavutil55 libbind9-140 libcups2 libcurl3 libcurl3-gnutls libdns-export162 libdns162 libelf1 libexpat1 libfaad2 libflac8 libfribidi0 libgmp10
  libgnutls30 libgssapi-krb5-2 libhogweed4 libicu57 libisc-export160 libisc160 libisccc140 libisccfg140 libk5crypto3 libkrb5-3 libkrb5support0 liblwres141 liblzma5 libnettle6 libnghttp2-14 libopencv-calib3d2.4v5 libopencv-core2.4v5
  libopencv-features2d2.4v5 libopencv-flann2.4v5 libopencv-highgui2.4-deb0 libopencv-imgproc2.4v5 libopencv-objdetect2.4v5 libopencv-video2.4v5 libopenexr22 libopenjp2-7 libpcap0.8 libpostproc54 libpython2.7 libpython2.7-minimal
  libpython2.7-stdlib libsamplerate0 libsasl2-2 libsasl2-modules-db libssh2-1 libssl-dev libssl1.0.2 libssl1.1 libswresample2 libswscale4 libtag1v5 libtag1v5-vanilla libtiff5 libvorbis0a libvorbisenc2 libvorbisfile3 libvpx4 lighttpd
  mariadb-client-10.1 mariadb-client-core-10.1 mariadb-common mariadb-server mariadb-server-10.1 mariadb-server-core-10.1 openssl python2.7 python2.7-minimal rsync tar tzdata xz-utils zlib1g zlib1g-dev
83 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Need to get 4,127 kB/56.4 MB of archives.
After this operation, 87.0 kB of additional disk space will be used.
Do you want to continue? [Y/n] y
Err:1 https://mirrors.switch.ca/raspbian/raspbian stretch/main armhf libssl-dev armhf 1.1.0l-1~deb9u6
  server certificate verification failed. CAfile: /etc/ssl/certs/ca-certificates.crt CRLfile: none
Err:2 https://mirrors.switch.ca/raspbian/raspbian stretch/main armhf libssl1.1 armhf 1.1.0l-1~deb9u6
  server certificate verification failed. CAfile: /etc/ssl/certs/ca-certificates.crt CRLfile: none
Err:3 https://mirrors.switch.ca/raspbian/raspbian stretch/main armhf libssl1.0.2 armhf 1.0.2u-1~deb9u7
  server certificate verification failed. CAfile: /etc/ssl/certs/ca-certificates.crt CRLfile: none
Err:4 https://mirrors.switch.ca/raspbian/raspbian stretch/main armhf openssl armhf 1.1.0l-1~deb9u6
  server certificate verification failed. CAfile: /etc/ssl/certs/ca-certificates.crt CRLfile: none
E: Failed to fetch https://mirrors.switch.ca/raspbian/raspbian/pool/main/o/openssl/libssl-dev_1.1.0l-1~deb9u6_armhf.deb  server certificate verification failed. CAfile: /etc/ssl/certs/ca-certificates.crt CRLfile: none
E: Failed to fetch https://mirrors.switch.ca/raspbian/raspbian/pool/main/o/openssl/libssl1.1_1.1.0l-1~deb9u6_armhf.deb  server certificate verification failed. CAfile: /etc/ssl/certs/ca-certificates.crt CRLfile: none
E: Failed to fetch https://mirrors.switch.ca/raspbian/raspbian/pool/main/o/openssl1.0/libssl1.0.2_1.0.2u-1~deb9u7_armhf.deb  server certificate verification failed. CAfile: /etc/ssl/certs/ca-certificates.crt CRLfile: none
E: Failed to fetch https://mirrors.switch.ca/raspbian/raspbian/pool/main/o/openssl/openssl_1.1.0l-1~deb9u6_armhf.deb  server certificate verification failed. CAfile: /etc/ssl/certs/ca-certificates.crt CRLfile: none
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?

KenMoreira, 2022-05-21:
root@KensBridge:~# sed -i 's|http://raspbian.raspberrypi.org/raspbian|https://mirrors.switch.ca/raspbian/raspbian|' /etc/apt/sources.list
root@KensBridge:~# apt update
Ign:1 https://mirrors.switch.ca/raspbian/raspbian stretch InRelease
Err:3 https://mirrors.switch.ca/raspbian/raspbian stretch Release
  server certificate verification failed. CAfile: /etc/ssl/certs/ca-certificates.crt CRLfile: none
Hit:4 https://archive.raspberrypi.org/debian stretch InRelease
Ign:2 https://www.lesbonscomptes.com/upmpdcli/downloads/raspbian buster InRelease
Err:5 https://www.lesbonscomptes.com/upmpdcli/downloads/raspbian buster Release
  server certificate verification failed. CAfile: /etc/ssl/certs/ca-certificates.crt CRLfile: none
Reading package lists... Done
E: The repository 'https://mirrors.switch.ca/raspbian/raspbian stretch Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
E: The repository 'http://www.lesbonscomptes.com/upmpdcli/downloads/raspbian buster Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.

MichaIng, 2022-05-21:
> Err:4 https://www.lesbonscomptes.com/upmpdcli/downloads/raspbian

You still have this repository inside. Check in `/etc/apt/sources.list.d` where it is located, and comment out the line by putting a `#` at the start. Then repeat the commands provided above: https://github.com/MichaIng/DietPi/issues/5502#issuecomment-1133656809

KenMoreira, 2022-05-21:
root@KensBridge~# sed -i 's|https://mirrors.switch.ca/raspbian/raspbian|http://raspbian.raspberrypi.org/raspbian|' /etc/apt/sources.list

root@KensBridge:~# apt update
Get:2 http://raspbian.raspberrypi.org/raspbian stretch InRelease [15.0 kB]
Ign:1 https://www.lesbonscomptes.com/upmpdcli/downloads/raspbian buster InRelease
Hit:3 https://archive.raspberrypi.org/debian stretch InRelease
Err:4 https://www.lesbonscomptes.com/upmpdcli/downloads/raspbian buster Release
  server certificate verification failed. CAfile: /etc/ssl/certs/ca-certificates.crt CRLfile: none
Reading package lists... Done
E: The repository 'http://www.lesbonscomptes.com/upmpdcli/downloads/raspbian buster Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
root@KensBridge:~# sed -i 's|https://mirrors.switch.ca/raspbian/raspbian|http://raspbian.raspberrypi.org/raspbian|' /etc/apt/sources.list

root@KensBridge:~# apt upgrade
Reading package lists... Done
Building dependency tree
Reading state information... Done
Calculating upgrade... Done

root@KensBridge:~# sed -i 's|http://raspbian.raspberrypi.org/raspbian|https://mirrors.switch.ca/raspbian/raspbian|' /etc/apt/sources.list

root@KensBridge:~# apt update
Ign:1 https://mirrors.switch.ca/raspbian/raspbian stretch InRelease
Err:3 https://mirrors.switch.ca/raspbian/raspbian stretch Release
  server certificate verification failed. CAfile: /etc/ssl/certs/ca-certificates.crt CRLfile: none
Hit:4 https://archive.raspberrypi.org/debian stretch InRelease
Ign:2 https://www.lesbonscomptes.com/upmpdcli/downloads/raspbian buster InRelease
Err:5 https://www.lesbonscomptes.com/upmpdcli/downloads/raspbian buster Release
  server certificate verification failed. CAfile: /etc/ssl/certs/ca-certificates.crt CRLfile: none
Reading package lists... Done
E: The repository 'https://mirrors.switch.ca/raspbian/raspbian stretch Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
E: The repository 'http://www.lesbonscomptes.com/upmpdcli/downloads/raspbian buster Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.

MichaIng, 2022-05-21:
So the package upgrades which were available [above](https://github.com/MichaIng/DietPi/issues/5502#issuecomment-1133668386) were applied already?

KenMoreira, 2022-05-21:
I just copy n paste what it says. I don't always understand it

but when i do dietpi-update i still get errors. so i exit

MichaIng, 2022-05-21:
Okay let's get the DietPi update through first by also disabling the failing 3rd party repository:
```sh
sed -i 's|https://mirrors.switch.ca/raspbian/raspbian|http://raspbian.raspberrypi.org/raspbian|' /etc/apt/sources.list
for i in /etc/apt/sources.list.d/*.list; do grep -q 'lesbonscomptes' "$i" && mv "$i"{,.bak}
apt update
apt full-upgrade
dietpi-update
```

KenMoreira, 2022-05-21:
APT upgrade                                                                                                          │
           │  - Command: apt-get -qq upgrade                                                                                      │
           │  - Exit code: 100                                                                                                    │
           │  - DietPi version: v7.5.2 (MichaIng/master) | HW_MODEL: 3 | HW_ARCH: 2 | DISTRO: 4                                   │
           │  - Image creator: DietPi Core Team                                                                                   │
           │  - Pre-image: Raspbian Lite                                                                                          │
           │  - Error log:                                                                                                        │
           │ E: Failed to fetch                                                                                                   │
           │ https://mirrors.switch.ca/raspbian/raspbian/pool/main/o/openssl/libssl-dev_1.1.0l-1~deb9u6_armhf.deb  server         │
           │ certificate verification failed. CAfile: /etc/ssl/certs/ca-certificates.crt CRLfile: none                            │
           │ E: Failed to fetch                                                                                                   │
           │ https://mirrors.switch.ca/raspbian/raspbian/pool/main/o/openssl/libssl1.1_1.1.0l-1~deb9u6_armhf.deb  server          │
           │ certificate verification failed. CAfile: /etc/ssl/certs/ca-certificates.crt CRLfile: none                            │
           │ E: Failed to fetch                                                                                                   │
           │ https://mirrors.switch.ca/raspbian/raspbian/pool/main/o/openssl1.0/libssl1.0.2_1.0.2u-1~deb9u7_armhf.deb

MichaIng, 2022-05-21:
You did run the first command? Please show:
```sh
for i in /etc/apt/sources.list{,.d/*.list}; do echo "$i:"; cat "$i"; done
```

KenMoreira, 2022-05-21:
root@KensBridge:~# sed -i 's|https://mirrors.switch.ca/raspbian/raspbian|http://raspbian.raspberrypi.org/raspbian|' /etc/apt/sources.list
root@KensBridge:~# for i in /etc/apt/sources.list.d/*.list; do grep -q 'lesbonscomptes' "$i" && mv "$i"{,.bak}; done
root@KensBridge:~#

KenMoreira, 2022-05-21:
i did do all of it , yes.

Joulinar, 2022-05-21:
can you share following

```
cat /etc/apt/sources.list
```

MichaIng, 2022-05-21:
Or the command I suggested above to get the full picture: https://github.com/MichaIng/DietPi/issues/5502#issuecomment-1133760729

KenMoreira, 2022-05-21:
root@KensBridge:~# cat /etc/apt/sources.list
deb http://raspbian.raspberrypi.org/raspbian stretch main contrib non-free rpi
root@KensBridge:~# sed -i 's|https://mirrors.switch.ca/raspbian/raspbian|http://raspbian.raspberrypi.org/raspbian|' /etc/apt/sources.list
root@KensBridge:~# for i in /etc/apt/sources.list.d/*.list; do grep -q 'lesbonscomptes' "$i" && mv "$i"{,.bak}; done

Joulinar, 2022-05-21:
pls can you share following to get a full picture on your configuration

```
for i in /etc/apt/sources.list{,.d/*.list}; do echo "$i:"; cat "$i"; done
```

KenMoreira, 2022-05-21:
> for i in /etc/apt/sources.list{,.d/*.list}; do echo "$i:"; cat "$i"; done

root@KensBridge:~# for i in /etc/apt/sources.list{,.d/*.list}; do echo "$i:"; ca                                                                                                                                                             t "$i"; done
/etc/apt/sources.list:
deb http://raspbian.raspberrypi.org/raspbian stretch main contrib non-free rpi
/etc/apt/sources.list.d/raspi.list:
deb https://archive.raspberrypi.org/debian/ stretch main ui
root@KensBridge:~#

MichaIng, 2022-05-22:
Hmm, no `https://mirrors.switch.ca`, so why is it still pulling packages from there (question to myself). Please try:
```sh
/boot/dietpi/func/dietpi-set_software apt-cache clean
apt update
apt full-upgrade
```

Joulinar, 2022-05-22:
Maybe it is redirected from a global mirror?

MichaIng, 2022-05-22:
I've never seen this being done transparently, but you are right, actually from the first posts outputs it looks like `https://mirrors.switch.ca` was never an actual repo list entry. Probably the used mirror is shown only in case of an error.

Okay so then we need to use an actual mirror instead of the mirror director:
```sh
sed -i 's|http://raspbian.raspberrypi.org/raspbian|http://raspbian.mirror.globo.tech/raspbian|' /etc/apt/sources.list
/boot/dietpi/func/dietpi-set_software apt-cache clean
apt update
apt full-upgrade
```

KenMoreira, 2022-05-22:
root@KensBridge:~# /boot/dietpi/func/dietpi-set_software apt-cache clean

 DietPi-Set_software

 Mode: apt-cache (clean)

[  OK  ] apt-cache clean | Completed

root@KensBridge
:~# apt update

Get:1 http://raspbian.raspberrypi.org/raspbian stretch InRelease [15.0 kB]

Get:2 http://raspbian.raspberrypi.org/raspbian stretch/contrib armhf Packages [5                                                            6.9 kB]

Get:3 https://archive.raspberrypi.org/debian stretch InRelease [25.3 kB]

Get:4 http://raspbian.raspberrypi.org/raspbian stretch/non-free armhf Packages [                                                            98.9 kB]

Get:5 https://archive.raspberrypi.org/debian stretch/main armhf Packages [192 kB                                                            ]

Get:6 http://raspbian.raspberrypi.org/raspbian stretch/rpi armhf Packages [1,360                                                             B]

Get:7 http://raspbian.raspberrypi.org/raspbian stretch/main armhf Packages [11.7                                                             MB]

Get:8 https://archive.raspberrypi.org/debian stretch/ui armhf Packages [44.6 kB]

Fetched 12.1 MB in 10s (1,158 kB/s)

Reading package lists... Done

Building dependency tree

Reading state information... Done

83 packages can be upgraded. Run 'apt list --upgradable' to see them.
root@KensBridge:~# apt full upgrade

E: Invalid operation full

root@KensBridge:~# apt full-upgrade

Reading package lists... Done

Building dependency tree

Reading state information... Done

Calculating upgrade... Done

MichaIng, 2022-05-22:
It needs to be `apt full-upgrade` with a dash 😉.

MichaIng, 2022-05-22:
There is no confirmation after the last command to apply those 83 available upgrades?

KenMoreira, 2022-05-22:
The following packages will be upgraded:
  bind9-host cron curl gzip libavcodec57 libavfilter6 libavformat57 libavresample3 libavutil55 libbind9-140 libcups2 libcurl3
  libcurl3-gnutls libdns-export162 libdns162 libelf1 libexpat1 libfaad2 libflac8 libfribidi0 libgmp10 libgnutls30 libgssapi-krb5-2
  libhogweed4 libicu57 libisc-export160 libisc160 libisccc140 libisccfg140 libk5crypto3 libkrb5-3 libkrb5support0 liblwres141 liblzma5
  libnettle6 libnghttp2-14 libopencv-calib3d2.4v5 libopencv-core2.4v5 libopencv-features2d2.4v5 libopencv-flann2.4v5
  libopencv-highgui2.4-deb0 libopencv-imgproc2.4v5 libopencv-objdetect2.4v5 libopencv-video2.4v5 libopenexr22 libopenjp2-7 libpcap0.8
  libpostproc54 libpython2.7 libpython2.7-minimal libpython2.7-stdlib libsamplerate0 libsasl2-2 libsasl2-modules-db libssh2-1 libssl-dev
  libssl1.0.2 libssl1.1 libswresample2 libswscale4 libtag1v5 libtag1v5-vanilla libtiff5 libvorbis0a libvorbisenc2 libvorbisfile3 libvpx4
  lighttpd mariadb-client-10.1 mariadb-client-core-10.1 mariadb-common mariadb-server mariadb-server-10.1 mariadb-server-core-10.1 openssl
  python2.7 python2.7-minimal rsync tar tzdata xz-utils zlib1g zlib1g-dev
83 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Need to get 4,127 kB/56.4 MB of archives.
After this operation, 87.0 kB of additional disk space will be used.
Do you want to continue? [Y/n] y
Err:1 https://mirrors.switch.ca/raspbian/raspbian stretch/main armhf libssl-dev armhf 1.1.0l-1~deb9u6
  server certificate verification failed. CAfile: /etc/ssl/certs/ca-certificates.crt CRLfile: none
Err:2 https://mirrors.switch.ca/raspbian/raspbian stretch/main armhf libssl1.1 armhf 1.1.0l-1~deb9u6
  server certificate verification failed. CAfile: /etc/ssl/certs/ca-certificates.crt CRLfile: none
Err:3 https://mirrors.switch.ca/raspbian/raspbian stretch/main armhf libssl1.0.2 armhf 1.0.2u-1~deb9u7
  server certificate verification failed. CAfile: /etc/ssl/certs/ca-certificates.crt CRLfile: none
Err:4 https://mirrors.switch.ca/raspbian/raspbian stretch/main armhf openssl armhf 1.1.0l-1~deb9u6
  server certificate verification failed. CAfile: /etc/ssl/certs/ca-certificates.crt CRLfile: none
E: Failed to fetch https://mirrors.switch.ca/raspbian/raspbian/pool/main/o/openssl/libssl-dev_1.1.0l-1~deb9u6_armhf.deb  server certificate verification failed. CAfile: /etc/ssl/certs/ca-certificates.crt CRLfile: none
E: Failed to fetch https://mirrors.switch.ca/raspbian/raspbian/pool/main/o/openssl/libssl1.1_1.1.0l-1~deb9u6_armhf.deb  server certificate verification failed. CAfile: /etc/ssl/certs/ca-certificates.crt CRLfile: none
E: Failed to fetch https://mirrors.switch.ca/raspbian/raspbian/pool/main/o/openssl1.0/libssl1.0.2_1.0.2u-1~deb9u7_armhf.deb  server certificate verification failed. CAfile: /etc/ssl/certs/ca-certificates.crt CRLfile: none
E: Failed to fetch https://mirrors.switch.ca/raspbian/raspbian/pool/main/o/openssl/openssl_1.1.0l-1~deb9u6_armhf.deb  server certificate verification failed. CAfile: /etc/ssl/certs/ca-certificates.crt CRLfile: none
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?

KenMoreira, 2022-05-22:
doing stretch dist upgrade now

MichaIng, 2022-05-22:
Did you apply the first command? It still uses the mirror director:
```sh
sed -i 's|http://raspbian.raspberrypi.org/raspbian|http://raspbian.mirror.globo.tech/raspbian|' /etc/apt/sources.list
/boot/dietpi/func/dietpi-set_software apt-cache clean
apt update
apt full-upgrade
```

KenMoreira, 2022-05-22:
> Did you apply the first command? It still uses the mirror director:
> 
> ```shell
> sed -i 's|http://raspbian.raspberrypi.org/raspbian|http://raspbian.mirror.globo.tech/raspbian|' /etc/apt/sources.list
> /boot/dietpi/func/dietpi-set_software apt-cache clean
> apt update
> apt full-upgrade
> ```

ill do that after this dist upgrade

KenMoreira, 2022-05-22:
root@KensBridge:~# apt-get dist-upgrade
Reading package lists... Done
Building dependency tree
Reading state information... Done
Calculating upgrade... Done
The following packages will be upgraded:
  bind9-host cron curl gzip libavcodec57 libavfilter6 libavformat57 libavresample3 libavutil55 libbind9-140 libcups2 libcurl3
  libcurl3-gnutls libdns-export162 libdns162 libelf1 libexpat1 libfaad2 libflac8 libfribidi0 libgmp10 libgnutls30 libgssapi-krb5-2
  libhogweed4 libicu57 libisc-export160 libisc160 libisccc140 libisccfg140 libk5crypto3 libkrb5-3 libkrb5support0 liblwres141 liblzma5
  libnettle6 libnghttp2-14 libopencv-calib3d2.4v5 libopencv-core2.4v5 libopencv-features2d2.4v5 libopencv-flann2.4v5
  libopencv-highgui2.4-deb0 libopencv-imgproc2.4v5 libopencv-objdetect2.4v5 libopencv-video2.4v5 libopenexr22 libopenjp2-7 libpcap0.8
  libpostproc54 libpython2.7 libpython2.7-minimal libpython2.7-stdlib libsamplerate0 libsasl2-2 libsasl2-modules-db libssh2-1 libssl-dev
  libssl1.0.2 libssl1.1 libswresample2 libswscale4 libtag1v5 libtag1v5-vanilla libtiff5 libvorbis0a libvorbisenc2 libvorbisfile3 libvpx4
  lighttpd mariadb-client-10.1 mariadb-client-core-10.1 mariadb-common mariadb-server mariadb-server-10.1 mariadb-server-core-10.1 openssl
  python2.7 python2.7-minimal rsync tar tzdata xz-utils zlib1g zlib1g-dev
83 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Need to get 4,127 kB/56.4 MB of archives.
After this operation, 87.0 kB of additional disk space will be used.
Do you want to continue? [Y/n] y
Err:1 https://mirrors.switch.ca/raspbian/raspbian stretch/main armhf libssl-dev armhf 1.1.0l-1~deb9u6
  server certificate verification failed. CAfile: /etc/ssl/certs/ca-certificates.crt CRLfile: none
Err:2 https://mirrors.switch.ca/raspbian/raspbian stretch/main armhf libssl1.1 armhf 1.1.0l-1~deb9u6
  server certificate verification failed. CAfile: /etc/ssl/certs/ca-certificates.crt CRLfile: none
Err:3 https://mirrors.switch.ca/raspbian/raspbian stretch/main armhf libssl1.0.2 armhf 1.0.2u-1~deb9u7
  server certificate verification failed. CAfile: /etc/ssl/certs/ca-certificates.crt CRLfile: none
Err:4 https://mirrors.switch.ca/raspbian/raspbian stretch/main armhf openssl armhf 1.1.0l-1~deb9u6
  server certificate verification failed. CAfile: /etc/ssl/certs/ca-certificates.crt CRLfile: none
E: Failed to fetch https://mirrors.switch.ca/raspbian/raspbian/pool/main/o/openssl/libssl-dev_1.1.0l-1~deb9u6_armhf.deb  server certificate verification failed. CAfile: /etc/ssl/certs/ca-certificates.crt CRLfile: none
E: Failed to fetch https://mirrors.switch.ca/raspbian/raspbian/pool/main/o/openssl/libssl1.1_1.1.0l-1~deb9u6_armhf.deb  server certificate verification failed. CAfile: /etc/ssl/certs/ca-certificates.crt CRLfile: none
E: Failed to fetch https://mirrors.switch.ca/raspbian/raspbian/pool/main/o/openssl1.0/libssl1.0.2_1.0.2u-1~deb9u7_armhf.deb  server certificate verification failed. CAfile: /etc/ssl/certs/ca-certificates.crt CRLfile: none
E: Failed to fetch https://mirrors.switch.ca/raspbian/raspbian/pool/main/o/openssl/openssl_1.1.0l-1~deb9u6_armhf.deb  server certificate verification failed. CAfile: /etc/ssl/certs/ca-certificates.crt CRLfile: none
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?

MichaIng, 2022-05-22:
You need to do this first, otherwise the command fails, as you can see.

KenMoreira, 2022-05-22:
root@KensBridge:~# sed -i 's|http://raspbian.raspberrypi.org/raspbian|http://raspbian.mirror.globo.tech/raspbian|' /etc/apt/sources.list
root@KensBridge:~# /boot/dietpi/func/dietpi-set_software apt-cache clean

 DietPi-Set_software
─────────────────────────────────────────────────────
 Mode: apt-cache (clean)

[  OK  ] apt-cache clean | Completed
root@KensBridge:~# apt update
Get:1 http://raspbian.mirror.globo.tech/raspbian stretch InRelease [15.0 kB]
Get:2 http://raspbian.mirror.globo.tech/raspbian stretch/non-free armhf Packages [98.9 kB]
Get:3 http://raspbian.mirror.globo.tech/raspbian stretch/contrib armhf Packages [56.9 kB]
Get:4 http://raspbian.mirror.globo.tech/raspbian stretch/rpi armhf Packages [1,360 B]
Get:5 https://archive.raspberrypi.org/debian stretch InRelease [25.3 kB]
Get:6 http://raspbian.mirror.globo.tech/raspbian stretch/main armhf Packages [11.7 MB]
Get:7 https://archive.raspberrypi.org/debian stretch/ui armhf Packages [44.6 kB]
Get:8 https://archive.raspberrypi.org/debian stretch/main armhf Packages [192 kB]
Fetched 12.1 MB in 9s (1,252 kB/s)
Reading package lists... Done
Building dependency tree
Reading state information... Done
83 packages can be upgraded. Run 'apt list --upgradable' to see them.

MichaIng, 2022-05-22:
And now the `apt-get dist-upgrade`.

KenMoreira, 2022-05-22:
root@KensBridge:~# apt full-upgrade
Reading package lists... Done
Building dependency tree
Reading state information... Done
Calculating upgrade... Done
The following packages will be upgraded:
  bind9-host cron curl gzip libavcodec57 libavfilter6 libavformat57 libavresample3 libavutil55 libbind9-140 libcups2 libcurl3
  libcurl3-gnutls libdns-export162 libdns162 libelf1 libexpat1 libfaad2 libflac8 libfribidi0 libgmp10 libgnutls30 libgssapi-krb5-2
  libhogweed4 libicu57 libisc-export160 libisc160 libisccc140 libisccfg140 libk5crypto3 libkrb5-3 libkrb5support0 liblwres141 liblzma5
  libnettle6 libnghttp2-14 libopencv-calib3d2.4v5 libopencv-core2.4v5 libopencv-features2d2.4v5 libopencv-flann2.4v5
  libopencv-highgui2.4-deb0 libopencv-imgproc2.4v5 libopencv-objdetect2.4v5 libopencv-video2.4v5 libopenexr22 libopenjp2-7 libpcap0.8
  libpostproc54 libpython2.7 libpython2.7-minimal libpython2.7-stdlib libsamplerate0 libsasl2-2 libsasl2-modules-db libssh2-1 libssl-dev
  libssl1.0.2 libssl1.1 libswresample2 libswscale4 libtag1v5 libtag1v5-vanilla libtiff5 libvorbis0a libvorbisenc2 libvorbisfile3 libvpx4
  lighttpd mariadb-client-10.1 mariadb-client-core-10.1 mariadb-common mariadb-server mariadb-server-10.1 mariadb-server-core-10.1 openssl
  python2.7 python2.7-minimal rsync tar tzdata xz-utils zlib1g zlib1g-dev
83 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Need to get 56.4 MB of archives.
After this operation, 84.0 kB of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:1 http://raspbian.mirror.globo.tech/raspbian stretch/main armhf gzip armhf 1.6-5+deb9u1 [103 kB]
Get:2 http://raspbian.mirror.globo.tech/raspbian stretch/main armhf tar armhf 1.29b-1.1+deb9u1 [722 kB]
Get:3 http://raspbian.mirror.globo.tech/raspbian stretch/main armhf cron armhf 3.0pl1-128+deb9u2 [92.1 kB]
Get:4 http://raspbian.mirror.globo.tech/raspbian stretch/main armhf mariadb-common all 10.1.48-0+deb9u2 [28.9 kB]
Get:5 http://raspbian.mirror.globo.tech/raspbian stretch/main armhf zlib1g-dev armhf 1:1.2.8.dfsg-5+deb9u1 [198 kB]
Get:6 http://raspbian.mirror.globo.tech/raspbian stretch/main armhf zlib1g armhf 1:1.2.8.dfsg-5+deb9u1 [82.0 kB]
Get:7 http://raspbian.mirror.globo.tech/raspbian stretch/main armhf mariadb-client-core-10.1 armhf 10.1.48-0+deb9u2 [4,370 kB]
Get:8 http://raspbian.mirror.globo.tech/raspbian stretch/main armhf mariadb-server-10.1 armhf 10.1.48-0+deb9u2 [4,693 kB]
Get:9 http://raspbian.mirror.globo.tech/raspbian stretch/main armhf mariadb-client-10.1 armhf 10.1.48-0+deb9u2 [5,322 kB]
Get:10 http://raspbian.mirror.globo.tech/raspbian stretch/main armhf mariadb-server-core-10.1 armhf 10.1.48-0+deb9u2 [4,666 kB]
Get:11 http://raspbian.mirror.globo.tech/raspbian stretch/main armhf rsync armhf 3.1.2-1+deb9u3 [355 kB]
Get:12 http://raspbian.mirror.globo.tech/raspbian stretch/main armhf libpython2.7 armhf 2.7.13-2+deb9u6 [914 kB]
Get:13 http://raspbian.mirror.globo.tech/raspbian stretch/main armhf python2.7 armhf 2.7.13-2+deb9u6 [286 kB]
Get:14 http://raspbian.mirror.globo.tech/raspbian stretch/main armhf libpython2.7-stdlib armhf 2.7.13-2+deb9u6 [1,828 kB]
Get:15 http://raspbian.mirror.globo.tech/raspbian stretch/main armhf python2.7-minimal armhf 2.7.13-2+deb9u6 [1,181 kB]
Get:16 http://raspbian.mirror.globo.tech/raspbian stretch/main armhf libpython2.7-minimal armhf 2.7.13-2+deb9u6 [390 kB]
Get:17 http://raspbian.mirror.globo.tech/raspbian stretch/main armhf libexpat1 armhf 2.2.0-2+deb9u5 [63.8 kB]
Err:18 http://raspbian.mirror.globo.tech/raspbian stretch/main armhf libssl-dev armhf 1.1.0l-1~deb9u5
  404  Not Found
Err:19 http://raspbian.mirror.globo.tech/raspbian stretch/main armhf libssl1.1 armhf 1.1.0l-1~deb9u5
  404  Not Found
Get:20 http://raspbian.mirror.globo.tech/raspbian stretch/main armhf liblzma5 armhf 5.2.2-1.2+deb9u1 [225 kB]
Get:21 http://raspbian.mirror.globo.tech/raspbian stretch/main armhf tzdata all 2021a-0+deb9u3 [281 kB]
Get:22 http://raspbian.mirror.globo.tech/raspbian stretch/main armhf xz-utils armhf 5.2.2-1.2+deb9u1 [261 kB]
Get:23 http://raspbian.mirror.globo.tech/raspbian stretch/main armhf libssl1.0.2 armhf 1.0.2u-1~deb9u7 [904 kB]
Get:24 http://raspbian.mirror.globo.tech/raspbian stretch/main armhf libgssapi-krb5-2 armhf 1.15-1+deb9u3 [132 kB]
Get:25 http://raspbian.mirror.globo.tech/raspbian stretch/main armhf libkrb5-3 armhf 1.15-1+deb9u3 [265 kB]
Get:26 http://raspbian.mirror.globo.tech/raspbian stretch/main armhf libkrb5support0 armhf 1.15-1+deb9u3 [58.4 kB]
Get:27 http://raspbian.mirror.globo.tech/raspbian stretch/main armhf libk5crypto3 armhf 1.15-1+deb9u3 [115 kB]
Get:28 http://raspbian.mirror.globo.tech/raspbian stretch/main armhf bind9-host armhf 1:9.10.3.dfsg.P4-12.3+deb9u12 [232 kB]
Get:29 http://raspbian.mirror.globo.tech/raspbian stretch/main armhf libisc160 armhf 1:9.10.3.dfsg.P4-12.3+deb9u12 [368 kB]
Get:30 http://raspbian.mirror.globo.tech/raspbian stretch/main armhf libdns162 armhf 1:9.10.3.dfsg.P4-12.3+deb9u12 [949 kB]
Get:31 http://raspbian.mirror.globo.tech/raspbian stretch/main armhf libisccc140 armhf 1:9.10.3.dfsg.P4-12.3+deb9u12 [198 kB]
Get:32 http://raspbian.mirror.globo.tech/raspbian stretch/main armhf libisccfg140 armhf 1:9.10.3.dfsg.P4-12.3+deb9u12 [217 kB]
Get:33 http://raspbian.mirror.globo.tech/raspbian stretch/main armhf libbind9-140 armhf 1:9.10.3.dfsg.P4-12.3+deb9u12 [205 kB]
Get:34 http://raspbian.mirror.globo.tech/raspbian stretch/main armhf liblwres141 armhf 1:9.10.3.dfsg.P4-12.3+deb9u12 [211 kB]
[... cut ...]
E: Failed to fetch http://raspbian.mirror.globo.tech/raspbian/pool/main/o/openssl/libssl-dev_1.1.0l-1~deb9u5_armhf.deb  404  Not Found
E: Failed to fetch http://raspbian.mirror.globo.tech/raspbian/pool/main/o/openssl/libssl1.1_1.1.0l-1~deb9u5_armhf.deb  404  Not Found
E: Failed to fetch http://raspbian.mirror.globo.tech/raspbian/pool/main/o/openssl/openssl_1.1.0l-1~deb9u5_armhf.deb  404  Not Found
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
root@KensBridge:~# ^C
root@KensBridge:~# ^C
root@KensBridge:~# apt-get dist-upgrade.
E: Invalid operation dist-upgrade.

MichaIng, 2022-05-22:
`apt-get dist-upgrade` without the `.` dot at the end.

KenMoreira, 2022-05-22:
root@KensBridge:~# apt-get dist-upgrade
Reading package lists... Done
Building dependency tree
Reading state information... Done
Calculating upgrade... Done
The following packages will be upgraded:
  bind9-host cron curl gzip libavcodec57 libavfilter6 libavformat57 libavresample3 libavutil55 libbind9-140 libcups2 libcurl3
  libcurl3-gnutls libdns-export162 libdns162 libelf1 libexpat1 libfaad2 libflac8 libfribidi0 libgmp10 libgnutls30 libgssapi-krb5-2
  libhogweed4 libicu57 libisc-export160 libisc160 libisccc140 libisccfg140 libk5crypto3 libkrb5-3 libkrb5support0 liblwres141 liblzma5
  libnettle6 libnghttp2-14 libopencv-calib3d2.4v5 libopencv-core2.4v5 libopencv-features2d2.4v5 libopencv-flann2.4v5
  libopencv-highgui2.4-deb0 libopencv-imgproc2.4v5 libopencv-objdetect2.4v5 libopencv-video2.4v5 libopenexr22 libopenjp2-7 libpcap0.8
  libpostproc54 libpython2.7 libpython2.7-minimal libpython2.7-stdlib libsamplerate0 libsasl2-2 libsasl2-modules-db libssh2-1 libssl-dev
  libssl1.0.2 libssl1.1 libswresample2 libswscale4 libtag1v5 libtag1v5-vanilla libtiff5 libvorbis0a libvorbisenc2 libvorbisfile3 libvpx4
  lighttpd mariadb-client-10.1 mariadb-client-core-10.1 mariadb-common mariadb-server mariadb-server-10.1 mariadb-server-core-10.1 openssl
  python2.7 python2.7-minimal rsync tar tzdata xz-utils zlib1g zlib1g-dev
83 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Need to get 3,224 kB/56.4 MB of archives.
After this operation, 84.0 kB of additional disk space will be used.
Do you want to continue? [Y/n] y
Err:1 http://raspbian.mirror.globo.tech/raspbian stretch/main armhf libssl-dev armhf 1.1.0l-1~deb9u5
  404  Not Found
Err:2 http://raspbian.mirror.globo.tech/raspbian stretch/main armhf libssl1.1 armhf 1.1.0l-1~deb9u5
  404  Not Found
Err:3 http://raspbian.mirror.globo.tech/raspbian stretch/main armhf openssl armhf 1.1.0l-1~deb9u5
  404  Not Found
E: Failed to fetch http://raspbian.mirror.globo.tech/raspbian/pool/main/o/openssl/libssl-dev_1.1.0l-1~deb9u5_armhf.deb  404  Not Found
E: Failed to fetch http://raspbian.mirror.globo.tech/raspbian/pool/main/o/openssl/libssl1.1_1.1.0l-1~deb9u5_armhf.deb  404  Not Found
E: Failed to fetch http://raspbian.mirror.globo.tech/raspbian/pool/main/o/openssl/openssl_1.1.0l-1~deb9u5_armhf.deb  404  Not Found
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?

KenMoreira, 2022-05-22:
running 
:~# sudo apt-get update --fix-missing

KenMoreira, 2022-05-22:
then ill do sudo apt-get install -f

KenMoreira, 2022-05-22:
root@KensBridge:~# sudo apt-get update --fix-missing
Hit:1 http://raspbian.mirror.globo.tech/raspbian stretch InRelease
Hit:2 https://archive.raspberrypi.org/debian stretch InRelease
^Cading package lists... 41%
root@KensBridge:~# ^C
root@KensBridge:~# ^C
root@KensBridge:~# ^C
root@KensBridge:~# sudo apt-get update --fix-missing
Hit:1 http://raspbian.mirror.globo.tech/raspbian stretch InRelease
Hit:2 https://archive.raspberrypi.org/debian stretch InRelease
Reading package lists... Done
root@KensBridge:~# sudo apt-get update --fix-missing
Hit:1 http://raspbian.mirror.globo.tech/raspbian stretch InRelease
Hit:2 https://archive.raspberrypi.org/debian stretch InRelease
Reading package lists... Done
root@KensBridge:~# $ sudo apt-get install -f
-bash: $: command not found
root@KensBridge:~# ^C
root@KensBridge:~# sudo apt-get install -f
Reading package lists... Done
Building dependency tree
Reading state information... Done
0 upgraded, 0 newly installed, 0 to remove and 83 not upgraded.

MichaIng, 2022-05-22:
```
  404  Not Found
```
The exact packages which solve the HTTPS certificate error cannot be found... Let me have a look into this repo: http://raspbian.mirror.globo.tech/raspbian/pool/main/o/openssl/
Indeed `1.1.0l-1~deb9u5` is not compiled, even the sources are completely missing for any Stretch package version above `[1.1.0l-1~deb9u1](http://raspbian.mirror.globo.tech/raspbian/pool/main/o/openssl/openssl_1.1.0l-1~deb9u1.dsc)`, so the link in the package list is invalid. Comparing with the mirror director: http://raspbian.raspberrypi.org/raspbian/pool/main/o/openssl/
All packages are available and compiled there... So we need to find another mirror, `http://raspbian.mirror.globo.tech` is faulty...

Another one in Canada: https://raspbian.freemirror.org/raspbian/pool/main/o/openssl/
- `1.1.0l-1~deb9u6` packages available (one build newer)
- Checking back whether these are correctly linked in the package list: https://raspbian.freemirror.org/raspbian/dists/stretch/main/binary-armhf/
- Yes that should work now.

Please run:
```sh
sed -i 's|http://raspbian.mirror.globo.tech/raspbian|http://raspbian.mirror.globo.tech/raspbian|' /etc/apt/sources.list
/boot/dietpi/func/dietpi-set_software apt-cache clean
apt update
apt-get dist-upgrade
```

KenMoreira, 2022-05-22:
funny thing is , all this time my end goal was to get DSF/DSD to play over GMRender cause its not. so i decided to start updating stuff.  And here I am still struggling to do even that. LOL

KenMoreira, 2022-05-22:
root@KensBridge:~# sed -i 's|http://raspbian.mirror.globo.tech/raspbian|http://r                                                  aspbian.mirror.globo.tech/raspbian|' /etc/apt/sources.list
root@KensBridge:~# /boot/dietpi/func/dietpi-set_software apt-cache clean

 DietPi-Set_software
─────────────────────────────────────────────────────
 Mode: apt-cache (clean)

[  OK  ] apt-cache clean | Completed
root@KensBridge:~# apt update
Get:1 http://raspbian.mirror.globo.tech/raspbian stretch InRelease [15.0 kB]
Get:2 http://raspbian.mirror.globo.tech/raspbian stretch/non-free armhf Packages                                                   [98.9 kB]
Get:3 https://archive.raspberrypi.org/debian stretch InRelease [25.3 kB]
Get:4 http://raspbian.mirror.globo.tech/raspbian stretch/rpi armhf Packages [1,3                                                  60 B]
Get:5 http://raspbian.mirror.globo.tech/raspbian stretch/main armhf Packages [11                                                  .7 MB]
Get:6 https://archive.raspberrypi.org/debian stretch/main armhf Packages [192 kB                                                  ]
Get:7 https://archive.raspberrypi.org/debian stretch/ui armhf Packages [44.6 kB]
Get:8 http://raspbian.mirror.globo.tech/raspbian stretch/contrib armhf Packages                                                   [56.9 kB]
Fetched 12.1 MB in 9s (1,252 kB/s)
Reading package lists... Done
Building dependency tree
Reading state information... Done
83 packages can be upgraded. Run 'apt list --upgradable' to see them.
root@KensBridge:~# apt-get dist-upgrade
Reading package lists... Done
Building dependency tree
Reading state information... Done
Calculating upgrade... Done
The following packages will be upgraded:
  bind9-host cron curl gzip libavcodec57 libavfilter6 libavformat57 libavresample3 libavutil55 libbind9-140 libcups2 libcurl3
  libcurl3-gnutls libdns-export162 libdns162 libelf1 libexpat1 libfaad2 libflac8 libfribidi0 libgmp10 libgnutls30
  libgssapi-krb5-2 libhogweed4 libicu57 libisc-export160 libisc160 libisccc140 libisccfg140 libk5crypto3 libkrb5-3
  libkrb5support0 liblwres141 liblzma5 libnettle6 libnghttp2-14 libopencv-calib3d2.4v5 libopencv-core2.4v5
  libopencv-features2d2.4v5 libopencv-flann2.4v5 libopencv-highgui2.4-deb0 libopencv-imgproc2.4v5 libopencv-objdetect2.4v5
  libopencv-video2.4v5 libopenexr22 libopenjp2-7 libpcap0.8 libpostproc54 libpython2.7 libpython2.7-minimal libpython2.7-stdlib
  libsamplerate0 libsasl2-2 libsasl2-modules-db libssh2-1 libssl-dev libssl1.0.2 libssl1.1 libswresample2 libswscale4 libtag1v5
  libtag1v5-vanilla libtiff5 libvorbis0a libvorbisenc2 libvorbisfile3 libvpx4 lighttpd mariadb-client-10.1
  mariadb-client-core-10.1 mariadb-common mariadb-server mariadb-server-10.1 mariadb-server-core-10.1 openssl python2.7
  python2.7-minimal rsync tar tzdata xz-utils zlib1g zlib1g-dev
83 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Need to get 56.4 MB of archives.
After this operation, 84.0 kB of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:1 http://raspbian.mirror.globo.tech/raspbian stretch/main armhf gzip armhf 1.6-5+deb9u1 [103 kB]
Get:2 http://raspbian.mirror.globo.tech/raspbian stretch/main armhf tar armhf 1.29b-1.1+deb9u1 [722 kB]
Get:3 http://raspbian.mirror.globo.tech/raspbian stretch/main armhf cron armhf 3.0pl1-128+deb9u2 [92.1 kB]
Get:4 http://raspbian.mirror.globo.tech/raspbian stretch/main armhf mariadb-common all 10.1.48-0+deb9u2 [28.9 kB]
Get:5 http://raspbian.mirror.globo.tech/raspbian stretch/main armhf zlib1g-dev armhf 1:1.2.8.dfsg-5+deb9u1 [198 kB]
Get:6 http://raspbian.mirror.globo.tech/raspbian stretch/main armhf zlib1g armhf 1:1.2.8.dfsg-5+deb9u1 [82.0 kB]
Get:7 http://raspbian.mirror.globo.tech/raspbian stretch/main armhf mariadb-client-core-10.1 armhf 10.1.48-0+deb9u2 [4,370 kB]
Get:8 http://raspbian.mirror.globo.tech/raspbian stretch/main armhf mariadb-server-10.1 armhf 10.1.48-0+deb9u2 [4,693 kB]
Get:9 http://raspbian.mirror.globo.tech/raspbian stretch/main armhf mariadb-client-10.1 armhf 10.1.48-0+deb9u2 [5,322 kB]
Get:10 http://raspbian.mirror.globo.tech/raspbian stretch/main armhf mariadb-server-core-10.1 armhf 10.1.48-0+deb9u2 [4,666 kB]
Get:11 http://raspbian.mirror.globo.tech/raspbian stretch/main armhf rsync armhf 3.1.2-1+deb9u3 [355 kB]
Get:12 http://raspbian.mirror.globo.tech/raspbian stretch/main armhf libpython2.7 armhf 2.7.13-2+deb9u6 [914 kB]
Get:13 http://raspbian.mirror.globo.tech/raspbian stretch/main armhf python2.7 armhf 2.7.13-2+deb9u6 [286 kB]
Get:14 http://raspbian.mirror.globo.tech/raspbian stretch/main armhf libpython2.7-stdlib armhf 2.7.13-2+deb9u6 [1,828 kB]
Get:15 http://raspbian.mirror.globo.tech/raspbian stretch/main armhf python2.7-minimal armhf 2.7.13-2+deb9u6 [1,181 kB]
Get:16 http://raspbian.mirror.globo.tech/raspbian stretch/main armhf libpython2.7-minimal armhf 2.7.13-2+deb9u6 [390 kB]
Get:17 http://raspbian.mirror.globo.tech/raspbian stretch/main armhf libexpat1 armhf 2.2.0-2+deb9u5 [63.8 kB]
Err:18 http://raspbian.mirror.globo.tech/raspbian stretch/main armhf libssl-dev armhf 1.1.0l-1~deb9u5
  404  Not Found
Err:19 http://raspbian.mirror.globo.tech/raspbian stretch/main armhf libssl1.1 armhf 1.1.0l-1~deb9u5
  404  Not Found
Get:20 http://raspbian.mirror.globo.tech/raspbian stretch/main armhf liblzma5 armhf 5.2.2-1.2+deb9u1 [225 kB]
Get:21 http://raspbian.mirror.globo.tech/raspbian stretch/main armhf tzdata all 2021a-0+deb9u3 [281 kB]
[... cut ...]
Fetched 53.2 MB in 9s (5,678 kB/s)
E: Failed to fetch http://raspbian.mirror.globo.tech/raspbian/pool/main/o/openssl/libssl-dev_1.1.0l-1~deb9u5_armhf.deb  404  Not Found
E: Failed to fetch http://raspbian.mirror.globo.tech/raspbian/pool/main/o/openssl/libssl1.1_1.1.0l-1~deb9u5_armhf.deb  404  Not Found
E: Failed to fetch http://raspbian.mirror.globo.tech/raspbian/pool/main/o/openssl/openssl_1.1.0l-1~deb9u5_armhf.deb  404  Not Found
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?

[60 more replies]

closed 2022-05-27 by MichaIng