MichaIng/DietPi #5503: Explaination of images

actraiser30, 2022-05-20:
Hello i have a Raspberry pi 3 and 4 i see in the download section many different dietpi for RaspberryPI
can you explain me what are the differences of Armv6 and Armv7 images i understand is that the arm7 will work from Rpi 2 and up
but the dietpi was always Armv6 and was working in all Raspberry pi models.

Also will the armv6 image work in all the Raspberry`s Pi`s including the RPi  4 ??

Thank you

Joulinar, 2022-05-20:
Basically it is already explained on the download page. Main difference between ARMv6 and ARMv7 is the used package repository. As stated on the download page as well, the ARMv6 version will be working on **ALL** models.

- ARMv6 32-bit image: Uses the Raspbian package repository and works on ALL Raspberry Pi models.
- ARMv7 32-bit image: Uses the Debian package repository and does NOT work on Raspberry Pi 1 and Zero (1) models.
- ARMv8 64-bit image: Uses the Debian package repository and does NOT work on Raspberry Pi 1, Zero (1) and Raspberry Pi 2 v1.1 models.

actraiser30, 2022-05-20:
thanks buddy

actraiser30, 2022-05-20:
is it possible to put also in the downloads the last buster images for dietpi and amibbery ??

Joulinar, 2022-05-20:
we don't offer Buster images anymore as Debian Buster is `oldstable`. Current Debian release is Bullseye. For Amiberry we offer ARMv6 as well as ARMv7 images directly on our download page.

MichaIng, 2022-05-20:
As additional background: ARMv7/ARMv8 use the regular Debian APT package repository, `armhf` and `aarch64` architectures: http://deb.debian.org/

Since these `armhf` packages do not support ARMv6 CPUs, but only ARMv7 and up, a "fork" of the Debian repository was required to get it running on the first Raspberry Pi models, which is the reason why "Raspbian" exists: It uses ~99,9% of the Debian package sources, with minor changes where required as of the architecture, but then provides the packages compiled against ARMv6 CPUs, i.e. without ARMv7-specific features like NEON.

> is it possible to put also in the downloads the last buster images for dietpi and amibbery ??

We do not provide Buster images anymore, support will be dropped next summer, so we would shoot into our own knees when still providing "oldstable" Debian images. Are there any issues with the Bullseye Amiberry image?

actraiser30, 2022-05-20:
buster somehow feels faster bullseye feels heavier as an image,is the last released buster image dated: 6/2/2021 ??

MichaIng, 2022-05-20:
> buster somehow feels faster bullseye feels heavier as an image

Not sure which aspect you mean by "heavier"? The Bullseye image is slightly larger, but shouldn't be somehow generally slower. If a specific application is slower, it would make sense to report this at respective maintainers or developers.

> is the last released buster image dated: 6/2/2021

I have no idea when the last image was generated. It however doesn't really matter since you can update everything via `dietpi-update` and `apt upgrade` to latest (on Buster available) versions.

MichaIng, 2022-07-06:
I'll mark this as closed. Feel free if there are still open questions.

closed 2022-07-06 by MichaIng