MichaIng/DietPi #5504: Failed to apply overlay '0_w1-gpio' (kernel)

ScottWoolven5, 2022-05-20:
I'm trying to enable the 1-wire protocol to measure temperature using a DS18B20 sensor. I was previously using Raspberry Pi OS however I've decided to switch to DietPi in order to reduce overheads and was able to read the temperature from the sensor. 

I've added `dtoverlay=w1-gpio,gpiopin=19` to /boot/config.txt but I there is no device folder in /sys/bus/w1/devices/. When I enter `sudo dtoverlay w1-gpio gpiopin=19` into the terminal, I get the following:

`* Failed to apply overlay '0_w1-gpio' (kernel)`

When running `dmesg`, I see that the driver for 1-wire Dallas network protocol is found. 

I'm using the latest ARMv8 Bullseye version of DietPi, testing on both RPi 3B+ & RPi 4.

MichaIng, 2022-05-21:
Did you reboot after applying the dtoverlay to `config.txt`? Most overlays cannot be applied at runtime via `dtoverlay` command.

MichaIng, 2022-07-23:
Marking as closed. Feel free to reopen if required.

closed 2022-07-23 by MichaIng