MichaIng/DietPi #5513: Err:9 https://download.mono-project.com/repo/debian buster Release

tanandry, 2022-05-23:
### Creating a bug report/issue

#### Required Information
- DietPi version | `8.4.2`
- Distro version | `buster 0`
- Kernel version | `Linux DietPi-RASPI 5.10.103-v8+`
- SBC model | `RPi 4 Model B (aarch64)`

#### Steps to reproduce

1. Do dietpi-update or apt-get update

#### Expected behaviour

- Update

#### Actual behaviour

- Failed to update

#### Extra details

Anything that I can do to solve this issue? I can't update my system.

[image]

Joulinar, 2022-05-23:
Could you try again. Probably it was a temp issue with the 3rd party repository for Mono. Btw there is no need to do screen prints. You should be able to copy everything from SSH terminal directly. This simplifies thinks for us.

tanandry, 2022-05-24:
I tried to do `sudo apt-get update --fix-missing` but still not possible.
```
Hit:1 https://deb.debian.org/debian buster InRelease                                                  
Hit:2 https://deb.debian.org/debian buster-updates InRelease                                          
Hit:3 https://deb.debian.org/debian-security buster/updates InRelease                                 
Hit:4 https://deb.debian.org/debian buster-backports InRelease                                        
Hit:5 https://apt.sonarr.tv/debian buster InRelease                                                   
Hit:6 https://archive.raspberrypi.org/debian buster InRelease                                         
Hit:7 https://repo.jellyfin.org/debian buster InRelease                                               
Ign:8 https://download.mono-project.com/repo/debian buster InRelease              
Get:9 https://download.mono-project.com/repo/debian buster Release [5337 B]
Get:10 https://download.mono-project.com/repo/debian buster Release.gpg [488 B]
Fetched 5825 B in 47s (124 B/s)
Reading package lists... Done
```

Joulinar, 2022-05-24:
Can you share following

```
apt update
```

tanandry, 2022-05-24:
```
Hit:1 https://download.mono-project.com/repo/debian buster InRelease
Hit:2 https://deb.debian.org/debian buster InRelease                                                  
Hit:3 https://deb.debian.org/debian buster-updates InRelease                                          
Hit:4 https://deb.debian.org/debian-security buster/updates InRelease                                 
Hit:5 https://deb.debian.org/debian buster-backports InRelease                                        
Hit:6 https://archive.raspberrypi.org/debian buster InRelease                                         
Hit:7 https://repo.jellyfin.org/debian buster InRelease                                    
Hit:8 https://apt.sonarr.tv/debian buster InRelease
Reading package lists... Done
Building dependency tree       
Reading state information... Done
168 packages can be upgraded. Run 'apt list --upgradable' to see them.
```

Joulinar, 2022-05-24:
That looks all fine, continue with package update

```
apt upgrade
```

tanandry, 2022-05-24:
```
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Calculating upgrade... Done
The following packages have been kept back:
  libreoffice-common
The following packages will be upgraded:
  ca-certificates-mono libldap-2.4-2 libldap-common libmono-2.0-1 libmono-2.0-dev
  libmono-accessibility4.0-cil libmono-cairo4.0-cil libmono-cecil-private-cil libmono-cil-dev
  libmono-codecontracts4.0-cil libmono-compilerservices-symbolwriter4.0-cil libmono-corlib4.5-cil
  libmono-cscompmgd0.0-cil libmono-csharp4.0c-cil libmono-custommarshalers4.0-cil
  libmono-data-tds4.0-cil libmono-db2-1.0-cil libmono-debugger-soft4.0a-cil libmono-http4.0-cil
  libmono-i18n-cjk4.0-cil libmono-i18n-mideast4.0-cil libmono-i18n-other4.0-cil
  libmono-i18n-rare4.0-cil libmono-i18n-west4.0-cil libmono-i18n4.0-all libmono-i18n4.0-cil
  libmono-ldap4.0-cil libmono-management4.0-cil libmono-messaging-rabbitmq4.0-cil
  libmono-messaging4.0-cil libmono-microsoft-build-engine4.0-cil
  libmono-microsoft-build-framework4.0-cil libmono-microsoft-build-tasks-v4.0-4.0-cil
  libmono-microsoft-build-utilities-v4.0-4.0-cil libmono-microsoft-build4.0-cil
  libmono-microsoft-csharp4.0-cil libmono-microsoft-visualc10.0-cil
  libmono-microsoft-web-infrastructure1.0-cil libmono-oracle4.0-cil libmono-parallel4.0-cil
  libmono-peapi4.0a-cil libmono-posix4.0-cil libmono-profiler libmono-rabbitmq4.0-cil
  libmono-relaxng4.0-cil libmono-security4.0-cil libmono-sharpzip4.84-cil libmono-simd4.0-cil
  libmono-smdiagnostics0.0-cil libmono-sqlite4.0-cil libmono-system-componentmodel-composition4.0-cil
  libmono-system-componentmodel-dataannotations4.0-cil libmono-system-configuration-install4.0-cil
  libmono-system-configuration4.0-cil libmono-system-core4.0-cil
  libmono-system-data-datasetextensions4.0-cil libmono-system-data-entity4.0-cil
  libmono-system-data-linq4.0-cil libmono-system-data-services-client4.0-cil
  libmono-system-data-services4.0-cil libmono-system-data4.0-cil libmono-system-deployment4.0-cil
  libmono-system-design4.0-cil libmono-system-drawing-design4.0-cil libmono-system-drawing4.0-cil
  libmono-system-dynamic4.0-cil libmono-system-enterpriseservices4.0-cil
  libmono-system-identitymodel-selectors4.0-cil libmono-system-identitymodel4.0-cil
  libmono-system-io-compression-filesystem4.0-cil libmono-system-io-compression4.0-cil
  libmono-system-json-microsoft4.0-cil libmono-system-json4.0-cil
  libmono-system-ldap-protocols4.0-cil libmono-system-ldap4.0-cil libmono-system-management4.0-cil
  libmono-system-messaging4.0-cil libmono-system-net-http-formatting4.0-cil
  libmono-system-net-http-webrequest4.0-cil libmono-system-net-http4.0-cil libmono-system-net4.0-cil
  libmono-system-numerics-vectors4.0-cil libmono-system-numerics4.0-cil
  libmono-system-reactive-core2.2-cil libmono-system-reactive-debugger2.2-cil
  libmono-system-reactive-experimental2.2-cil libmono-system-reactive-interfaces2.2-cil
  libmono-system-reactive-linq2.2-cil libmono-system-reactive-observable-aliases0.0-cil
  libmono-system-reactive-platformservices2.2-cil libmono-system-reactive-providers2.2-cil
  libmono-system-reactive-runtime-remoting2.2-cil libmono-system-reactive-windows-forms2.2-cil
  libmono-system-reactive-windows-threading2.2-cil libmono-system-reflection-context4.0-cil
  libmono-system-runtime-caching4.0-cil libmono-system-runtime-durableinstancing4.0-cil
  libmono-system-runtime-serialization-formatters-soap4.0-cil
  libmono-system-runtime-serialization4.0-cil libmono-system-runtime4.0-cil
  libmono-system-security4.0-cil libmono-system-servicemodel-activation4.0-cil
  libmono-system-servicemodel-discovery4.0-cil libmono-system-servicemodel-internals0.0-cil
  libmono-system-servicemodel-routing4.0-cil libmono-system-servicemodel-web4.0-cil
  libmono-system-servicemodel4.0a-cil libmono-system-serviceprocess4.0-cil
  libmono-system-threading-tasks-dataflow4.0-cil libmono-system-transactions4.0-cil
  libmono-system-web-abstractions4.0-cil libmono-system-web-applicationservices4.0-cil
  libmono-system-web-dynamicdata4.0-cil libmono-system-web-extensions-design4.0-cil
  libmono-system-web-extensions4.0-cil libmono-system-web-http-selfhost4.0-cil
  libmono-system-web-http-webhost4.0-cil libmono-system-web-http4.0-cil
  libmono-system-web-mobile4.0-cil libmono-system-web-mvc3.0-cil libmono-system-web-razor2.0-cil
  libmono-system-web-regularexpressions4.0-cil libmono-system-web-routing4.0-cil
  libmono-system-web-services4.0-cil libmono-system-web-webpages-deployment2.0-cil
  libmono-system-web-webpages-razor2.0-cil libmono-system-web-webpages2.0-cil
  libmono-system-web4.0-cil libmono-system-windows-forms-datavisualization4.0a-cil
  libmono-system-windows-forms4.0-cil libmono-system-windows4.0-cil
  libmono-system-workflow-activities4.0-cil libmono-system-workflow-componentmodel4.0-cil
  libmono-system-workflow-runtime4.0-cil libmono-system-xaml4.0-cil libmono-system-xml-linq4.0-cil
  libmono-system-xml-serialization4.0-cil libmono-system-xml4.0-cil libmono-system4.0-cil
  libmono-tasklets4.0-cil libmono-webbrowser4.0-cil libmono-webmatrix-data4.0-cil
  libmono-windowsbase4.0-cil libmono-xbuild-tasks4.0-cil libmonosgen-2.0-1 libmonosgen-2.0-dev
  libssl-dev libssl1.1 libxml2 mono-4.0-gac mono-4.0-service mono-complete mono-csharp-shell
  mono-devel mono-gac mono-mcs mono-roslyn mono-runtime mono-runtime-common mono-runtime-sgen
[... cut ...]
E: Failed to fetch https://download.mono-project.com/repo/debian/pool/main/m/msbuild/msbuild_16.10.1+xamarinxplat.2021.05.26.14.00-0xamarin2+debian10b1_all.deb  Unable to connect to download.mono-project.com:https:
E: Failed to fetch https://download.mono-project.com/repo/debian/pool/main/m/mono/ca-certificates-mono_6.12.0.179-0xamarin3+debian10b1_all.deb  Unable to connect to download.mono-project.com:https:
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
```

Joulinar, 2022-05-24:
That's weird. You where able to connect to `https://download.mono-project.com` while running the update process but your connection times out on the upgrade command some minutes later. Do you have some special network components like firewall or other security stuff that could block such a connection?

MichaIng, 2022-05-24:
I'm also able to download those packages 🤔.

tanandry, 2022-05-24:
I have routed my router connection through a wireguard server, updated again, and it downloaded perfectly fine. So perhaps my isp is doing some weird things? Not sure.

Joulinar, 2022-05-24:
I guess we will not find out 🤔 

I'm going to close this issue, feel free to reopen if needed.

closed 2022-05-24 by Joulinar