MichaIng/DietPi #5515: error: could not compile vaultwarden when upgrading to version 1.25.0

anubis-genix, 2022-05-26:
Tried to update to the latest vaultwarden version 1.25.0 from version 1.24.0. 
This is the output I get after it fails.

NanoPi`` K1 Plus (aarch64)
│  - Command: cargo build --features sqlite --release
│  - Exit code: 101
│  - DietPi version: v8.4.2 (MichaIng/master) | HW_MODEL: 67 | HW_ARCH: 3 | DISTRO: 5
│  - Image creator: DietPi Core Team
│  - Pre-image: Armbian
│    Compiling yubico v0.11.0
│ error: could not compile `vaultwarden`
│
│ Caused by:
│   process didn't exit successfully: `rustc --crate-name vaultwarden --edition=2021 src/main.rs --error-format=json
│ --json=diagnostic-rendered-ansi,artifacts,future-incompat --crate-type bin --emit=dep-info,link -C opt-level=3 -C
│ lto=thin --cfg 'feature="libsqlite3-sys"' --cfg 'feature="sqlite"' -C metadata=a83be2069b6a9a76 -C
│ extra-filename=-a83be2069b6a9a76 --out-dir /opt/vaultwarden/target/release/deps -C strip=debuginfo -L
│ dependency=/opt/vaultwarden/target/release/deps --extern
│ backtrace=/opt/vaultwarden/target/release/deps/libbacktrace-8bb6fc68e4a228c4.rlib --extern
│ bytes=/opt/vaultwarden/target/release/deps/libbytes-478b01a89291ace7.rlib --extern
│ cached=/opt/vaultwarden/target/release/deps/libcached-74afb4fec5cf8bff.rlib --extern
│ chashmap=/opt/vaultwarden/target/release/deps/libchashmap-41a966aeb2079801.rlib --extern
│ chrono=/opt/vaultwarden/target/release/deps/libchrono-f906c7b429ef21d1.rlib --extern
│ chrono_tz=/opt/vaultwarden/target/release/deps/libchrono_tz-e51beec82635ba89.rlib --extern
│ cookie=/opt/vaultwarden/target/release/deps/libcookie-db15f2d62e25cc7b.rlib --extern
│ cookie_store=/opt/vaultwarden/target/release/deps/libcookie_store-8e62dd1e342778ec.rlib --extern
│ ctrlc=/opt/vaultwarden/target/release/deps/libctrlc-ed7507bb65349fad.rlib --extern
│ data_encoding=/opt/vaultwarden/target/release/deps/libdata_encoding-92d07f4aa32f63fa.rlib --extern
│ data_url=/opt/vaultwarden/target/release/deps/libdata_url-dec429fca4dad6b3.rlib --extern
│ diesel=/opt/vaultwarden/target/release/deps/libdiesel-31d7c72973f2d6b4.rlib --extern
│ diesel_migrations=/opt/vaultwarden/target/release/deps/libdiesel_migrations-d7e2074742e476a7.rlib --extern
│ dotenvy=/opt/vaultwarden/target/release/deps/libdotenvy-3d6d0a3db794d756.rlib --extern
│ fern=/opt/vaultwarden/target/release/deps/libfern-4cc021884f39495b.rlib --extern
│ futures=/opt/vaultwarden/target/release/deps/libfutures-384b26d0435018d7.rlib --extern
│ governor=/opt/vaultwarden/target/release/deps/libgovernor-409f50b1e04a3bb9.rlib --extern
│ handlebars=/opt/vaultwarden/target/release/deps/libhandlebars-38fc3cfbeab72305.rlib --extern
│ html5gum=/opt/vaultwarden/target/release/deps/libhtml5gum-ed05d7c00ce31a96.rlib --extern
│ idna=/opt/vaultwarden/target/release/deps/libidna-fcd039b6e5e635b4.rlib --extern
│ job_scheduler=/opt/vaultwarden/target/release/deps/libjob_scheduler-4eea64efa53e95ca.rlib --extern
│ jsonwebtoken=/opt/vaultwarden/target/release/deps/libjsonwebtoken-637a354696b69dbc.rlib --extern
│ lettre=/opt/vaultwarden/target/release/deps/liblettre-6e78b97a391eee9c.rlib --extern
│ libsqlite3_sys=/opt/vaultwarden/target/release/deps/liblibsqlite3_sys-0152b5f1b60f43dc.rlib --extern
│ log=/opt/vaultwarden/target/release/deps/liblog-5ecf186a9c1eeffd.rlib --extern
│ num_derive=/opt/vaultwarden/target/release/deps/libnum_derive-dfcf91356734d9a0.so --extern
│ num_traits=/opt/vaultwarden/target/release/deps/libnum_traits-a4e07e8c1577493e.rlib --extern
│ once_cell=/opt/vaultwarden/target/release/deps/libonce_cell-f718696775d31ca4.rlib --extern
│ openssl=/opt/vaultwarden/target/release/deps/libopenssl-2f0de8783b34a34d.rlib --extern
│ ws=/opt/vaultwarden/target/release/deps/libparity_ws-cc0dccc042441c00.rlib --extern
│ paste=/opt/vaultwarden/target/release/deps/libpaste-78a55d2c1e7869b1.so --extern
│ percent_encoding=/opt/vaultwarden/target/release/deps/libpercent_encoding-405bb8c0f9c06788.rlib --extern
│ pico_args=/opt/vaultwarden/target/release/deps/libpico_args-500f27ca02f5ceff.rlib --extern
│ percent_encoding=/opt/vaultwarden/target/release/deps/libpercent_encoding-405bb8c0f9c06788.rlib --extern
│ pico_args=/opt/vaultwarden/target/release/deps/libpico_args-500f27ca02f5ceff.rlib --extern
│ rand=/opt/vaultwarden/target/release/deps/librand-f5abd520473e7064.rlib --extern
│ regex=/opt/vaultwarden/target/release/deps/libregex-5bfba6bb4a5bcebf.rlib --extern
│ reqwest=/opt/vaultwarden/target/release/deps/libreqwest-e57d169f4e894a9d.rlib --extern
│ ring=/opt/vaultwarden/target/release/deps/libring-56c6e8d9fce7baf4.rlib --extern
│ rmpv=/opt/vaultwarden/target/release/deps/librmpv-ec1a546a1ede950f.rlib --extern
│ rocket=/opt/vaultwarden/target/release/deps/librocket-05cb9c0d5f8a765e.rlib --extern
│ serde=/opt/vaultwarden/target/release/deps/libserde-bdd97c66da62a898.rlib --extern
│ serde_json=/opt/vaultwarden/target/release/deps/libserde_json-249816585476010f.rlib --extern
│ syslog=/opt/vaultwarden/target/release/deps/libsyslog-db8f95f44d259d49.rlib --extern
│ time=/opt/vaultwarden/target/release/deps/libtime-7070b35a3e8c7a36.rlib --extern
│ tokio=/opt/vaultwarden/target/release/deps/libtokio-04533621b22d98a4.rlib --extern
│ totp_lite=/opt/vaultwarden/target/release/deps/libtotp_lite-d39873b930371ce0.rlib --extern
│ tracing=/opt/vaultwarden/target/release/deps/libtracing-2e31b13690d8a076.rlib --extern
│ url=/opt/vaultwarden/target/release/deps/liburl-890d913f7834966e.rlib --extern
│ uuid=/opt/vaultwarden/target/release/deps/libuuid-a478bb99599ff3c7.rlib --extern
│ webauthn_rs=/opt/vaultwarden/target/release/deps/libwebauthn_rs-edc2a41d171db62e.rlib --extern
│ yubico=/opt/vaultwarden/target/release/deps/libyubico-f7ad989c38497fa7.rlib --cfg sqlite -L
│ native=/opt/vaultwarden/target/release/build/libsqlite3-sys-e52fafab49b35746/out -L
│ native=/opt/vaultwarden/target/release/build/ring-ee21f17f297afff6/out` (signal: 9, SIGKILL: `kill)

Joulinar, 2022-05-26:
Hi,

looks like something killed the the process `(signal: 9, SIGKILL: kill)`

Could you have a look to the `journalctl` to check if there was some memory issues?

anubis-genix, 2022-05-26:
Restoring from backup right now. I'll try to reproduce this error on my second device and give proper feedback tomorrow or the day ofter tomorrow.

anubis-genix, 2022-05-26:
Performed another upgrade after restoring from backup on my main device. This time without errors. I don't get it to be honest. 2048MB swap was left untouched. I think this issue can be closed.

closed 2022-05-26 by anubis-genix