MichaIng/DietPi #5516: Gitea service fails to start after update

twikedk, 2022-05-26:
### Creating a bug report/issue

#### Required Information
- DietPi version | 
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=4
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
- Distro version | bullseye
- Kernel version | Linux development 5.15.25-sunxi64 #22.02.1 SMP Sun Feb 27 09:24:04 UTC 2022 aarch64 GNU/Linux
- SBC model | PINE A64 (aarch64)
- Power supply used | 5V 1A
- SD card used | SanDisk

#### Additional Information (if applicable)
- Software title | Gitea
- Was the software title installed freshly or updated/migrated? Updated
- Can this issue be replicated on a fresh installation of DietPi? Haven't tried

#### Steps to reproduce

1. dietpi-software reinstall 165
2. restart sbc

#### Expected behaviour

- Gitea service running and able to login via web
- 
#### Actual behaviour

- Gitea service failed to start

#### Extra details

From systemd status log:

● gitea.service - Gitea (DietPi)
Loaded: loaded (/etc/systemd/system/gitea.service; disabled; vendor preset: enabled)
Active: failed (Result: exit-code) since Thu 2022-05-26 12:12:59 CEST; 3min 11s ago
Process: 1143 ExecStart=/mnt/dietpi_userdata/gitea/gitea web (code=exited, status=1/FAILURE)
Main PID: 1143 (code=exited, status=1/FAILURE)
CPU: 1.094s

May 26 12:12:58 development systemd[1]: Started Gitea (DietPi). 
May 26 12:12:59 development gitea[1143]: 2022/05/26 12:12:59 cmd/web.go:102:runWeb() [I] Starting Gitea on PID: 1143
May 26 12:12:59 development gitea[1143]: 2022/05/26 12:12:59 ...s/setting/setting.go:957:loadFromConf() [F] Expect
user 'dietpi' but current user is: gitea
May 26 12:12:59 development systemd[1]: gitea.service: Main process exited, code=exited, status=1/FAILURE
May 26 12:12:59 development systemd[1]: gitea.service: Failed with result 'exit-code'.
May 26 12:12:59 development systemd[1]: gitea.service: Consumed 1.094s CPU time.

MichaIng, 2022-05-26:
Many thanks for your report.

This is much likely since we switched the service user from "dietpi" to "gitea", but it seems the old user is stored in some config. I'll check which one it is and add the migration to dietpi-software.

MichaIng, 2022-05-26:
Found it, to fix this, please run:
```sh
G_CONFIG_INJECT 'RUN_USER[[:blank:]]' 'RUN_USER = gitea' /mnt/dietpi_userdata/gitea/custom/conf/app.ini
```

twikedk, 2022-05-26:
I did find it in the app.ini but would not change it until I had reported the issue.
Used your fix and everything is running fine now. :-)

closed 2022-05-26 by twikedk

MichaIng, 2022-05-26:
Fixed with: https://github.com/MichaIng/DietPi/commit/2a1db41

anselal, 2023-03-06:
I am getting the following error and can't start the service

[image]

Joulinar, 2023-03-06:
Pls open a new issue and fill the give questionnaire.