MichaIng/DietPi #5527: qbittorrent webgui not running whereas qbittorrent-nox is running

dasrocky, 2022-05-29:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
```
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=4
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
```
- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
`bullseye 0`
- Kernel version | `uname -a`
`Linux DietPi 5.15.32-v8+ #1538 SMP PREEMPT Thu Mar 31 19:40:39 BST 2022 aarch64 GNU/Linux`
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
`RPi 4 Model B (aarch64)`
- SD card: Sandisk 32GB
- RAM: 4GB

#### Additional Information (if applicable)
- Software title | Qbittorrent
- Was the software title installed freshly or updated/migrated?
Kinda. This problem existed from fresh installation of qbittorrent using dietpi scripts(Debian default package for bullseye). Later updated it to the latest version with manual compilation for RPi from [qbittorrent docs](https://github.com/qbittorrent/qBittorrent/wiki/Compilation:-Raspberry-Pi-OS-and-DietPi) and the problem still exist.
- Can this issue be replicated on a fresh installation of DietPi?
Not sure, in my RPi4 I think yes.

#### Steps to reproduce

1. Fresh installation of qbittorrent from `dietpi-software`.
2. After a while, qbittorrent GUI stops loading or just goes unresponsive.(maybe after 1-5+hrs). Weirdly, qbittorent-nox works fine.

#### Expected behaviour

- Well, qbittorrent GUI should be working or at least load in browser, which it doesn't.

#### Extra details

For some reason, qbittorrent GUI stops working or loading in browsers after a while(fresh install). Not sure why. I tried investigating the problem, by restarting it, and then it works fine for a while, I also checked it doesn't contain much of torrents in it too, so I don't assume it isn't the issue. After it stops loading again (not immediately) I checked both `dietpi-services` and `systemctl status qbittorrent.service` they both were running fine/active which is weird. I tried looking for issues in search engines, GitHub dietpi & qbittorrent regarding this, well no luck there. Later found out that the package provided by dietpi scripts is default/old version from Debian repo for bullseye. So I thought if I install new version the problem might have been resolved by now. So, I started working on it.

First, I removed the qbittorrent using uninstall from `dietpi-software`. The started compiling the latest version(v4.4.x) from [qbittorrent RPi compilation docs](https://github.com/qbittorrent/qBittorrent/wiki/Compilation:-Raspberry-Pi-OS-and-DietPi) and I did stumble upon some [problems](https://github.com/qbittorrent/qBittorrent/discussions/17068#discussion-4088479) on the way, but luckily @kolcha helped me in solving it. After successfully compiling it, it worked for a while and the same problem persist.

I am not sure what is the problem with this. I will create an issue in qbittorrent repo too.

Screenshot of qbittorrent-nox running: (v4.4.x sadly I don't screenshot of default install, instead only the compiled version)
[image]

Screenshot of qbittorrent gui not loading:
[image]
[image]

MichaIng, 2022-05-29:
Can you show the following (regarding the "Assignment outside of section" error):
```sh
systemctl cat qbittorrent
```
And after the web interface is not available anymore:
```sh
ss -tlpn '( sport = :1340 )'
journalctl -u qbittorrent
```

dasrocky, 2022-05-29:
My title of the issue is little misleading, qbittorrent webui works if the system boots or reboots, but it stops working after a while, whereas qbittorrent-nox works fine all the time(just to be sure). 
> Can you show the following (regarding the "Assignment outside of section" error):
> 
> ```shell
> systemctl cat qbittorrent
> ```

```
root@DietPi:~# systemctl cat qbittorrent
# /etc/systemd/system/qbittorrent.service
Description=qBittorrent Daemon Service
After=network.target

[Service]
Environment="LD_LIBRARY_PATH=/usr/local/lib:"
User=qbittorrent
#Group=dietpi
UMask=002
ExecStart=/usr/local/bin/qbittorrent-nox
ExecStop=/usr/bin/killall -w qbittorrent-nox

[Install]
WantedBy=multi-user.target
```
"UMask" command, I got it from dietpi source script(i checked online regarding it too). Also commented `Group:dietpi` didn't find it in dietpi script. 

> And after the web interface is not available anymore:
> 
> ```shell
> ss -tlpn '( sport = :1340 )'
> journalctl -u qbittorrent
> ```

```
root@DietPi:~# ss -tlpn '( sport = :1340 )'
State               Recv-Q              Send-Q                           Local Address:Port                           Peer Address:Port             Process              
LISTEN              51                  50                                           *:1340                                      *:*                 users:(("qbittorrent-nox",pid=872,fd=39))

```
```
root@DietPi:~# journalctl -u qbittorrent
-- Journal begins at Sun 2022-05-29 15:06:31 +04, ends at Sun 2022-05-29 20:30:27 +04. --
-- No entries --
```

MichaIng, 2022-05-29:
This is to restore our original service file:
```
cat << _EOF_ > /etc/systemd/system/qbittorrent.service
[Unit]
Description=qBittorrent (DietPi)
Documentation=man:qbittorrent-nox(1)
Wants=network-online.target
After=network-online.target

[Service]
User=qbittorrent
UMask=002
LogsDirectory=qbittorrent
ExecStart=/usr/local/bin/qbittorrent-nox

[Install]
WantedBy=multi-user.target
_EOF_
systemctl daemon-reload
```
The `[Unit]` section declaration is missing in your's. Was it like that when compiling it from source + `make install` or so? If so there is a bug in the `Makefile` which we should report.

We do not set `Group=dietpi` anymore but instead make `dietpi` the primary group for the `qbittorrent` user:
```sh
usermod -g dietpi qbittorrent
```

So the web UI listener is up. It cannot be due to a somehow changed IP or other network connection issue? Strange is the completely missing journal log, as if the service was not starting at all 🤔. But the service status still shows "active" while `journalctl -u qbittorrent` is empty?
```sh
systemctl status qbittorrent
```

dasrocky, 2022-05-29:
> So the web UI listener is up. It cannot be due to a somehow changed IP or other network connection issue? Strange is the completely missing journal log, as if the service was not starting at all thinking. But the service status still shows "active" while `journalctl -u qbittorrent` is empty?

See, that is the weird part, I have no idea why it isn't open webui although qbit-nox is running. I thought by now it is bugged or something, I wanted to post issue in qbittorrent repo, but I thought it is better to do it here as I thought it might be only because I am doing something wrong.

> ```shell
> systemctl status qbittorrent
> ```

I have already shared the pic with that code(first screenshot on top). Here is another screenshot with different time. I will do that again after make changes to .service file as u suggested and share it here.
[image]

> The `[Unit]` section declaration is missing in your's. Was it like that when compiling it from source + `make install` or so? If so there is a bug in the `Makefile` which we should report.

No, this one is just copy and paste with little modification from [qbittorrent compilation docs for rpi](https://github.com/qbittorrent/qBittorrent/wiki/Compilation:-Raspberry-Pi-OS-and-DietPi#create-systemd-service-file). i completely missed [unit] part even comparing with dietpi script for qbitttorrent 😅
screenshot of qbittorrent.service file for dietpi 
[image]

I will modify the service file and let u know, does the problem still exist or not

MichaIng, 2022-05-29:
Ah, so this needs to be fixed in that Wiki 🙂.

dasrocky, 2022-05-30:
> Ah, so this needs to be fixed in that Wiki 🙂.

I can do that, although never did something like that, well first time for everything.
```
[Unit]
Description=qBittorrent (DietPi)
Documentation=man:qbittorrent-nox(1)
Wants=network-online.target
After=network-online.target

[Service]
User=qbittorrent
UMask=002
LogsDirectory=qbittorrent
ExecStart=/usr/local/bin/qbittorrent-nox
ExecStop=/usr/bin/killall -w qbittorrent-nox

[Install]
WantedBy=multi-user.target
```

Okay updates on qbittorrent webui, it still stops working after a while, but this time journal records something:
```
root@DietPi:~# systemctl status qbittorrent
● qbittorrent.service - qBittorrent (DietPi)
     Loaded: loaded (/etc/systemd/system/qbittorrent.service; disabled; vendor preset: enabled)
    Drop-In: /etc/systemd/system/qbittorrent.service.d
             └─dietpi-services_edit.conf
     Active: active (running) since Mon 2022-05-30 01:54:10 +04; 9h ago
       Docs: man:qbittorrent-nox(1)
   Main PID: 2675 (qbittorrent-nox)
      Tasks: 13 (limit: 3948)
        CPU: 16min 50.714s
     CGroup: /system.slice/qbittorrent.service
             └─2675 /usr/local/bin/qbittorrent-nox

May 30 01:54:10 DietPi systemd[1]: Started qBittorrent (DietPi).
root@DietPi:~# journalctl -u qbittorrent
-- Journal begins at Mon 2022-05-30 01:50:57 +04, ends at Mon 2022-05-30 11:43:37 +04. --
May 30 01:52:03 DietPi systemd[1]: Started qBittorrent (DietPi).
May 30 01:52:41 DietPi systemd[1]: Stopping qBittorrent (DietPi)...
May 30 01:52:41 DietPi qbittorrent-nox[738]: Catching signal: SIGTERM
May 30 01:52:41 DietPi qbittorrent-nox[738]: Exiting cleanly
May 30 01:52:46 DietPi qbittorrent-nox[738]: ******** Information ********
May 30 01:52:46 DietPi qbittorrent-nox[738]: To control qBittorrent, access the WebUI at: http://localhost:1340
May 30 01:52:48 DietPi systemd[1]: qbittorrent.service: Succeeded.
May 30 01:52:48 DietPi systemd[1]: Stopped qBittorrent (DietPi).
May 30 01:52:48 DietPi systemd[1]: qbittorrent.service: Consumed 3.690s CPU time.
May 30 01:54:10 DietPi systemd[1]: Started qBittorrent (DietPi).
```
My present .service file: environment ld_library_path needed to run the app.
```
[Unit]
Description=qBittorrent (DietPi)
Documentation=man:qbittorrent-nox(1)
Wants=network-online.target
After=network-online.target

[Service]
User=qbittorrent
UMask=002
LogsDirectory=qbittorrent
ExecStart=/usr/local/bin/qbittorrent-nox
ExecStop=/usr/bin/killall -w qbittorrent-nox
Environment=LD_LIBRARY_PATH=/usr/local/lib:${LD_LIBRARY_PATH}

[Install]
WantedBy=multi-user.target
```

Now we can say it is bugged or something. So I will make issue in qbittorrent repo.

closed 2022-05-31 by dasrocky

MichaIng, 2022-06-03:
Great, but strange indeed, since the directives below `[Unit]` are without practical effect. `dietpi-services` starts the service anyway manually at a later boot stage (regarding `network-online.target`).