MichaIng/DietPi #5530: Error compiling Vaulvarden

Gluca235, 2022-05-29:
#### Details:
- Date           | Sun May 29 12:01:05 CEST 2022
- DietPi version | v8.5.1 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | From scratch
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | `Linux 5.15.32-v8+ #1538 SMP PREEMPT Thu Mar 31 19:40:39 BST 2022 aarch64 GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=0)
- Command        | `cargo build --features sqlite --release`
- Exit code      | 101
- Software title | DietPi-Software
- 
#### Steps to reproduce:

1. Trying to update Vaultvarden with  "sudo dietpi-software reinstall 183" 

- ...
#### Extra details:
 The service still working, but can't access to the Vault page anymore 
- ...
#### Additional logs:
```
info: syncing channel updates for 'stable-aarch64-unknown-linux-gnu'
info: latest update on 2022-05-19, rust version 1.61.0 (fe5b13d68 2022-05-18)
info: downloading component 'cargo'
info: downloading component 'rust-std'
info: downloading component 'rustc'
info: installing component 'cargo'
info: installing component 'rust-std'
info: installing component 'rustc'
    Updating git repository `https://github.com/BlackDex/job_scheduler`
    Updating crates.io index
 Downloading crates ...
  Downloaded async-stream-impl v0.3.3
  Downloaded async-stream v0.3.3
  Downloaded async-compression v0.3.13
  Downloaded base64 v0.13.0
  Downloaded block-buffer v0.10.2
  Downloaded byteorder v1.4.3
  Downloaded mime v0.3.16
  Downloaded cached_proc_macro v0.12.0
  Downloaded miniz_oxide v0.5.1
  Downloaded multer v2.0.2
  Downloaded nix v0.24.1
  Downloaded nom v7.1.1
  Downloaded num-integer v0.1.45
  Downloaded num-traits v0.2.15
  Downloaded ctr v0.8.0
  Downloaded once_cell v1.10.0
  Downloaded openssl v0.10.40
  Downloaded parking_lot v0.12.0
  Downloaded pear_codegen v0.2.3
  Downloaded paste v1.0.7
  Downloaded pest_meta v2.1.3
  Downloaded pin-utils v0.1.0
  Downloaded ppv-lite86 v0.2.16
  Downloaded proc-macro-hack v0.5.19
  Downloaded quick-error v1.2.3
  Downloaded rand v0.4.6
  Downloaded r2d2 v0.8.9
  Downloaded ref-cast v1.0.7
  Downloaded remove_dir_all v0.5.3
  Downloaded resolv-conf v0.7.0
  Downloaded parking_lot_core v0.2.14
  Downloaded percent-encoding v2.1.0
  Downloaded rand v0.7.3
  Downloaded quick-error v2.0.1
  Downloaded reqwest v0.11.10
  Downloaded quanta v0.9.3
  Downloaded ryu v1.0.9
[... 577 lines cut ...]
   Compiling phf_codegen v0.10.0
   Compiling migrations_macros v1.4.2
   Compiling syslog v6.0.1
   Compiling simple_asn1 v0.6.1
   Compiling governor v0.4.2
   Compiling cached v0.34.0
   Compiling cookie_store v0.15.1
   Compiling trust-dns-resolver v0.20.4
   Compiling job_scheduler v1.2.1 (https://github.com/BlackDex/job_scheduler?rev=9100fc596a083fd9c0b560f8f11f108e0a19d07e#9100fc59)
   Compiling rocket_http v0.5.0-rc.2
   Compiling chrono-tz-build v0.0.2
   Compiling cookie_store v0.16.0
   Compiling fern v0.6.1
   Compiling jsonwebtoken v8.1.0
   Compiling chrono-tz v0.6.1
   Compiling diesel_migrations v1.4.0
   Compiling hyper-tls v0.5.0
   Compiling rocket_codegen v0.5.0-rc.2
   Compiling reqwest v0.11.10
   Compiling yubico v0.11.0
error: could not compile `vaultwarden`

Caused by:
  process didn't exit successfully: `rustc --crate-name vaultwarden --edition=2021 src/main.rs --error-format=json --json=diagnostic-rendered-ansi,artifacts,future-incompat --crate-type bin --emit=dep-info,link -C opt-level=3 -C lto=thin --cfg 'feature="libsqlite3-sys"' --cfg 'feature="sqlite"' -C metadata=a83be2069b6a9a76 -C extra-filename=-a83be2069b6a9a76 --out-dir /opt/vaultwarden/target/release/deps -C strip=debuginfo -L dependency=/opt/vaultwarden/target/release/deps --extern backtrace=/opt/vaultwarden/target/release/deps/libbacktrace-8bb6fc68e4a228c4.rlib --extern bytes=/opt/vaultwarden/target/release/deps/libbytes-478b01a89291ace7.rlib --extern cached=/opt/vaultwarden/target/release/deps/libcached-74afb4fec5cf8bff.rlib --extern chashmap=/opt/vaultwarden/target/release/deps/libchashmap-41a966aeb2079801.rlib --extern chrono=/opt/vaultwarden/target/release/deps/libchrono-f906c7b429ef21d1.rlib --extern chrono_tz=/opt/vaultwarden/target/release/deps/libchrono_tz-e51beec82635ba89.rlib --extern cookie=/opt/vaultwarden/target/release/deps/libcookie-db15f2d62e25cc7b.rlib --extern cookie_store=/opt/vaultwarden/target/release/deps/libcookie_store-8e62dd1e342778ec.rlib --extern ctrlc=/opt/vaultwarden/target/release/deps/libctrlc-ed7507bb65349fad.rlib --extern data_encoding=/opt/vaultwarden/target/release/deps/libdata_encoding-92d07f4aa32f63fa.rlib --extern data_url=/opt/vaultwarden/target/release/deps/libdata_url-dec429fca4dad6b3.rlib --extern diesel=/opt/vaultwarden/target/release/deps/libdiesel-31d7c72973f2d6b4.rlib --extern diesel_migrations=/opt/vaultwarden/target/release/deps/libdiesel_migrations-d7e2074742e476a7.rlib --extern dotenvy=/opt/vaultwarden/target/release/deps/libdotenvy-3d6d0a3db794d756.rlib --extern fern=/opt/vaultwarden/target/release/deps/libfern-4cc021884f39495b.rlib --extern futures=/opt/vaultwarden/target/release/deps/libfutures-384b26d0435018d7.rlib --extern governor=/opt/vaultwarden/target/release/deps/libgovernor-409f50b1e04a3bb9.rlib --extern handlebars=/opt/vaultwarden/target/release/deps/libhandlebars-38fc3cfbeab72305.rlib --extern html5gum=/opt/vaultwarden/target/release/deps/libhtml5gum-ed05d7c00ce31a96.rlib --extern idna=/opt/vaultwarden/target/release/deps/libidna-fcd039b6e5e635b4.rlib --extern job_scheduler=/opt/vaultwarden/target/release/deps/libjob_scheduler-4eea64efa53e95ca.rlib --extern jsonwebtoken=/opt/vaultwarden/target/release/deps/libjsonwebtoken-637a354696b69dbc.rlib --extern lettre=/opt/vaultwarden/target/release/deps/liblettre-6e78b97a391eee9c.rlib --extern libsqlite3_sys=/opt/vaultwarden/target/release/deps/liblibsqlite3_sys-0152b5f1b60f43dc.rlib --extern log=/opt/vaultwarden/target/release/deps/liblog-5ecf186a9c1eeffd.rlib --extern num_derive=/opt/vaultwarden/target/release/deps/libnum_derive-dfcf91356734d9a0.so --extern num_traits=/opt/vaultwarden/target/release/deps/libnum_traits-a4e07e8c1577493e.rlib --extern once_cell=/opt/vaultwarden/target/release/deps/libonce_cell-f718696775d31ca4.rlib --extern openssl=/opt/vaultwarden/target/release/deps/libopenssl-2f0de8783b34a34d.rlib --extern ws=/opt/vaultwarden/target/release/deps/libparity_ws-cc0dccc042441c00.rlib --extern paste=/opt/vaultwarden/target/release/deps/libpaste-78a55d2c1e7869b1.so --extern percent_encoding=/opt/vaultwarden/target/release/deps/libpercent_encoding-405bb8c0f9c06788.rlib --extern pico_args=/opt/vaultwarden/target/release/deps/libpico_args-500f27ca02f5ceff.rlib --extern rand=/opt/vaultwarden/target/release/deps/librand-f5abd520473e7064.rlib --extern regex=/opt/vaultwarden/target/release/deps/libregex-5bfba6bb4a5bcebf.rlib --extern reqwest=/opt/vaultwarden/target/release/deps/libreqwest-e57d169f4e894a9d.rlib --extern ring=/opt/vaultwarden/target/release/deps/libring-56c6e8d9fce7baf4.rlib --extern rmpv=/opt/vaultwarden/target/release/deps/librmpv-ec1a546a1ede950f.rlib --extern rocket=/opt/vaultwarden/target/release/deps/librocket-05cb9c0d5f8a765e.rlib --extern serde=/opt/vaultwarden/target/release/deps/libserde-bdd97c66da62a898.rlib --extern serde_json=/opt/vaultwarden/target/release/deps/libserde_json-249816585476010f.rlib --extern syslog=/opt/vaultwarden/target/release/deps/libsyslog-db8f95f44d259d49.rlib --extern time=/opt/vaultwarden/target/release/deps/libtime-7070b35a3e8c7a36.rlib --extern tokio=/opt/vaultwarden/target/release/deps/libtokio-04533621b22d98a4.rlib --extern totp_lite=/opt/vaultwarden/target/release/deps/libtotp_lite-d39873b930371ce0.rlib --extern tracing=/opt/vaultwarden/target/release/deps/libtracing-2e31b13690d8a076.rlib --extern url=/opt/vaultwarden/target/release/deps/liburl-890d913f7834966e.rlib --extern uuid=/opt/vaultwarden/target/release/deps/libuuid-a478bb99599ff3c7.rlib --extern webauthn_rs=/opt/vaultwarden/target/release/deps/libwebauthn_rs-edc2a41d171db62e.rlib --extern yubico=/opt/vaultwarden/target/release/deps/libyubico-f7ad989c38497fa7.rlib --cfg sqlite -L native=/opt/vaultwarden/target/release/build/libsqlite3-sys-e52fafab49b35746/out -L native=/opt/vaultwarden/target/release/build/ring-ee21f17f297afff6/out` (signal: 9, SIGKILL: kill)
```

Gluca235, 2022-05-29:
process didn't exit successfully: `rustc --crate-name vaultwarden --edition=2021 src/main.rs --error-format=json --json=diagnostic-rendered-ansi,artifacts,future-incompat --crate-type bin --emit=dep-info,link -C opt-level=3 -C lto=thin --cfg 'feature="libsqlite3-sys"' --cfg 'feature="sqlite"' -C metadata=a83be2069b6a9a76 -C extra-filename=-a83be2069b6a9a76 --out-dir /opt/vaultwarden/target/release/deps -C strip=debuginfo -L dependency=/opt/vaultwarden/target/release/deps --extern backtrace=/opt/vaultwarden/target/release/deps/libbacktrace-8bb6fc68e4a228c4.rlib --extern bytes=/opt/vaultwarden/target/release/deps/libbytes-478b01a89291ace7.rlib --extern cached=/opt/vaultwarden/target/release/deps/libcached-74afb4fec5cf8bff.rlib --extern chashmap=/opt/vaultwarden/target/release/deps/libchashmap-41a966aeb2079801.rlib --extern chrono=/opt/vaultwarden/target/release/deps/libchrono-f906c7b429ef21d1.rlib --extern chrono_tz=/opt/vaultwarden/target/release/deps/libchrono_tz-e51beec82635ba89.rlib --extern cookie=/opt/vaultwarden/target/release/deps/libcookie-db15f2d62e25cc7b.rlib --extern cookie_store=/opt/vaultwarden/target/release/deps/libcookie_store-8e62dd1e342778ec.rlib --extern ctrlc=/opt/vaultwarden/target/release/deps/libctrlc-ed7507bb65349fad.rlib --extern data_encoding=/opt/vaultwarden/target/release/deps/libdata_encoding-92d07f4aa32f63fa.rlib --extern data_url=/opt/vaultwarden/target/release/deps/libdata_url-dec429fca4dad6b3.rlib --extern diesel=/opt/vaultwarden/target/release/deps/libdiesel-31d7c72973f2d6b4.rlib --extern diesel_migrations=/opt/vaultwarden/target/release/deps/libdiesel_migrations-d7e2074742e476a7.rlib --extern dotenvy=/opt/vaultwarden/target/release/deps/libdotenvy-3d6d0a3db794d756.rlib --extern fern=/opt/vaultwarden/target/release/deps/libfern-4cc021884f39495b.rlib --extern futures=/opt/vaultwarden/target/release/deps/libfutures-384b26d0435018d7.rlib --extern governor=/opt/vaultwarden/target/release/deps/libgovernor-409f50b1e04a3bb9.rlib --extern handlebars=/opt/vaultwarden/target/release/deps/libhandlebars-38fc3cfbeab72305.rlib --extern html5gum=/opt/vaultwarden/target/release/deps/libhtml5gum-ed05d7c00ce31a96.rlib --extern idna=/opt/vaultwarden/target/release/deps/libidna-fcd039b6e5e635b4.rlib --extern job_scheduler=/opt/vaultwarden/target/release/deps/libjob_scheduler-4eea64efa53e95ca.rlib --extern jsonwebtoken=/opt/vaultwarden/target/release/deps/libjsonwebtoken-637a354696b69dbc.rlib --extern lettre=/opt/vaultwarden/target/release/deps/liblettre-6e78b97a391eee9c.rlib --extern libsqlite3_sys=/opt/vaultwarden/target/release/deps/liblibsqlite3_sys-0152b5f1b60f43dc.rlib --extern log=/opt/vaultwarden/target/release/deps/liblog-5ecf186a9c1eeffd.rlib --extern num_derive=/opt/vaultwarden/target/release/deps/libnum_derive-dfcf91356734d9a0.so --extern num_traits=/opt/vaultwarden/target/release/deps/libnum_traits-a4e07e8c1577493e.rlib --extern once_cell=/opt/vaultwarden/target/release/deps/libonce_cell-f718696775d31ca4.rlib --extern openssl=/opt/vaultwarden/target/release/deps/libopenssl-2f0de8783b34a34d.rlib --extern ws=/opt/vaultwarden/target/release/deps/libparity_ws-cc0dccc042441c00.rlib --extern paste=/opt/vaultwarden/target/release/deps/libpaste-78a55d2c1e7869b1.so --extern percent_encoding=/opt/vaultwarden/target/release/deps/libpercent_encoding-405bb8c0f9c06788.rlib --extern pico_args=/opt/vaultwarden/target/release/deps/libpico_args-500f27ca02f5ceff.rlib --extern rand=/opt/vaultwarden/target/release/deps/librand-f5abd520473e7064.rlib --extern regex=/opt/vaultwarden/target/release/deps/libregex-5bfba6bb4a5bcebf.rlib --extern reqwest=/opt/vaultwarden/target/release/deps/libreqwest-e57d169f4e894a9d.rlib --extern ring=/opt/vaultwarden/target/release/deps/libring-56c6e8d9fce7baf4.rlib --extern rmpv=/opt/vaultwarden/target/release/deps/librmpv-ec1a546a1ede950f.rlib --extern rocket=/opt/vaultwarden/target/release/deps/librocket-05cb9c0d5f8a765e.rlib --extern serde=/opt/vaultwarden/target/release/deps/libserde-bdd97c66da62a898.rlib --extern serde_json=/opt/vaultwarden/target/release/deps/libserde_json-249816585476010f.rlib --extern syslog=/opt/vaultwarden/target/release/deps/libsyslog-db8f95f44d259d49.rlib --extern time=/opt/vaultwarden/target/release/deps/libtime-7070b35a3e8c7a36.rlib --extern tokio=/opt/vaultwarden/target/release/deps/libtokio-04533621b22d98a4.rlib --extern totp_lite=/opt/vaultwarden/target/release/deps/libtotp_lite-d39873b930371ce0.rlib --extern tracing=/opt/vaultwarden/target/release/deps/libtracing-2e31b13690d8a076.rlib --extern url=/opt/vaultwarden/target/release/deps/liburl-890d913f7834966e.rlib --extern uuid=/opt/vaultwarden/target/release/deps/libuuid-a478bb99599ff3c7.rlib --extern webauthn_rs=/opt/vaultwarden/target/release/deps/libwebauthn_rs-edc2a41d171db62e.rlib --extern yubico=/opt/vaultwarden/target/release/deps/libyubico-f7ad989c38497fa7.rlib --cfg sqlite -L native=/opt/vaultwarden/target/release/build/libsqlite3-sys-e52fafab49b35746/out -L native=/opt/vaultwarden/target/release/build/ring-ee21f17f297afff6/out` (signal: 9, SIGKILL: kill)
```

MichaIng, 2022-05-29:
Many thanks for your report.

It got SIGKILLed, probably due to insufficient memory. Can you show:
```sh
free -m
swapon
```

Gluca235, 2022-05-29:
Thanks to you, to keep this work going on

              total        used        free      shared  buff/cache   available
Mem:            3844          92        3576          14         174        3671
Swap:            251           0         251

I delted some folders before reinstall 183, but not rebooted. I use my Raspberry for NAS, PiHole-Unbound, Vaultwarden

MichaIng, 2022-05-29:
Can you try to reduce the build to run two jobs concurrently only:
```sh
sudo sed -i 's/cargo build/cargo build -j 2/' /boot/dietpi/dietpi-software
sudo dietpi-software reinstall 183
```

Gluca235, 2022-05-30:
I tried, and now it fail at "cargo build"

info: syncing channel updates for 'stable-aarch64-unknown-linux-gnu'
info: latest update on 2022-05-19, rust version 1.61.0 (fe5b13d68 2022-05-18)
info: downloading component 'cargo'
info: downloading component 'rust-std'
info: downloading component 'rustc'
info: installing component 'cargo'
info: installing component 'rust-std'
info: installing component 'rustc'
error: The argument '--jobs <N>' was provided more than once, but cannot be used multiple times

Gluca235, 2022-05-30:
I think the problem now is this command:
cargo build -j 2 -j 2 --features sqlite --release__

Gluca235, 2022-05-30:
Ok, it worked this time. Took a very long time:

Finished release [optimized] target(s) in 54m 41s

closed 2022-05-30 by Joulinar

MichaIng, 2022-05-30:
> cargo build -j 2 -j 2 --features sqlite --release

Did you run the `sed` two times accidentally? 🙂