MichaIng/DietPi #5531: dietpi scripts: don't restart disabled/stopped services

checkmate777, 2022-05-29:
Got grafana-server installed, however I stopped and disabled it through systemctl, as I'm not using it.

While upgrading to dietpi v8.5.1 through dietpi-update at the end grafana-server was (re)started.

That shall not happed. dietpi scripts shall honor systemctl settings for any service.

MichaIng, 2022-05-29:
Via `dietpi-services` you can exclude services from being started/stopped. Background is explained here: https://github.com/MichaIng/DietPi/issues/5470#issuecomment-1116037377

checkmate777, 2022-05-29:
Done, thank you.

> Why don't you uninstall Grafana when you don't use it?
Because I have the db set up, however the ingestion process is not done yet.

MichaIng, 2022-05-29:
Okay, I agree it is confusing and non-intuitive, hence we plan to change it (as explained in linked issue). But there are always more tasks/requests than time to address them 😅.

I'll mark this issue as closed then, feel free to reopen if required.

closed 2022-05-29 by MichaIng