MichaIng/DietPi #5532: Dietpi-VPN statred config, canceled, but shows up in services and cannot start

NukeManDan, 2022-05-30:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=5
G_DIETPI_VERSION_RC=1
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. open `dietpi-vpn`
2. close dialog (select custom first)
3. check running services

`dietpi-services status`

 DietPi-Services
─────────────────────────────────────────────────────
 Mode: status 

[FAILED] DietPi-Services | ● dietpi-vpn.service - VPN Client (DietPi)
     Loaded: loaded (/etc/systemd/system/dietpi-vpn.service; disabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Sun 2022-05-29 15:43:17 MDT; 20h ago
    Process: 31833 ExecStart=/usr/sbin/openvpn --suppress-timestamps --nobind --config /etc/openvpn/client.ovpn (code=exited, status=203/EXEC)
   Main PID: 31833 (code=exited, status=203/EXEC)
        CPU: 18ms

#### Expected behaviour

- closing dialog cleans up, no dangling server
- detects VPN is installed. I used PiVPN, with wiregaurd. Not OpenVPN.

Joulinar, 2022-05-30:
> closing dialog cleans up, no dangling server

There is a reset button within the provider configuration menu. This should clean up all config files. As well you should be ask if `openvpn` should be removed.

> detects VPN is installed. I used PiVPN, with wiregaurd. Not OpenVPN.

This feature is not supported. We do not detect existing 3rd party VPN clients. 

`dietpi-vpn` is acting as VPN client and not as server like PiVPN. As well we support OpenVPN only (as client)

MichaIng, 2022-09-17:
Not sure how the service could be running (and failing), which is only started when selecting "Apply", which is not part of your steps to reproduce. However, "Reset" will now also try to remove /etc/openvpn, if existing and empty, which was not done before.

I'll mark this as closed.

closed 2022-09-17 by MichaIng