MichaIng/DietPi #5535: Ping to dns9.quad9.net failed

almirus, 2022-05-30:
### Creating a bug report/issue

#### Required Information
- DietPi version | `G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=5
G_DIETPI_VERSION_RC=1
G_GITBRANCH='master'
G_GITOWNER='MichaIng'`
- Distro version | `bullseye`
- Kernel version | `Linux DietPi 5.10.69-11-amlogic-g57a25a2f2f47 #amlogic SMP PREEMPT Sat Mar 12 07:47:53 UTC 2022 aarch64 GNU/Linux`
- SBC model | `Radxa Zero (aarch64)`
- Power supply used | 5V 2A 
- SD card used | SanDisk ultra

#### Additional Information (if applicable)
- Software title | `try to install Python via dietpi-software`

- Bug report ID | `094922a5-e81d-47d3-95c4-65f472de5d43`

#### Steps to reproduce

1. dietpi-software
2. search Python
3. Ping error

But ping to google.com OK

Joulinar, 2022-05-30:
You could adjust the following in `/boot/dietpi.txt` to point to `dns.google.com` and `8.8.8.8`

```
# General connection and DNS testing
# - IP to ping when checking network connectivity. Default: 9.9.9.9 (Quad9 DNS IP)
CONFIG_CHECK_CONNECTION_IP=9.9.9.9
# - Domain to ping when checking DNS resolver. Default: dns9.quad9.net (Quad9 DNS domain)
CONFIG_CHECK_DNS_DOMAIN=dns9.quad9.net
```

MichaIng, 2022-07-23:
Marking as closed. Feel free to reopen if required.

closed 2022-07-23 by MichaIng