MichaIng/DietPi #5536: Amiberry activity leds (DF0: etc.) not working

DJP75, 2022-05-31:
### Creating a bug report/issue

#### Additional Information (if applicable)
- Software title | Amiberry
- Was the software title installed freshly or updated/migrated? Never workded
- Can this issue be replicated on a fresh installation of DietPi? Yes

#### Steps to reproduce

1. Assign a funcion (power , DF0: DF1: etc) to keyboard LEDs (numlock,caps,scroll) in Amiberry 

#### Expected behaviour
- LED blinking as disk activity in emulator

#### Actual behaviour
- LED stays off

#### Extra details
- pressing numlock key and using setleds +num command,LED turn on and off as expected

MichaIng, 2022-06-04:
Many thanks for your report.

Did this work with the version we provided before DietPi v8.5?

Are there probably some related error logs, either:
```sh
journalctl -u amiberry
```
or when you hit ALT+F3 to switch to TTY3 console where Amiberry runs on?

I see now there is a new optional build option `USE_GPIOD`, which we do not set, but it is for GPIO controlled LEDs.

@midwan
May I ask you quickly, I guess this is managed via SDL2, probably you know some library/headers/flag that is required to make keyboard LEDs functional?

midwan, 2022-06-04:
@MichaIng 
This is handled by calls to `ioctl` directly, not through SDL2. From what I remember, this might be related to the user permissiosn on the system, but I don't recall which group exactly might be responsible for it. I remember that I saw different results when logging in as a normal user vs root for example.

MichaIng, 2022-06-04:
Hmm, on DietPi, Amiberry (still) runs as root, though I actually want to change this when I find time. So at least there shouldn't be permission issues.

midwan, 2022-06-04:
I think it was when running as root that it _didn't_ work actually. Perhaps `root` is not a member of some groups that are required?

midwan, 2022-06-04:
Alternatively, perhaps the console it's running under has something to do with it.
Do you still launch Amiberry under X? That might also be related.

MichaIng, 2022-06-04:
Usually root has simple full permissions for everything, at least when it's about file permissions, it doesn't need to be member of any group and can even edit 000 mode files etc. But I'm not 100% sure how it is with system calls.

We use KMS/DRM, on RPi with DispmanX, so no X server involved. It's started as systemd service with `ExecStartPre=/bin/chvt 3` as at least with previous versions every keyboard input was printed to the console, and we don't want to have all this output on TTY1 😉.

midwan, 2022-06-04:
Hm, could it be that starting it as a service is related then?
It should be possible to test this scenario, if it is launched manually after logging in, and seeing if the behavior is different.

MichaIng, 2022-06-04:
@DJP75 
Could you try this:
```sh
usermod -aG tty,input root
```

And if it doesn't help, following midwan's idea:
```sh
export LD_LIBRARY_PATH=/mnt/dietpi_userdata/amiberry/lib
cd /mnt/dietpi_userdata/amiberry
/mnt/dietpi_userdata/amiberry/amiberry
```

DJP75, 2022-06-05:
Wow! The second solution works!!

MichaIng, 2022-06-25:
Interesting, so it is related to the systemd service environment somehow. The service does basically the same, but only switches to a different TTY to keep the main TTY  free from keyboard input spam and avoid interference with Amiberry and ongoing boot messages. Could be actually tested if this is still required with latest SDL2:
```
[Unit]
Description=Amiberry Amiga Emulator (DietPi)
Documentation=https://github.com/midwan/amiberry/wiki

[Service]
WorkingDirectory=/mnt/dietpi_userdata/amiberry
Environment=LD_LIBRARY_PATH=/mnt/dietpi_userdata/amiberry/lib
ExecStartPre=/bin/chvt 3
ExecStart=/mnt/dietpi_userdata/amiberry/amiberry
ExecStopPost=/bin/chvt 1

[Install]
WantedBy=local-fs.target
```
Is it probably required to set `StandardInput=tty` or even `StandardOutput=tty` so that the console device is occupied by the service? Not sure how that would be related to `ioctl` but lacking of deeper understanding I have no other idea 😅.

MichaIng, 2023-01-27:
It can be replicate quite easily:
```sh
setleds +num # numlock enabled
setleds -num # numlock disabled
systemd-run setleds +num # nothing changes
```
This is since `setleds` sets flags for its STDIN TTY, and within a systemd service environment, there is not STDIN by default. Good that one can assign a STDIN TTY and that we do already move to `/dev/tty3` to assure the local console is not polluted. For some reason, doing so with Amiberry also resolves an issue where `systemctl stop amibery`, respectively SIGTERM and even SIGKILLing it didn't really work. Looks like a STDIN VT is for some reason helpful in general. So this is how to do it:
```sh
cat << '_EOF_' > lib/systemd/system/amiberry.service
[Unit]
Description=Amiberry Amiga Emulator (DietPi)
Documentation=https://github.com/BlitterStudio/amiberry/wiki

[Service]
WorkingDirectory=/mnt/dietpi_userdata/amiberry
Environment=LD_LIBRARY_PATH=/mnt/dietpi_userdata/amiberry/lib
StandardInput=tty
TTYPath=/dev/tty3
ExecStartPre=/bin/chvt 3
ExecStart=/mnt/dietpi_userdata/amiberry/amiberry
ExecStopPost=/bin/chvt 1

[Install]
WantedBy=local-fs.target
_EOF_
systemctl daemon-reload
systemctl start amiberry
```

MichaIng, 2023-01-27:
Fixed with: https://github.com/MichaIng/DietPi/commit/f2684fb

DJP75, 2023-01-31:
It works!!! Thanks a lot!!!

DJP75, 2023-01-31:
It works!! Thanks a lot!!

MichaIng, 2023-01-31:
Great, many thanks for testing. I moved the builds to stable downloads as well, so issue can be indeed closed.

closed 2023-01-31 by MichaIng

EMPI-PL, 2024-01-25:
Hi. I'm facing the same problem on ARM and x86 boards. You may want to take a look at @midwan's linked issue.
It would seem the solution here would also resolve mine but I don't quite understand this part in https://github.com/MichaIng/DietPi/issues/5536#issuecomment-1146656889 by @MichaIng:
```sh
export LD_LIBRARY_PATH=/mnt/dietpi_userdata/amiberry/lib
```
There's no /lib directory under amiberry code. Is this something available only for DietPi? I guess not but still...
What would be then a solution for x86 boards using Amiberry?
Or did something change since the resolution in amiberry's directory structure?

thanks