MichaIng/DietPi #5537: I can't install radarr from DietPi-Software

eddiermar, 2022-05-31:
### Creating a bug report/issue

#### Details:
- Date           | Tue May 31 14:24:11 -05 2022
- DietPi version | v8.5.1 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi B+ (armv6l) (ID=1)
- Kernel version | `Linux 5.15.32+ #1538 Thu Mar 31 19:37:58 BST 2022 arm                                                                                        v6l GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=1)
- Command        | `curl -ILfvm 20 https://github.com/Radarr/Radarr/releases/dow                                                                                        nload/v4.0.4.5922/Radarr.master.4.0.4.5922.linux.tar.gz`
- Exit code      | 22
- Software title | DietPi-Software
#### Steps to reproduce:

1. Open Dietpi-software select Radarr and install
2. Radarr do not install!
#### Expected behaviour:

- Radarr installs
#### Actual behaviour:

- Error 404
#### Extra details:

- 
#### Additional logs:
```
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0*                                                                                           Trying 140.82.113.4:443...
* Connected to github.com (140.82.113.4) port 443 (#0)
* ALPN, offering h2
* ALPN, offering http/1.1
* successfully set certificate verify locations:
*  CAfile: /etc/ssl/certs/ca-certificates.crt
*  CApath: /etc/ssl/certs
} [5 bytes data]
* TLSv1.3 (OUT), TLS handshake, Client hello (1):
} [512 bytes data]
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0*                                                                                         TLSv1.3 (IN), TLS handshake, Server hello (2):
{ [122 bytes data]
* TLSv1.3 (IN), TLS handshake, Encrypted Extensions (8):
{ [19 bytes data]
* TLSv1.3 (IN), TLS handshake, Certificate (11):
{ [2459 bytes data]
* TLSv1.3 (IN), TLS handshake, CERT verify (15):
{ [79 bytes data]
* TLSv1.3 (IN), TLS handshake, Finished (20):
{ [36 bytes data]
* TLSv1.3 (OUT), TLS change cipher, Change cipher spec (1):
} [1 bytes data]
* TLSv1.3 (OUT), TLS handshake, Finished (20):
} [36 bytes data]
* SSL connection using TLSv1.3 / TLS_AES_128_GCM_SHA256
* ALPN, server accepted to use h2
* Server certificate:
*  subject: C=US; ST=California; L=San Francisco; O=GitHub, Inc.; CN=github.com
*  start date: Mar 15 00:00:00 2022 GMT
*  expire date: Mar 15 23:59:59 2023 GMT
*  subjectAltName: host "github.com" matched cert's "github.com"
*  issuer: C=US; O=DigiCert Inc; CN=DigiCert TLS Hybrid ECC SHA384 2020 CA1
*  SSL certificate verify ok.
* Using HTTP2, server supports multi-use
* Connection state changed (HTTP/2 confirmed)
* Copying HTTP/2 data in stream buffer to connection buffer after upgrade: len=0
} [5 bytes data]
* Using Stream ID: 1 (easy handle 0x1681d50)
} [5 bytes data]
> HEAD /Radarr/Radarr/releases/download/v4.0.4.5922/Radarr.master.4.0.4.5922.lin                                                                                        ux.tar.gz HTTP/2
> Host: github.com
> user-agent: curl/7.74.0
> accept: */*
>
{ [5 bytes data]
* TLSv1.3 (IN), TLS handshake, Newsession Ticket (4):
{ [57 bytes data]
* TLSv1.3 (IN), TLS handshake, Newsession Ticket (4):
{ [57 bytes data]
* old SSL session ID is stale, removing
{ [5 bytes data]
* Connection state changed (MAX_CONCURRENT_STREAMS == 100)!
} [5 bytes data]
* The requested URL returned error: 404
* stopped the pause stream!
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
* Connection #0 to host github.com left intact
curl: (22) The requested URL returned error: 404

Joulinar, 2022-05-31:
@MichaIng could it be that Radarr removed support for ARMv6 since version4? At least I don't see a `Radarr.master.v4xx.linux.tar.gz` file anymore on Radarr GitHub since v4. It seems latest version available is [3.2.2.5080](https://github.com/Radarr/Radarr/releases/tag/v3.2.2.5080)

MichaIng, 2022-05-31:
Yes it is 😞: https://github.com/Radarr/Radarr/pull/6427

Shall we disable Servarr applications for ARMv6 RPi models or serve the last functional (non-maintained/supported) build?

eddiermar, 2022-06-01:
> Yes it is 😞: [Radarr/Radarr#6427](https://github.com/Radarr/Radarr/pull/6427)
> 
> Shall we disable Servarr applications for ARMv6 RPi models or serve the last functional (non-maintained/supported) build?

Difficult question ... I would like ti use radarr. Maybe just serve last functional version. By the way I have been using lidarr, sonarr and bazaar without problems yet.

MichaIng, 2022-06-01:
PR up to fix this: #5539
Precautionary I did the same for Lidarr, as v1 won't support Mono anymore, too.

closed 2022-06-25 by MichaIng