MichaIng/DietPi #5538: NanoPi R2S image : LAN LED always on and can't be changed via 'dietpi-config'

TheLinuxGuy, 2022-05-31:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
```
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=5
G_DIETPI_VERSION_RC=1
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
```
- Distro version : bullseye
- Kernel version | Linux DietPi 5.15.35-rockchip64 #22.05.1 SMP PREEMPT Sat May 28 08:37:27 UTC 2022 aarch64 GNU/Linux
- SBC model | NanoPi R2S (aarch64)
- Power supply used | Raspberry Pi official usb-c PS
- SD card used | Sandisk

#### Additional Information (if applicable)
- Software title | DietPi base OS
- Was the software title installed freshly or updated/migrated? 
- Can this issue be replicated on a fresh installation of DietPi? Yes

- Bug report ID | e0ef442a-e947-45e5-9490-e2ab28daba9a

#### Steps to reproduce

1. Flash NanoPi R2S image, complete all basic steps (includes update and forced reboot)
2. dietpi-config > display > 14 LED control > lan_led > set to "heartbeat" or any option. It wont' work. Even post reboots.

#### Expected behaviour

- Changing lan_led setting to heartbeat or any setting should apply and be seen on the actual device LED. It works fine on other LEDs on this board (system LED and wan LED work fine, even changing settings).

#### Actual behaviour

- LAN LED always solid after system boot.

MichaIng, 2022-06-26:
Many thanks for your report.

I actually recognised the same on some other Armbian kernel based systems. Sometimes 2-3 of the ~15 trigger options work, all others have no effect. Does the "activity" trigger work in your case?

MichaIng, 2022-09-15:
Probably the same works as here: #5679
I'll test this.

MichaIng, 2022-11-18:
The Ethernet status LEDs are now enabled by default, reacting on cable link and blinking on RX/TX: https://github.com/MichaIng/DietPi/commit/427a899

closed 2022-11-18 by MichaIng