MichaIng/DietPi #5540: DietPi version | v8.4.2 (MichaIng/master)

kebabia, 2022-06-01:
#### Details:
- Date           | Wed Jun  1 18:08:40 CEST 2022
- DietPi version | v8.4.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspberry Pi OS Lite (64-bit)
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | `Linux DietPi 5.10.103-v8+ #1529 SMP PREEMPT Tue Mar 8 12:26:46 GMT 2022 aarch64 GNU/Linux`
- Distro         | buster (ID=5,RASPBIAN=0)
- Command        | `ping -nc 1 -W 10 dns9.quad9.net`
- Exit code      | 2
- Software title | DietPi-Update
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
ping: dns9.quad9.net: Temporary failure in name resolution

Joulinar, 2022-06-01:
Looks like a DNS issue as you have a challenge to resolve Quad9 host name. Do you know what DNS server you are using? Your system is correctly connected to the internet?

kebabia, 2022-06-01:
yes know waht a DNS is,
i use my Own DNS...

Joulinar, 2022-06-01:
Have a look to you DNS server if it is resolving Quad9 host correctly. As well check that your Dietpi device has the correct DNS server assigned

kebabia, 2022-06-01:
ok thanks i do that and i ll give feedback .., Thank s

Joulinar, 2022-06-01:
Ok but still a little bit confusing. Usually a local DNS is using a global upstream DNS to resolve DNS queries and there shouldn't be a relationship to the update status of the local DNS server operating system. Anyhow, good it is working now. I will close the issue. Feel free to reopen if needed.