MichaIng/DietPi #5542: Automatic latest Radarr download URL detection failed.

lmerega, 2022-06-01:
When installing this appears:
 DietPi-Software
─────────────────────────────────────────────────────
 Step: Installing Radarr: Automatically download movies

[FAILED] DietPi-Software | Automatic latest Radarr download URL detection failed.
[FAILED] DietPi-Software | "https://github.com/Radarr/Radarr/releases/download/v4.0.4.5922/Radarr.master.4.0.4.5922.linux-core-arm64.tar.gz" will be used as fallback, but a newer version might be available.
[FAILED] DietPi-Software | Please report this at: https://github.com/MichaIng/DietPi/issues

Joulinar, 2022-06-01:
Does the installation finished at the end using the fallback URL?

MichaIng, 2022-06-01:
#5539 fixes this as well. However, not an issue on other architectures since the internal updater will update it to latest version easily.

Joulinar, 2022-07-05:
@MichaIng I guess this one can be closed as we already released DietPi v8.6?

closed 2022-07-05 by MichaIng