MichaIng/DietPi #5543: Odroid C4 Wiring

Guchshenskaya, 2022-06-03:
#### Details:
- Date           | Fri Jun  3 14:44:57 BST 2022
- DietPi version | v8.5.1 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | from scratch
- Hardware       | Odroid C4/HC4 (aarch64) (ID=16)
- Kernel version | `Linux DietPi 5.10.110-meson64 #22.05.1 SMP PREEMPT Sat May 28 07:50:27 UTC 2022 aarch64 GNU/Linux`
- Distro         | bullseye (ID=6)
- Command        | `./configure CFLAGS=-g0 -O3`
- Exit code      | 2
- Software title | DietPi-Software
#### Steps to reproduce:

1. `dietpi-software` -> select 
```70 WiringPi: GPIO interface library (C)```
#### Expected behaviour:

- Library should be installed
#### Actual behaviour:

- Error
#### Extra details:

- ...
#### Additional logs:
```
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking for a BSD-compatible install... /usr/bin/install -c
checking whether ln -s works... yes
checking for ranlib... ranlib
./configure: line 3511: syntax error near unexpected token `LIBCRYPT,'
./configure: line 3511: `PKG_CHECK_MODULES(LIBCRYPT, libcrypt)'
```

MichaIng, 2022-06-26:
Many thanks for your report.

Can you try the following:
```sh
apt install pkg-config
dietpi-software install 70
```

MichaIng, 2022-07-02:
Yes that was it. Fixed for release: https://github.com/MichaIng/DietPi/commit/086f91e

closed 2022-07-02 by MichaIng