MichaIng/DietPi #5552: Dietpi-Update Failing

5oundless, 2022-06-07:
#### Details:
- Date           | Tue Jun  7 09:01:18 CDT 2022
- DietPi version | v8.4.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspberry Pi OS (32-bit) Lite
- Hardware       | RPi 3 Model B+ (armv7l) (ID=3)
- Kernel version | `Linux Pi 5.10.103-v7+ #1529 SMP Tue Mar 8 12:21:37 GMT 2022 armv7l GNU/Linux`
- Distro         | buster (ID=5,RASPBIAN=1)
- Command        | `ping -nc 1 -W 10 dns9.quad9.net`
- Exit code      | 2
- Software title | DietPi-Update
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
ping: dns9.quad9.net: Name or service not known
```

5oundless, 2022-06-07:
Yes, I have ad blocking. Disabling does not resolve the issue.

Yes, the ping to google.com works.

Joulinar, 2022-06-07:
could it be blocked somewhere else? On your ISP? Or some other network component?

Otherwise have a look to following setting in `/boot/dietpi.txt` and adjust it on your needs

```
# General connection and DNS testing
# - IP to ping when checking network connectivity. Default: 9.9.9.9 (Quad9 DNS IP)
CONFIG_CHECK_CONNECTION_IP=9.9.9.9
# - Domain to ping when checking DNS resolver. Default: dns9.quad9.net (Quad9 DNS domain)
CONFIG_CHECK_DNS_DOMAIN=dns9.quad9.net
```

closed 2022-06-07 by 5oundless