MichaIng/DietPi #5553: DietPi-Software | DietPi-VPN: fails to connect when Killswitch is ON

kinoushe, 2022-06-08:
### Creating a bug report/issue

#### Required Information
- DietPi version | `DietPi v8.5.1`
- Distro version | `Bullseye`
- Kernel version | `Linux DietPi64 5.15.32-v8+ #1538 SMP PREEMPT Thu Mar 31 19:40:39 BST 2022 aarch64 GNU/Linux`
- SBC model | `RPi 4 Model B (aarch64)`
- Power supply used | Official

#### Additional Information (if applicable)
- Software title | DietPi-VPN

#### Steps to reproduce

1. Start/Restart VPN connection with Killswitch : [On]

#### Expected behaviour

- VPN should connect

#### Actual behaviour

- Connection times out after 20 attempts
- Connection works correctly when Killswitch is set to OFF, and used to work with Killswitch ON

#### Extra details

```
Connection failed/timeout: au656.nordvpn.com.udp.ovpn
Please verify account details are correct.
If problems persist, please check the service status: "systemctl -l status dietpi-vpn"
```

systemctl -l status dietpi-vpn
```
...
Jun 08 21:22:34 DietPi64 openvpn[44197]: iptables-restore v1.8.7 (nf_tables): host/network `au656.nordvpn.com.udp.ovpn' not found
Jun 08 21:22:34 DietPi64 openvpn[44197]: Error occurred at line: 10
Jun 08 21:22:34 DietPi64 openvpn[44197]: Try `iptables-restore -h' or 'iptables-restore --help' for more information.
Jun 08 21:22:34 DietPi64 openvpn[44166]: WARNING: Failed running command (--up/--down): external program exited with error status: 2
Jun 08 21:22:34 DietPi64 openvpn[44166]: Exiting due to fatal error
Jun 08 21:22:34 DietPi64 systemd[1]: dietpi-vpn.service: Main process exited, code=exited, status=1/FAILURE
Jun 08 21:22:34 DietPi64 systemd[1]: dietpi-vpn.service: Failed with result 'exit-code'.
```

Joulinar, 2022-06-08:
Just for reference, same issue as this one on our forum https://dietpi.com/forum/t/dietpi-vpn-blocking-443/13348/13

Issue seems to be restoring `iptables` information

kinoushe, 2022-06-08:
Oh thank you! I'm not sure why I didn't check the forum first

kinoushe, 2022-06-09:
Just comparing the error messages, is it usual to see the grave accent (`) instead of apostrophes?

```
`au656.nordvpn.com.udp.ovpn'
...
Try `iptables-restore -h'
```

MichaIng, 2022-06-25:
Fixed with: https://github.com/MichaIng/DietPi/commit/70f7fa2

closed 2022-06-25 by MichaIng