MichaIng/DietPi #5554: Error installding docker on Thinker Board

sandvaer, 2022-06-08:
#### Details:
- Date           | Wed Jun  8 18:36:36 CEST 2022
- Bug report     | 55fd4c6d-4156-4901-ae3a-c8ae52f4592a
- DietPi version | v8.5.1 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Armbian
- Hardware       | ASUS Tinker Board (armv7l) (ID=52)
- Kernel version | `Linux DietPi 5.15.35-rockchip #22.05.1 SMP PREEMPT Sat May 28 08:18:16 UTC 2022 armv7l GNU/Linux`
- Distro         | bullseye (ID=6)
- Command        | `docker run -d -p 9002:9000 --name=portainer --restart=always -v /run/docker.sock:/var/run/docker.sock -v portainer_data:/data portainer/portainer-ce`
- Exit code      | 125
- Software title | DietPi-Software
#### Steps to reproduce:

Install portainer on Thinker Board using the dietpi-software

docker shoud start and protainer should start.
#### Actual behaviour:

Docker is not starting.
#### Extra details:

- ...
#### Additional logs:
```
Unable to find image 'portainer/portainer-ce:latest' locally
latest: Pulling from portainer/portainer-ce
772227786281: Pulling fs layer
96fd13befc87: Pulling fs layer
5b2faccdf576: Pulling fs layer
af0be5406135: Pulling fs layer
af0be5406135: Waiting
96fd13befc87: Verifying Checksum
96fd13befc87: Download complete
772227786281: Verifying Checksum
772227786281: Download complete
772227786281: Pull complete
af0be5406135: Verifying Checksum
96fd13befc87: Pull complete
5b2faccdf576: Verifying Checksum
5b2faccdf576: Download complete
5b2faccdf576: Pull complete
af0be5406135: Pull complete
Digest: sha256:52f9fdee1e4acfb1b5c4ddd15c88905287efb6e8f8058d2c5a2543ddc72e9dc0
Status: Downloaded newer image for portainer/portainer-ce:latest
e94d7ba5e3183c2553cbbbf24b184a306213f12c301fa2e24acd19c227e88f7f
docker: Error response from daemon: failed to create shim task: OCI runtime create failed: runc create failed: unable to start container process: error during container init: error setting cgroup config for procHooks process: bpf_prog_query(BPF_CGROUP_DEVICE) failed: function not implemented: unknown.
```

sandvaer, 2022-06-09:
I found solution here: https://github.com/MichaIng/DietPi/issues/4980

I added systemd.unified_cgroup_hierarchy=0 to /boot/armbianEnv.txt and rebooted, but it did not work.

Then I added systemd.unified_cgroup_hierarchy=0 to /boot/boot.cmd, ran: mkimage -C none -A arm -T script -d /boot/boot.cmd /boot/boot.scr and rebooted.

Now its working!

MichaIng, 2022-06-10:
That is strange, Linux 5.15 supports cgroupsv2 / unified cgroup hierarchy, so that shouldn't be required 🤔. Ah, now I read again the missing `BPF_CGROUP_DEVICE`. We need to report this to Armbian. Let me reopen the issue.

preethaml7, 2022-09-29:
Both the workarounds do not work for me and strangely the added parameter does not show up when cat /proc/cmdline even though I have tried the below: 

I added systemd.unified_cgroup_hierarchy=0 to /boot/armbianEnv.txt and rebooted, it did not work.

Then I tried addding systemd.unified_cgroup_hierarchy=0 to /boot/boot.cmd, ran: mkimage -C none -A arm -T script -d /boot/boot.cmd /boot/boot.scr and rebooted, and I still cannot find the added parameter in cat /proc/cmdline.

MichaIng, 2022-09-30:
`systemd.unified_cgroup_hierarchy=0` needs to be appended to the `extraarga=` line in `/etc/armbianEnv.txt` or `setenv bootargs` line in `/boot/boot.cmd`. It is no valid U-Boot environment variable but a kernel cmdline argument.

onorbe, 2022-10-08:
I have the same issue can someone explain steps to solve it.
Thanks in advance

MichaIng, 2022-10-08:
My last comment just above yours? 🙂

onorbe, 2022-10-08:
Yes but i try it and continúe with the sale issue.
Cant install portainer.

Joulinar, 2022-10-08:
What exactly you changed and what exactly is your error message

onorbe, 2022-10-08:
This is the error.

dietpi@DietPi:~$ sudo docker run -d -p 9000:9000 --name portainer --restar
t=always -v /var/run/docker.sock:/var/run/docker.sock -v portainer_data:/d
ata portainer/portainer-ce:latest
294629ec7a1a671f1f6e07fe958749ec3b67c3aa8902c6f992321e6871008677
docker: Error response from daemon: failed to create shim task: OCI runtim
e create failed: runc create failed: unable to start container process: er
ror during container init: error setting cgroup config for
procHooks proce
ss: bpf_prog_query(BPF_CGROUP_DEVICE) failed: function not
implemented: un
known.

Thanks again

preethaml7, 2022-10-08:
@onorbe ssh into your tinkerboard and edit the file below 
sudo nano /boot/boot.cmd
and update the existing line 
<artuuid} usb-storage.quirks=${usbstoragequirks} ${extraargs} ${extraboardargs}" 
to <artuuid} usb-storage.quirks=${usbstoragequirks} ${extraargs} ${extraboardargs} systemd.unified_cgroup_hierarchy=0"
and then run the command sudo mkimage -C none -A arm -T script -d /boot/boot.cmd /boot/boot.scr and reboot. Docker should be working now.

MichaIng, 2022-10-08:
Better go the path with `/etc/armbianEnv.txt`, which can be easier reverted, no compiling required, no chance to break anything:
```sh
sed -i '/^extraargs=/s/$/ systemd.unified_cgroup_hierarchy=0/' /etc/armbianEnv.txt
grep '^extraargs=' /etc/armbianEnv.txt
reboot
```

preethaml7, 2022-10-08:
> Better go the path with `/etc/armbianEnv.txt`, which can be easier reverted, no compiling required, no chance to break anything:
> 
> ```shell
> sed -i '/^extraargs=/s/$/ systemd.unified_cgroup_hierarchy=0/' /etc/armbianEnv.txt
> grep '^extraargs=' /etc/armbianEnv.txt
> reboot
> ```

I tried the above and it did not work for me.

MichaIng, 2022-10-08:
Then there must be something applied wrong. You can see the `${extraargs}` explicitly applied in the `/boot/boot.cmd` line and the `net.ifnames=0` applied the same way works as expected. Can you remove it from `/boot/boot.cmd`, apply it the above way and after reboot show the output of:
```sh
grep '^extraargs=' /etc/armbianEnv.txt
cat /proc/cmdline
```

onorbe, 2022-10-08:
> ```shell
> grep
> ```

Works perfect! You are a crack!

MichaIng, 2022-11-17:
Known to Armbian already, but no real fix known, just the workaround we know already: https://github.com/armbian/build/pull/3830

So I'll add this to our Tinker Board images by default.

MichaIng, 2022-11-18:
Solved with: https://github.com/MichaIng/DietPi/commit/e41cd73

MichaIng, 2022-11-18:
Better approach: https://github.com/MichaIng/DietPi/commit/3248247

closed 2022-11-18 by MichaIng