MichaIng/DietPi #5556: rpi4 unable to update sabnzb (dietpi-software reinstall 139)

josh3003, 2022-06-10:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=5
G_DIETPI_VERSION_RC=1
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
 buster 1

- Kernel version | `uname -a`
-Linux DietPi 5.10.103-v8+ #1529 SMP PREEMPT aarch64 GNU/Linux

- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
 RPi 4 Model B (aarch64)
- Power supply used | (EG: 5V 1A RAVpower)
 official rpi4
- SD card used | (EG: SanDisk ultra)
sandisk endurance

#### Additional Information (if applicable)
- Software title | Sabnzbd
- Was the software title installed freshly or updated/migrated?
-updated
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. Try to run dietpi-software reinstall 139
2. https://pastebin.com/JpKsBXa6 

#### Expected behaviour

Should update to the latest. 

#### Actual behaviour

/usr/lib/gcc/arm-linux-gnueabihf/8/include/arm_neon.h:6749:49: fatal error: You must enable NEON instructions (e.g. -mfloat-abi=softfp -mfpu=neon) to use these intrinsics.

#### Extra details

- tried switching between 32 bit and 64 bit kernels and updating to latest dietpi.

josh3003, 2022-06-10:
Seems to be affecting 32bit builds. No longer can install sabnzbd even through dietpi-software now. So now my softwares borked can't even run an reinstall

Joulinar, 2022-06-10:
left a comment on the sabnzbd issue you created as we well. It doesn't make sense to investigate the same case on multiple platforms.

josh3003, 2022-06-10:
> left a comment on the sabnzbd issue you created as we well. It doesn't make sense to investigate the same case on multiple platforms.

Too easy, wasn't sure if it was an issue with dietpi or sabyenc3. As you tested bullseye and I am on buster

Joulinar, 2022-06-13:
just an update on this one. Sabnzb guys will go to update their build script to fix the issue on Buster.

MichaIng, 2022-06-26:
Issue solved upstream 👍.

closed 2022-06-26 by MichaIng

Joulinar, 2022-06-26:
Not fully as still a new release of `SABnzbd` is required to pull the correct dependencys on Buster 😉

MichaIng, 2022-06-26:
Ah right, I see: https://github.com/sabnzbd/sabnzbd/pull/2209