MichaIng/DietPi #5561: Motioneye install fails (No module named 'pip')

Wixely, 2022-06-12:
#### Details:
- Date           | Sun Jun 12 16:54:11 BST 2022
- DietPi version | v8.5.1 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | from scratch
- Hardware       | RPi 3 Model B (aarch64) (ID=3)
- Kernel version | `Linux DietPi 5.15.32-v8+ #1538 SMP PREEMPT Thu Mar 31 19:40:39 BST 2022 aarch64 GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=0)
- Command        | `pip3 install --no-cache-dir -U https://github.com/motioneye-project/motioneye/archive/dev.tar.gz`
- Exit code      | 1
- Software title | DietPi-Software
#### Steps to reproduce:
1. Install motioneye on latest version of dietpi.
#### Expected behaviour:
Motioneye should install without errors
- ...
#### Actual behaviour:
Motioneye fails to install because 'pip3' is missing
- ...
#### Extra details:
I've tried apt-get install pip, which installs python3-pip but the install process for motioneye uninstalls it then throws the error. I've also used the change command feature and tried 'pip' instead which also fails.

Reference code: f41d8936-5828-46ca-8d80-1ca7ea7a4576
- ...
#### Additional logs:
```
Traceback (most recent call last):
  File "/usr/bin/pip3", line 5, in <module>
    from pip._internal.cli.main import main
ModuleNotFoundError: No module named 'pip'
```

MichaIng, 2022-06-12:
Many thanks for your report.

> /usr/bin/pip3

This is not the `pip3` executable that was intended to be installed, which would be `/usr/local/bin/pip3`. I remember another such case, but we couldn't find out or replicate how `get-pip.py` would install an invalid pip to an unintended location.

Just to verify:
```sh
ls -l /usr/bin/pip3
ls -l /use/local/bin/pip3
```
This first exists, while it shouldn't, the second does not exist, while it should?

Do you have any other Python software installed? Otherwise I suggest to purge everything around Python down to the roots and reinstall it:
```sh
apt purge --autoremove 'python3*' 'libpython3*'
rm -R /usr/local/lib/python3* /root/.cache/pip /usr/local/bin/pip*
dietpi-software reinstall 130
ls -l /usr/local/bin/pip3
```

Wixely, 2022-06-12:
Correct it exists in '''/usr/bin/pip3''' but doesn't exist in '''/use/local/bin/pip3'''.

I didn't realise python was a dietpi package on its own. The main purpose of this pi is for octoprint which I know uses python in some way. I'll make a backup and try your suggestion, thanks!

MichaIng, 2022-06-12:
Ah okay, you didn't install OctoPrint via `dietpi-software`? After removing all Python libraries and binaries, you'll need to reinstall OctoPrint either way, it's data however will be unaffected, so no worry on that.

Wixely, 2022-06-12:
I installed octoprint using dietpi-software but it was a long time ago on a very early version of dietpi 8.x.x
I updated to 8.5.1 today, I assume some change has happened with pip in the meantime that's caused this error.
Thanks for your advice, I'm in the middle of some prints but I will let you know if your suggestions work asap. Much appreciated

MichaIng, 2022-06-12:
> very early version of dietpi 8.x.x

v8.0 is just 5 months ago, nothing changed on the Python 3 implementation since then. We did install the `python3-pip` package a longer time ago, instead of using `get-pip.py` now. But also installing those in any other on top of each other, I wasn't able to replicate this strange issue, that `get-pip.py` fails to install a working pip at all 🤔.

Wixely, 2022-06-19:
Thanks @MichaIng your instructions worked. I backed up my octoprint config just to be sure and I'm glad I did. My config was set up for port 5000 but new default is 5001 so I uninstalled octoprint it thinking it didn't quite work and it removed all my settings. But all resolved now and I have my full setup back now. Motioneye installed flawlessly afterwards too. Thanks!

closed 2022-06-19 by Wixely

MichaIng, 2022-06-19:
Okay great, so indeed it looks like an issue with `python3-pip` + `get-pip.py` on top, probably even with a specific `pip` or `setuptools` version only, so that I wasn't able to replicate anymore. There have been quite some issues with those the last months, which have been solved in the meantime.

Yes port `5001` is used to avoid a conflict with Shairport Sync, which uses port `5000` already. An existing config file however shouldn't be touched on reinstall, so that port is only applied on fresh OctoPrint installs.