MichaIng/DietPi #5562: 2 copies of GMrender

KenMoreira, 2022-06-12:
After updating from 9 to bullseye I have 2 copies of GMrender, only one works.  It's causing issues where my streamer (allo usbridge) diet pi. Stops working and needs reboot.

[image]

MichaIng, 2022-06-19:
Were you able to solve it yourself? What was the issue? Probably we need to add a step to our Stretch => Buster or Buster => Bullseye upgrade guide.

KenMoreira, 2022-06-19:
I ran uninstall, and it uninstalled the newest copy which wasn't ever able to work.

Still unable to even find the version that is installed.
 In my Dir. I was hoping to update it. It's an old one

MichaIng, 2022-06-19:
Please check the following:
```sh
dpkg -l | grep gmediarender
command -v gmediarender gmrender
systemctl cat gmediarender
```

KenMoreira, 2022-06-19:
root@KensBridge:~# dpkg -l | grep -E 'gmediarender|gmrender'
root@KensBridge:~# command -v gmediarender gmrender
/usr/local/bin/gmediarender
root@KensBridge:~# systemctl cat gmediarender
No files found for gmediarender.service.
root@KensBridge:~# systemctl cat gmrender
# /etc/systemd/system/gmrender.service
[Unit]
Description=GMediaRender (DietPi)
Documentation=https://github.com/hzeller/gmrender-resurrect/blob/master/INSTALL.md#commandline-options
Wants=network-online.target
After=network-online.target sound.target dietpi-boot.service

[Service]
User=gmrender
ExecStartPre=+/bin/bash -c '. /boot/dietpi/func/dietpi-globals; systemctl set-environment ACTIVE_IP=$(G_GET_NET ip)'
ExecStart=/usr/local/bin/gmediarender -u 'ed987da7-8247-4ac6-ac0e-f58a33c89e02' -f 'KensBridge' --gstout-audiosink=al
sasink --gstout-audiodevice=default --logfile=stdout -I "$ACTIVE_IP"

[Install]
WantedBy=multi-user.target

MichaIng, 2022-06-19:
That looks all correct like the latest GMediaRender implementation. It doesn't look like you uninstalled GMediaRender via `dietpi-software`, else those files all shouldn't exist.

When you do `dietpi-software install 163`, how do you know that there are two instances, can you show the related logs? Might it be simply that the player still has the old instance ID/name in cache and you need to change the new service ExecStart options to restore the previous ID/name?

KenMoreira, 2022-06-19:
there WAS two copies , i uninstalled one on my own from google searching the other day

MichaIng, 2022-06-19:
Ah okay, I guess it was the `gmediarender` APT package then, which installs a service called `gmediarender.service`, but the two services fail due to network port conflict.

MichaIng, 2022-06-23:
I made an enhancement out of this issue, with the goal to change the service and user name on our implementation to "gmediarender" to align it with the Debian package. That way it wouldn't be possible anymore to have two conflicting services running.

MichaIng, 2022-10-16:
Implemented for DietPi v8.10, along with a general update and packaging of GMediaRender: #5811

closed 2022-10-16 by MichaIng