MichaIng/DietPi #5564: Error installing Jellyfin on ASUS Tinker Board

rgabbo, 2022-06-13:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
```
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=5
G_DIETPI_VERSION_RC=1
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
```
- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
`bullseye`
- Kernel version | `uname -a`
`Linux DietPi 5.15.35-rockchip #22.05.1 SMP PREEMPT Sat May 28 08:18:16 UTC 2022 armv7l GNU/Linux`
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
`ASUS Tinker Board (armv7l)`
- Power supply used | (EG: 5V 1A RAVpower)
5V 1A
- SD card used | (EG: SanDisk ultra)
Samsung

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
Jellyfin
- Was the software title installed freshly or updated/migrated?
Freshly
- Can this issue be replicated on a fresh installation of DietPi?
Yes

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. Freshly install Dietpi
2. At dietpi-software step choose Jellyfin

#### Expected behaviour

- Jellyfin should end installation fine and launch its web GUI

#### Actual behaviour

- I get this two errors in sequence and jellyfin service is not started:
[image]
[image]

#### Extra details

- ...

Joulinar, 2022-06-13:
We are already aware on this https://dietpi.com/forum/t/jellyfin-installation-error/13402

PR to fix it: #5559

MichaIng, 2022-06-25:
Issue solved in `dev`.

closed 2022-06-25 by MichaIng