MichaIng/DietPi #5565: Upgrade failed -> please help

omavoss, 2022-06-16:
#### Details:
- Date           | Thu Jun 16 22:01:39 CEST 2022
- DietPi version | v8.4.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | from scratch
- Hardware       | RPi 3 Model B (armv7l) (ID=3)
- Kernel version | `Linux RasPi-3 5.15.32-v7+ #1538 SMP Thu Mar 31 19:38:48 BST 2022 armv7l GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=0)
- Command        | `apt-get -y upgrade`
- Exit code      | 100
- Software title | DietPi-Update
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Reading package lists...
Building dependency tree...
Reading state information...
Calculating upgrade...
The following packages will be upgraded:
  firmware-atheros firmware-brcm80211 firmware-iwlwifi firmware-misc-nonfree
  firmware-realtek libpam-runtime tzdata
debconf: delaying package configuration, since apt-utils is not installed
7 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
1 not fully installed or removed.
Need to get 0 B/39.6 MB of archives.
After this operation, 21.5 kB of additional disk space will be used.
(Reading database ... 84373 files and directories currently installed.)
Preparing to unpack .../libpam-runtime_1.4.0-9+deb11u1+rpt2_all.deb ...
Unpacking libpam-runtime (1.4.0-9+deb11u1+rpt2) over (1.4.0-9+deb11u1+rpt1) ...
dpkg: error processing archive /var/cache/apt/archives/libpam-runtime_1.4.0-9+deb11u1+rpt2_all.deb (--unpack):
 unable to stat './usr/share/locale/bg/LC_MESSAGES' (which was about to be installed): Bad message
dpkg-deb: error: paste subprocess was killed by signal (Broken pipe)
Errors were encountered while processing:
 /var/cache/apt/archives/libpam-runtime_1.4.0-9+deb11u1+rpt2_all.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

AAS-Crypt, 2022-06-17:
Аre all package lists correct and up-to date also are the for Pi? Will you be able to try to upgrade every package seperately?

Joulinar, 2022-06-17:
You could try following as a first step

```
/boot/dietpi/func/dietpi-set_software apt-cache clean
apt update
apt upgrade
```

omavoss, 2022-06-17:
Hey Joulinar,
I executed this and after this I rebooted the Pi and dietpi-update was successful.
Many thanks und greetings.

closed 2022-06-17 by omavoss