MichaIng/DietPi #5566: Dietpi Crontab Purge after 80 hours of heavy load computing

AAS-Crypt, 2022-06-17:
### Creating a bug report/issue

#### Required Information
- DietPi version | 8.5.1
- Distro version | bullseye 0
- Kernel version | Linux DietPi 5.10.92-v8+ #1514 SMP PREEMPT Mon Jan 17 17:39:38 GMT 2022 aarch64 GNU/Linux
- SBC model | `RPi 4 Model B (aarch64)
- Power supply used | 5V 3.5A Samsung
- SD card used | DataTraveler 100 G3 64GB - USB stick

#### Steps to reproduce
To note I've been running on DietPi v8.2.1 after 2 days from issue I've updated Pi.
I've been running heavy MariaDB computing. After aprox. 80 hrs of work on morning Dietpi purged my cron jobs. On reboot `crontab -l` have been shown to be empty. So after that I thought that issue might be high temperature of Pi for 80 hours so after it cold down I rebooted Pi and was unable to retrieve my crontab list.

Is there anyone who have been exposed to similar issues where Dietpi would purge down?

Joulinar, 2022-06-19:
hmm usually there is nothing inside DietPi scripts that would purge a crontab. Maybe due to the heavy load, the base system was not able to complete some task and left the crontab in an undefined status? @MichaIng any ideas?

MichaIng, 2022-06-19:
Those per-user crontab's are stored in `/var/spool/cron/crontabs`. In theory, if there are I/O errors or the system crashes before the filesystem cache has been `sync`ed, it could be lost. Check for kernel errors:
```sh
dmesg -l emerg,alert,crit,err
```

AAS-Crypt, 2022-06-19:
Can't give new information as I updated kernel and dietpi. Next time definetily will give proper logs.

MichaIng, 2022-07-23:
Marking as closed. Feel free to reopen if the issue appears again.

closed 2022-07-23 by MichaIng