MichaIng/DietPi #5569: Can someone help me with MPD

KenMoreira, 2022-06-20:
Trying to play music from usb stick on ALLO USBRIDGE Dietpi (Bullseye) 

I know i need MPD and some software to control it.  My understanding Bullseye comes with MPD. I just need help getting it to read the the thumb drive as a drive. And then setting up a web based UI to be the controller.

KenMoreira, 2022-06-20:
I have done Nothing. I do not know what to do .

KenMoreira, 2022-06-20:
I just put two songs on a small drive to test it. If i can get it to work ill buy a 256Gb stick and then add music n  plug it in after

KenMoreira, 2022-06-20:
Ok i mounted the drive. in drive manager.
next step ? :D

MichaIng, 2022-06-26:
You have the [Allo USBridge](https://allo.com/sparky/usbridge.html) with Sparky SBC, or [Allo USBridge Sig](https://allo.com/sparky/usbridge-signature-pcb.html) with Raspberry Pi Compute Module 3+?

Those two can be ordered with DietPi and Allo GUI + audio software pre-installed indeed, which includes MPD and the O!MPD frontend. Brief instructions about how to access and use O!MPD can be found on our docs: https://dietpi.com/docs/software/media/#ompd

There are some other endpoints pre-installed, like the Roon Bridge, Shairport Sync as AirPlay backend, Squeezelite as Logitech Media Server/Squeezebox Server backend, DLNA and such.

And overview and some configuration can be also done with the Allo GUI itself at `http://dietpi.local/` with `admin@allo.com` username and `allo` or `dietpi` as password (not sure which version of Allo GUI it was shipped with 😉). More info here: https://dietpi.com/forum/t/dietpi-allo-com-web-gui-image/1523
_Ay, we need to port this to our docs!_

KenMoreira, 2022-06-26:
yes i have the allo usbridge signature.   I'll read that stuff to start

KenMoreira, 2022-06-26:
Im not sure how this works. I try doing various FTP with my dietpi's IP but they cannot connect.  Maybe i have to do something first? 

O!MPD[¶](https://dietpi.com/docs/software/media/#ompd)

trying to install that one first..

Joulinar, 2022-06-26:
You need to install FTP or Samba server to be able to transfer music.

MichaIng, 2022-06-26:
> You need to install FTP or Samba server

Or the NFS server. The Samba server however should be installed by default with the Allo GUI image. Was that really pre-installed on the SD card or did you flash a DietPi image manually? If so, then yes, those need to be installed manually. `dietpi-software` offers a large number of MPD frontends and other media software btw: https://dietpi.com/docs/software/media/

KenMoreira, 2022-06-26:
I Got my dietpi directly from allo , and then upgraded from here to bullseye.

KenMoreira, 2022-06-26:
i don't have access to allow GUI since upgrading though.

Joulinar, 2022-06-26:
Can share following

`dietpi-services status`

MichaIng, 2022-06-26:
Ah, in this case I suggest to flash our Bullseye image in the first place. Upgrading the distro can work but also can fail, especially with such a large number of installed software titles, and it isn't as clean as a fresh Bullseye installation for sure. We have an image with Allo GUI and everything pre-installed as well, even a 64-bit variant: https://dietpi.com/downloads/images/DietPi_RPi-ARMv8-Bullseye_AlloGUI.7z

KenMoreira, 2022-06-26:
root@KensBridge:~# dietpi-services status

 DietPi-Services
─────────────────────────────────────────────────────
 Mode: status

[  OK  ] DietPi-Services | avahi-daemon          active (running) since Fri 2022                                                                                                                                                             -06-24 17:55:05 BST; 1 day 22h ago
[  OK  ] DietPi-Services | nmbd                  active (running) since Fri 2022                                                                                                                                                             -06-24 17:55:05 BST; 1 day 22h ago
[  OK  ] DietPi-Services | smbd                  active (running) since Fri 2022                                                                                                                                                             -06-24 17:55:06 BST; 1 day 22h ago
[  OK  ] DietPi-Services | mariadb               active (running) since Fri 2022                                                                                                                                                             -06-24 17:55:08 BST; 1 day 22h ago
[FAILED] DietPi-Services | ● lighttpd.service - Lighttpd Daemon
     Loaded: loaded (/lib/systemd/system/lighttpd.service; disabled; vendor pres                                                                                                                                                             et: enabled)
     Active: failed (Result: exit-code) since Fri 2022-06-24 17:55:09 BST; 1 day                                                                                                                                                              22h ago
    Process: 704 ExecStartPre=/usr/sbin/lighttpd -tt -f /etc/lighttpd/lighttpd.c                                                                                                                                                             onf (code=exited, status=255/EXCEPTION)
        CPU: 81ms

Jun 24 17:55:09 KensBridge systemd[1]: lighttpd.service: Scheduled restart job,                                                                                                                                                              restart counter is at 5.
Jun 24 17:55:09 KensBridge systemd[1]: Stopped Lighttpd Daemon.
Jun 24 17:55:09 KensBridge systemd[1]: lighttpd.service: Start request repeated                                                                                                                                                              too quickly.
Jun 24 17:55:09 KensBridge systemd[1]: lighttpd.service: Failed with result 'exi                                                                                                                                                             t-code'.
Jun 24 17:55:09 KensBridge systemd[1]: Failed to start Lighttpd Daemon.
[  OK  ] DietPi-Services | mpd                   active (running) since Fri 2022                                                                                                                                                             -06-24 17:55:05 BST; 1 day 22h ago
[  OK  ] DietPi-Services | shairport-sync        active (running) since Fri 2022                                                                                                                                                             -06-24 17:55:08 BST; 1 day 22h ago
[  OK  ] DietPi-Services | squeezelite           active (running) since Fri 2022                                                                                                                                                             -06-24 17:55:08 BST; 1 day 22h ago
[  OK  ] DietPi-Services | gmrender              active (running) since Fri 2022                                                                                                                                                             -06-24 17:55:08 BST; 1 day 22h ago
[  OK  ] DietPi-Services | networkaudiod         active (running) since Fri 2022                                                                                                                                                             -06-24 17:55:09 BST; 1 day 22h ago
[  OK  ] DietPi-Services | roonbridge            active (running) since Fri 2022                                                                                                                                                             -06-24 17:55:09 BST; 1 day 22h ago
[ INFO ] DietPi-Services | logitechmediaserver   inactive (dead)
[ INFO ] DietPi-Services | noip2                 inactive (dead)
[  OK  ] DietPi-Services | netdata               active (running) since Fri 2022                                                                                                                                                             -06-24 17:55:11 BST; 1 day 22h ago
[  OK  ] DietPi-Services | cron                  active (running) since Fri 2022                                                                                                                                                             -06-24 17:55:11 BST; 1 day 22h ago
[  OK  ] DietPi-Services | ssh                   active (running) since Fri 2022                                                                                                                                                    [... cut ...]
[  OK  ] DietPi-Services | dietpi-preboot        active (exited) since Fri 2022-                                                                                                                                                             06-24 17:54:54 BST; 1 day 22h ago
[  OK  ] DietPi-Services | dietpi-postboot       active (exited) since Fri 2022-                                                                                                                                                             06-24 17:55:00 BST; 1 day 22h ago
[ INFO ] DietPi-Services | dietpi-wifi-monitor   inactive (dead)

Joulinar, 2022-06-26:
Ok web server is failing. Can you share as well

`/usr/sbin/lighttpd -tt -f /etc/lighttpd/lighttpd.conf`

KenMoreira, 2022-06-26:
i just flashed using that IMG u provided.  I am gonna login to SSH  . putty then we can take it from there ?

MichaIng, 2022-06-26:
Jep, also try again to connect to the Allo GUI, which should work now OOTB.

KenMoreira, 2022-06-26:
the GUI does work and it shows me the link to O!MPD in there

KenMoreira, 2022-06-26:
just runnig apt upgrade.  so now that everything is AS it should be lol.  Ill buy a larger thumb USD drive to host ALL my audio. 

however now we can actually get to work

KenMoreira, 2022-06-26:
Ok well now that i have all that setup . what do i do/ ( i will be heading out so ill follow up later)

MichaIng, 2022-06-26:
Attach the external drive via `dietpi-drive_manager`. I suggest to move DietPi userdata to this external drive as well, which can be done in `dietpi-drive_manager` as well. `/mnt/dietpi_userdata/Music` will then be pointing to the external drive. Then transfer music e.g. via Samba to this directory.

closed 2022-09-17 by MichaIng