MichaIng/DietPi #5570: help to uninstall docker

useronegit, 2022-06-20:
i use raspberrypi 3b+ with adguard home and i wanted to try searx with docker but icant uninstall docker for now

> │ APT purge for: docker-ce docker-ce-cli docker-engine                                                                 │
                                                          │  - Command: apt-get -y --allow-change-held-packages autopurge docker-ce docker-ce-cli docker-engine                  │
                                                          │  - Exit code: 100                                                                                                    │
                                                          │  - DietPi version: v8.5.1 (MichaIng/master) | HW_MODEL: 3 | HW_ARCH: 3 | DISTRO: 6                                   │
                                                          │  - Error log:                                                                                                        │
                                                          │ Reading package lists...                                                                                             │
                                                          │ Building dependency tree...                                                                                          │
                                                          │ Reading state information...                                                                                         │
                                                          │ The following packages will be REMOVED:                                                                              │
                                                          │   docker-ce* docker-ce-cli*                                                                                          │
                                                          │ 0 upgraded, 0 newly installed, 2 to remove and 0 not upgraded.                                                       │
                                                          │ After this operation, 232 MB disk space will be freed.                                                               │
                                                          │ (Reading database ... ^Mdpkg: warning: files list file for package 'bash' missing; assuming package has no files     │
                                                          │ currently installed^M                                                                                                │
                                                          │ dpkg: warning: files list file for package 'console-setup' missing; assuming package has no files currently          │
                                                          │ installed^M                                                                                                          │
                                                          │ dpkg: warning: files list file for package 'bsdextrautils' missing; assuming package has no files currently          │
                                                          │ installed^M                                                                                                          │
                                                          │ dpkg: warning: files list file for package 'dosfstools' missing; assuming package has no files currently installed^M │
                                                          │ dpkg: warning: files list file for package 'firmware-misc-nonfree' missing; assuming package has no files currently  │
                                                          │ installed^M                                                                                                          │
                                                          │ dpkg: warning: files list file for package 'libprotobuf-c1:arm64' missing; assuming package has no files currently   │
                                                          │ installed^M                                                                                                          │
                                                          │ dpkg: warning: files list file for package 'libsystemd0:arm64' missing; assuming package has no files currently      │
                                                          │ installed^M                                                                                                          │
                                                          │ dpkg: warning: files list file for package 'libcap2:arm64' missing; assuming package has no files currently          │
                                                          │ installed^M                                                                                                          │
                                                          │ dpkg: warning: files list file for package 'libkrb5support0:arm64' missing; assuming package has no files currently  │
                                                          │ installed^M                                                                                                          │
                                                          │ dpkg: warning: files list file for package 'perl' missing; assuming package has no files currently installed^M       │
                                                          │ dpkg: warning: files list file for package 'sysvinit-utils' missing; assuming package has no files currently         │
                                                          │ installed^M                                                                                                          │
                                                          │ dpkg: warning: files list file for package 'console-setup-linux' missing; assuming package has no files currently    │
                                                          │ installed^M                                                                                                          │
                                                          │ dpkg: warning: files list file for package 'libdebconfclient0:arm64' missing; assuming package has no files          │
                                                          │ currently installed^M                                                                                                │
                                                          │ dpkg: warning: files list file for package 'libsemanage1:arm64' missing; assuming package has no files currently     │
                                                          │ installed^M                                                                                                          │
                                                          │ dpkg: warning: files list file for package 'base-passwd' missing; assuming package has no files currently installed^M│
                                                          │ dpkg: warning: files list file for package 'gnupg-utils' missing; assuming package has no files currently installed^M│
                                                          │ (Reading database ... 5%^M(Reading database ... 10%^M(Reading database ... 15%^M(Reading database ... 20%^M(Reading  │
                                                          │ ... 25%^M(Reading database ... 30%^M(Reading database ... 35%^M(Reading database ... 40%^M(Reading database ... 45%^M│
                                                          │ database ... 50%^M(Reading database ... 55%^M(Reading database ... 60%^M(Reading database ... 65%^M(Reading database │
                                                          │ (Reading database ... 75%^M(Reading database ... 80%^M(Reading database ... 85%^M(Reading database ... 90%^M(Reading │
                                                          │ database ... 95%^Mdpkg: unrecoverable fatal error, aborting:^M                                                       │
                                                          │  files list file for package 'libnetfilter-conntrack3:arm64' is missing final newline^M                              │
                                                          │ E: Sub-process /usr/bin/dpkg returned an error code (2)

MichaIng, 2022-06-20:
File lists for a lot of packages are missing in your case, most likely due to a filesystem corruption. Did you have a system crash recently? Please check for kernel errors and do an `fsck` on reboot before we're going to try fixing it:
```sh
dmesg -l emerg,alert,crit,err
> /forcefsck
reboot
```
Do you have a `dietpi-backup` where you could restore those missing files from?

MichaIng, 2022-07-23:
Marking as closed. Feel free to reopen if required.

closed 2022-07-23 by MichaIng