MichaIng/DietPi #5573: Open Beta v8.6 | Please help testing and hardening the upcoming release

MichaIng, 2022-06-25:
RC version | v8.6.0
---------- | -------
v8.6.0 PR | #5572
Raw changelog | https://github.com/MichaIng/DietPi/blob/beta/CHANGELOG.txt
Code changes | https://github.com/MichaIng/DietPi/compare/master...beta
Release date | **2022-07-02**

### How to apply: https://github.com/MichaIng/DietPi/blob/master/BRANCH_SYSTEM.md

#### Related/solved issues: https://github.com/MichaIng/DietPi/issues?q=is%3Aissue+milestone%3Av8.6
____
### Beta v8.6.0
_(2022-06-25)_

#### New images
- Quartz64 | Support and images for the PINE64 Quartz64 series have been added to DietPi, modern feature-rich SBCs based on the new Rockchip RK3366 SoC.

#### New software
- Prometheus Node Exporter: A Prometheus exporter for hardware and OS metrics, written in Go, has been added to the DietPi software catalogue. Many thanks to @alexiri for implementing this software option: https://github.com/MichaIng/DietPi/pull/5551
- Tailscale: A new VPN server option has been added to the DietPi software catalogue. Many thanks to @goldfix for implementing this software option: https://github.com/MichaIng/DietPi/pull/5558
- Rclone: "rsync for cloud storage" has been added to the DietPi software catalogue. Many thanks to @goldfix for implementing this software option: https://github.com/MichaIng/DietPi/pull/5560
- ZeroTier: The free easy to deploy cloud-hosted VPN service has been added to the DietPi software catalogue. Many thanks to @AAS-Crypt for implementing this software option: https://github.com/MichaIng/DietPi/pull/5567

#### New features
- DietPi-Software | PiVPN: It is now possible to do an unattended install by placing a config file named "unattended_pivpn.conf" into the boot partition/directory. For example configs, have a look at https://github.com/pivpn/pivpn/tree/master/examples. Many thanks to @bastianpaetzold for implementing this feature: https://github.com/MichaIng/DietPi/pull/5546

#### Enhancements
- DietPi-Software | HAProxy: It received some updates and enhancements, using v2.6.0 sources now, enabled support for prometheus metrics, updated cipher suites, and the initial password for the admin panel is now the global software password instead of hardcoded "dietpi". All config file changes affect fresh installs only, while reinstalls will never touch the config file, but update the HAProxy binary only: "dietpi-software reinstall 98"

#### Bug fixes
- DietPi-VPN | Resolved an issue where NordVPN, IPVanish and PIA connections failed if the killswitch was enabled. Many thanks to `@jonyskids` for reporting this issue: https://dietpi.com/forum/t/dietpi-vpn-blocking-443/13348
- DietPi-LetsEncrypt | Resolved an issue where Lighttpd needed to be reloaded manually after certificate renewals to serve the new certificate. Many thanks to `@Rednox` for reporting this issue: https://dietpi.com/forum/t/letsencrypt-renew-cron-issue/13471/6
- DietPi-Software | Radarr: Resolved an issue where the installation on ARMv6 Raspberry Pi models (Raspberry Pi 1 and Zero (1)) failed, since Radarr v4 does not support Mono anymore. The latest v3 will be installed now on these models. Many thanks to @eddiermar for reporting this issue: https://github.com/MichaIng/DietPi/issues/5537
- DietPi-Software | Lidarr: Precautionary, on ARMv6 Raspberry Pi models, the latest v0.8 will be installed from now on, since the upcoming v1 won't support Mono anymore. 
- DietPi-Software | Jellyfin: Resolved an issue where the installation failed due to a missing config file. Many thanks to `@TomEighty15` for reporting this issue: https://dietpi.com/forum/t/jellyfin-installation-error/13402

alexiri, 2022-06-26:
The Prometheus metrics for HAProxy aren't working because they're not properly configured, #5574 should fix it.

By the way, I would also like to start exposing Prometheus metrics for other software, such as Pi-hole and Unbound. Those services will require external exporters, much like the base OS one. What would be the best way of doing it? Should it be done by default whenever Pi-hole (for example) is installed, only conditionally when the user installs both Prometheus Node Exporter and Pi-hole, or should it be an entirely separate software option (ie. "Prometheus Exporter for Pi-Hole")?

MichaIng, 2022-06-26:
Generally it's not an issue when admins need to actively configure additional features, instead of enabling each and every OOTB. In case of Prometheus metrics on HAProxy it's sufficiently cheap, behind authentication and duplicating that stats shown on the stats page already, so I'm fine with it.

For other software, however, I'd be against such, especially when it requires additional plugins to be installed. We may instead add it to our documentation, especially when we plan to add a Prometheus frontend.

alexiri, 2022-06-26:
So I understand that you wouldn't want the first two options, but having separate software options (ie. "Prometheus Exporter for Pi-Hole") that had to be installed explicitly would be ok?

Joulinar, 2022-06-26:
Better to document this on our online docs. Personally I don't think there is a huge demand for these Prometheus Exporter configurations.

MichaIng, 2022-06-27:
Btw, Quartz64 (A, B and SOQuartz) images can be found here: https://dietpi.com/downloads/images/

Check out the dedicated issue: https://github.com/MichaIng/DietPi/issues/4649
I wasn't able yet to successfully boot the model B, which is most likely due to a bad power cable: It has a different jack than the other two, bought a too cheap adapter cable, I think 😅. Would be great if someone could verify that it boots just as fine as model A and SOQuartz.

Joulinar, 2022-06-27:
There has been no change on Wireguard and it's configuration. According your error message, you have a local network issue to resolve hast name correctly. Please check your DNS settings.

GHRoss, 2022-06-27:
Not withstanding the effort that's gone in to making this work (I applaud you), I think #5546 should be re-thought. While I support making services easier for people to setup and configure, upon reading the release notes of this, I couldn't help but think that the configuration could enable someone who doesn't particularly know what they're doing to expose their device through their router in a way that doesn't just expose the VPN service port or encourage the use of certificate auth (in the instance of an exposed SSH port) or other measures to harden the device.

I'm more than happy to be wrong about this, and I'm sure someone who is far more intelligent than I will correct me (and I hope they do!). I suspect the answer will be "well, if they're going to install PiVPN manually, they'd probably get it wrong doing it that way too", then sobeit.

Again, I'd like to reiterate that I don't want to put a downer on the work that's been done to make this possible, just want to make sure people are as secure as they can be when exposing stuff to the internet. It might even be that DietPi covers this off somewhere else that I'm not aware of.

Sorry for the essay!

Joulinar, 2022-06-28:
I'm not 100% sure what you are trying say, but the linked pull request will not install any software automatically, nor will it open any ports to the internet on the user router. It's just an additional option to install the already available PiVPN software package, to be able to complete the installation without user input by placing a configuration file on a specific place. This would need to be done by the user manually. The configuration file contains the input that is needed to complete PiPVN installation without user intervention. Otherwise you would need to follow the interactive dialog. The result will be exactly the same, a running VPN server.

Anyway following your arguments, we would need to disable more than a hundert software options within our catalogue as nearly all of them could be accessed from internet if respective ports are opened by the user on his own router.

MichaIng, 2022-06-28:
Right, no ports are automatically opened at the router. That would be only possible via UPnP which I see more as security vulnerability than as feature and hence recommend everyone to disable UPnP in router settings, if available. And the PiVPN installer does not make use of it.

So the only automated thing is that the VPN server is listening on that port, same as with automated OpenVPN installation, but to access it, the client configs/keys need to be transferred to clients manually. And OpenVPN as well as WireGuard (both options in PiVPN) are well trusted applications and can hence be seen as secure on application level.

So this is really just an option for users to manually pre-create a config for automated deployment instead of manually going through installer dialogs.

Isildur981, 2022-06-30:
I just wanted to make sure that everyone was aware that Netdata has a Prometheus stats exporting function(along with several other exporters) built in.  It just has to be enabled.  I understand the use case for just a Prometheus exporter without all of the additional overhead that Netdata brings, but it would probably be good to add a note to Prometheus Node Exporter documentation that there is an alternative way to get Prometheus stats.  I'm sure there are people who just look at the Netdata charts and don't realize what other capabilities it has.  Just my 2 cents.  

[Export Netdata metrics](https://learn.netdata.cloud/docs/export/external-databases)

MichaIng, 2022-07-03:
### DietPi v8.6 has been released. Many thanks to all testers ❤️: #5591

closed 2022-07-03 by MichaIng