MichaIng/DietPi #5579: dietpi-config non reversible display resolution change

wubwubwubwubwubwubwubwub, 2022-06-28:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
- Kernel version | `uname -a`
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
- Power supply used | (EG: 5V 1A RAVpower)
- SD card used | (EG: SanDisk ultra)

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. ...
2. ...

#### Expected behaviour

- ...

#### Actual behaviour

- ...

#### Extra details

- ...

Joulinar, 2022-06-28:
What exactly is your issue. You basically did not provide any information nor filled the bug report.

wubwubwubwubwubwubwubwub, 2022-06-28:
I am not completely without understanding, but am truly without 
knowledge nor experience of GIT.  Bear with me, this answer needs context.

I had a problem, which seemed stable.  It started when I attempted to 
use dietpi-config to change my screen resolution. Turns out, for ODROID 
C4 this is not an option.  I then went to the "Advanced Options" item, 
simply out of curiosity.  This produced a black screen, no prompt.  I 
recovered with ^C and discovered my display resolution had dropped 
considerably.  I could not figure out how to revert to the original 
display resolution.

I decided this might be a bug, so I ran dietpi-bugreport, blithely 
assuming this was how one files a bug report.  I asked the system to 
generate a report, expecting to answer some questions, and was mildly 
horrified to discover that I had shipped a wad of system information to 
you all.  I had no intention to knocking on the door before I had done 
all my homework.

I did a very rudimentary search to attempt to locate an existing bug 
report and did not find one.  Then I tried to open a bug, and answer the 
questions about my system, and somewhere down the list I found that I 
should have tried to reproduce this before filing anything.  Sorry.

I did try to reproduce it on another ODROID C4, and did find the same 
behavior.  I guess one of my new C4s is mildly defective?

Also, as originator of a new, defective bug report, I could not figure 
out how to get GIT to let go of it.

Again sorry.  I will keep all my observations to myself.

I clearly have no business attempting to use the bug system for anything.

John.

On 6/28/22 09:31, Joulinar wrote:
> What exactly is your issue. You basically did not provide any information nor filled the bug report.
>

MichaIng, 2022-06-28:
No problem.

Are you trying to set the resolution for a desktop environment or for the plain console?

For console with nowadays used KMS it is not so trivial on anything else than Raspberry Pi or x86_64 systems. But there are generally ways, e.g.: https://wiki.archlinux.org/title/Kernel_mode_setting#Forcing_modes
Can you show the output of these command loop:
```bash
for i in /sys/class/drm/*; do [[ -e $i/status ]] && echo "$i: $(<$i/status)"; done
```

MichaIng, 2022-07-23:
I opened a new feature request to add support for DRM/KMS video modes in `dietpi-config` with mainline kernel: #5641

Marking this hence as closed.

closed 2022-07-23 by MichaIng