MichaIng/DietPi #5582: Unable to install LXDE desktop (DietPi has encountered an error)

Getafix17, 2022-06-29:
Hi please see log for the error I encountered:

#### Details:
- Date           | Tue Jun 28 10:16:13 IST 2022
- DietPi version | v8.5.1 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | from scratch
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | `Linux DietPiSSD 5.15.32-v8+ #1538 SMP PREEMPT Thu Mar 31 19:40:39 BST 2022 aarch64 GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=0)
- Command        | `ping -nc 1 -W 10 dns9.quad9.net`
- Exit code      | 2
- Software title | DietPi-Software
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
ping: dns9.quad9.net: Temporary failure in name resolution
```

Joulinar, 2022-06-29:
Hi,

this looks like a local network issue as you are not able to resolve Quad9 global DNS name.

Could you share following

```
ping -nc 1 -W 10 dns9.quad9.net
ping -nc 1 -W 10 google.com
```

Getafix17, 2022-06-29:
Hi thanks for the prompt response.
I am using Pi-hole with open DNS. Please see output.
root@DietPiSSD:~# ping -nc 1 -W 10 dns9.quad9.net
ping: dns9.quad9.net: Temporary failure in name resolution
root@DietPiSSD:~# ping -nc 1 -W 10 google.com
ping: google.com: Temporary failure in name resolution

Joulinar, 2022-06-29:
Something wrong with your DNS server settings on the DietPi network configuration. Can you share

```
cat /etc/resolv.conf
```

Getafix17, 2022-06-29:
root@DietPiSSD:~# cat /etc/resolv.conf
nameserver 192.168.0.100

Joulinar, 2022-06-29:
What IP is this device? DietPi or something else? Do you use DHCP or STATIC on DietPi?

Getafix17, 2022-06-29:
Static IP. 192.168.0.102

Joulinar, 2022-06-29:
and who is `192.168.0.100`

Getafix17, 2022-06-29:
Dietpi pi hole. on seperate machine is 192.168.0.100

Joulinar, 2022-06-29:
Do you see DNS request arriving/within the logs on the PiHole device?

Getafix17, 2022-06-29:
My apologies. I have another instance of Pi-hole on Docker @ 192.168.101. I disconnected Pi-hole 192.168.0.100 from my network. Pi-hole @ 192.168.0.101  is getting DNS request. There is another instace of Dietpi @ 192.168.0.115 that works perfectly. I managed to install LXDE just to test it out.

Getafix17, 2022-06-29:
Issue is with Diet-pi @192.168.0.102

Joulinar, 2022-06-29:
Well if you switch of DNS server on the system ending `.100`, you would need to adjust DNS settings on your system having issues as well. Have a look to network configuration on `.102` end ensure using the correct DNS server `.101`

Getafix17, 2022-06-29:
I changed the DNS to 192.168.0.101. System not responding. Cant SSH into the machine anymore. I must have done something wrong.

Joulinar, 2022-06-29:
Did you changed DNS server settings or STATIC IP address as well? Or did you switch to DHCP? Maybe you can connect keyboard+screen

Getafix17, 2022-06-29:
I will try to connect to a monitor. with Keybord and mouse.

closed 2022-06-29 by Joulinar