MichaIng/DietPi #5586: Portainer - upgrade without stopping docker

isarrider, 2022-07-02:
Hi,

as portainer isnt required to run docker and is deployed
as a container anyway, it shouldnt be an issue to leave docker
running and just upgrade portainer...

So I propose it is done like every container update...
If it is too much hassle to implement,
we could state it in the docu...

(because my understanding is that the container + image are removed and then
this is executed: https://github.com/MichaIng/DietPi/blob/master/dietpi/dietpi-software#L12261)

BR,
ALex

Joulinar, 2022-07-02:
This is working as expected as we stop all DietPi managed services during software installation and re-installation. However you could exclud the service from being DietPi controlled.

isarrider, 2022-07-02:
for docu purposes I out in here the 3 command lines to do it manually:

sudo docker container rm -f portainer
sudo docker image rm portainer/portainer-ce
sudo docker run -d -p '9002:9000' --name=portainer --restart=always -v '/run/docker.sock:/var/run/docker.sock' -v 'portainer_data:/data' 'portainer/portainer-ce'

documented & closed...

closed 2022-07-02 by isarrider

isarrider, 2022-07-29:
the official documentation suggests to install portainer like below...
https://docs.portainer.io/start/upgrade/docker

docker run -d -p 8000:8000 -p 9443:9443 --name=portainer --restart=always -v /var/run/docker.sock:/var/run/docker.sock -v portainer_data:/data portainer/portainer-ce:latest

Joulinar, 2022-07-29:
Yes and what are you trying to say with it? This is how we install it as well.

isarrider, 2022-07-29:
the way I see it different ports are specified...

Joulinar, 2022-07-29:
Yap as we use http. This is stated on the link your shared on one of the first red boxes.