MichaIng/DietPi #5594: fresh webmin install on fresh DietPi  install fails to start

alucard87pl, 2022-07-03:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version` 8.6.1
- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN` bullseye 0
- Kernel version | `uname -a`
- uname -a
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3) RPi4 B+ 4GB
- Power supply used | (EG: 5V 1A RAVpower) Official Rpi4 PSU
- SD card used | (EG: SanDisk ultra) 32GB Sandisk USB Drive, booting

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud) Webmin
- Was the software title installed freshly or updated/migrated? Fresh, same as DietPi
- Can this issue be replicated on a fresh installation of DietPi? Should be.

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. ... Install DietPi
2. ... Use `dietpi-sowftware` to install `webmin`

#### Expected behaviour

- ... `webmin` service starts

#### Actual behaviour

- ... `webmin` service fails to starts

#### Extra details

```
root@DietPi:~# systemctl status webmin
● webmin.service - Webmin (DietPi)
     Loaded: loaded (/etc/systemd/system/webmin.service; enabled; vendor preset: enabled)
    Drop-In: /etc/systemd/system/webmin.service.d
             └─dietpi-services_edit.conf
     Active: failed (Result: timeout) since Sun 2022-07-03 12:13:02 CEST; 3min 51s ago
       Docs: https://doxfer.webmin.com/Webmin/Main_Page
    Process: 9314 ExecStart=/etc/webmin/start (code=killed, signal=TERM)
        CPU: 8ms

lip 03 12:11:32 DietPi systemd[1]: Starting Webmin (DietPi)...
lip 03 12:13:02 DietPi systemd[1]: webmin.service: start operation timed out. Terminating.
lip 03 12:13:02 DietPi systemd[1]: webmin.service: Failed with result 'timeout'.
lip 03 12:13:02 DietPi systemd[1]: Failed to start Webmin (DietPi).
root@DietPi:~#
```

```
lip 03 12:11:32 DietPi systemd[1]: Starting Webmin (DietPi)...
░░ Subject: Rozpoczęto wykonywanie zadania uruchamiania dla jednostki webmin.service
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░
░░ Rozpoczęto wykonywanie zadania uruchamiania dla jednostki webmin.service.
░░
░░ Identyfikator zadania: 2703.
lip 03 12:13:02 DietPi systemd[1]: webmin.service: start operation timed out. Terminating.
lip 03 12:13:02 DietPi systemd[1]: webmin.service: Failed with result 'timeout'.
░░ Subject: Jednostka się nie powiodła
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░
░░ Jednostka webmin.service przeszła do stanu „failed” (niepowodzenia)
░░ z wynikiem „timeout”.
lip 03 12:13:02 DietPi systemd[1]: Failed to start Webmin (DietPi).
```

Joulinar, 2022-07-03:
can you share following

```
journalctl -u webmin
```

alucard87pl, 2022-07-03:
```
lip 03 11:33:21 DietPi webmin[5405]: Webmin starting
lip 03 11:33:22 DietPi systemd[1]: Started Webmin server daemon.
lip 03 11:33:23 DietPi systemd[1]: Stopping Webmin server daemon...
lip 03 11:33:23 DietPi systemd[1]: webmin.service: Succeeded.
lip 03 11:33:23 DietPi systemd[1]: Stopped Webmin server daemon.
lip 03 11:33:23 DietPi systemd[1]: webmin.service: Consumed 1.541s CPU time.
lip 03 11:33:28 DietPi systemd[1]: Starting Webmin (DietPi)...
lip 03 11:34:58 DietPi systemd[1]: webmin.service: start operation timed out. Terminating.
lip 03 11:34:58 DietPi systemd[1]: webmin.service: Failed with result 'timeout'.
lip 03 11:34:58 DietPi systemd[1]: Failed to start Webmin (DietPi).
lip 03 11:36:34 DietPi systemd[1]: Starting Webmin (DietPi)...
lip 03 11:38:04 DietPi systemd[1]: webmin.service: start operation timed out. Terminating.
lip 03 11:38:04 DietPi systemd[1]: webmin.service: Failed with result 'timeout'.
lip 03 11:38:04 DietPi systemd[1]: Failed to start Webmin (DietPi).
lip 03 11:44:35 DietPi systemd[1]: Starting Webmin (DietPi)...
lip 03 11:46:05 DietPi systemd[1]: webmin.service: start operation timed out. Terminating.
lip 03 11:46:05 DietPi systemd[1]: webmin.service: Failed with result 'timeout'.
lip 03 11:46:05 DietPi systemd[1]: Failed to start Webmin (DietPi).
lip 03 11:48:16 DietPi systemd[1]: Starting Webmin (DietPi)...
lip 03 11:49:46 DietPi systemd[1]: webmin.service: start operation timed out. Terminating.
lip 03 11:49:46 DietPi systemd[1]: webmin.service: Failed with result 'timeout'.
lip 03 11:49:46 DietPi systemd[1]: Failed to start Webmin (DietPi).
lip 03 11:51:07 DietPi systemd[1]: Starting Webmin (DietPi)...
lip 03 11:52:37 DietPi systemd[1]: webmin.service: start operation timed out. Terminating.
lip 03 11:52:37 DietPi systemd[1]: webmin.service: Failed with result 'timeout'.
lip 03 11:52:37 DietPi systemd[1]: Failed to start Webmin (DietPi).
lip 03 11:55:00 DietPi systemd[1]: Starting Webmin (DietPi)...
lip 03 11:56:30 DietPi systemd[1]: webmin.service: start operation timed out. Terminating.
lip 03 11:56:30 DietPi systemd[1]: webmin.service: Failed with result 'timeout'.
lip 03 11:56:30 DietPi systemd[1]: Failed to start Webmin (DietPi).
lip 03 12:04:21 DietPi systemd[1]: Starting Webmin (DietPi)...
lip 03 12:05:51 DietPi systemd[1]: webmin.service: start operation timed out. Terminating.
lip 03 12:05:51 DietPi systemd[1]: webmin.service: Failed with result 'timeout'.
lip 03 12:05:51 DietPi systemd[1]: Failed to start Webmin (DietPi).
lip 03 12:09:02 DietPi systemd[1]: Starting Webmin (DietPi)...
lip 03 12:10:32 DietPi systemd[1]: webmin.service: start operation timed out. Terminating.
lip 03 12:10:32 DietPi systemd[1]: webmin.service: Failed with result 'timeout'.
lip 03 12:10:32 DietPi systemd[1]: Failed to start Webmin (DietPi).
lip 03 12:11:32 DietPi systemd[1]: Starting Webmin (DietPi)...
lip 03 12:13:02 DietPi systemd[1]: webmin.service: start operation timed out. Terminating.
lip 03 12:13:02 DietPi systemd[1]: webmin.service: Failed with result 'timeout'.
lip 03 12:13:02 DietPi systemd[1]: Failed to start Webmin (DietPi).
```

alucard87pl, 2022-07-03:
Explanation:
These are a result of either a manual `systemctl reload webmin` or through `dietpi-services` which, I assume would be equivalent?

Joulinar, 2022-07-03:
ok we need to have a look. I'm not sure if there was a change on the way how webmin service is going to be started. But for now we seems to have some kind of loop as our service definition is trying to call `/etc/webmin/start`

```
root@DietPi4:~# systemctl cat webmin.service
# /etc/systemd/system/webmin.service
[Unit]
Description=Webmin (DietPi)
Documentation=https://doxfer.webmin.com/Webmin/Main_Page

[Service]
Type=forking
GuessMainPID=no
SyslogIdentifier=Webmin
ExecStart=/etc/webmin/start
ExecStop=/etc/webmin/stop
ExecReload=/etc/webmin/reload

[Install]
WantedBy=multi-user.target
```

which is again trying to start the service

```
root@DietPi4:~# cat /etc/webmin/start
#!/bin/sh
/usr/bin/systemctl start webmin
```

Joulinar, 2022-07-03:
as workaround try following

- stop the service

```
systemctl stop webmin.service
```

- copy following as a single block

```
cat << '_EOF_' > /etc/systemd/system/webmin.service
[Unit]
Description=Webmin server daemon

[Service]
#Environment="PERLLIB=WEBMIN_LIBDIR"
ExecStart=/usr/share/webmin/miniserv.pl /etc/webmin/miniserv.conf
ExecStop=/usr/bin/kill $MAINPID
ExecReload=/usr/bin/kill -USR1 $MAINPID
PIDFile=/var/webmin/miniserv.pid
Type=forking
Restart=always
RestartSec=2s
TimeoutSec=15s

[Install]
WantedBy=multi-user.target
_EOF_
```

- start the service

```
systemctl daemon-reload
systemctl start webmin.service
```

alucard87pl, 2022-07-03:
Worked like a charm, thank you :)

MichaIng, 2022-07-03:
Finally it has a native systemd service, that is great. But pretty strange that start/stop scripts are starting the service and not the other way round. If systemd is not installed, `/etc/webmin/start` is pointing to `/etc/webmin/.start-init` which starts `/usr/share/webmin/miniserv.pl` and `/etc/init.d/webmin` (the SysV service) also starts `/etc/webmin/.start-init`, so there is an inconsistency between service > script > binary (SysV) vs. script > service > binary (systemd), the letter being very uncommon. However, easy to fix our end.

Why doubling the systemd service?
```sh
rm /etc/systemd/system/webmin.service
systemctl daemon-reload
systemctl start webmin
```

Nasty that we did our release just yesterday. I'm thinking about how to fix this safely with a live patch.

Joulinar, 2022-07-03:
Let's do it If a live patch is possible

MichaIng, 2022-07-03:
PR up to fix for next release: #5595
Live patch will follow.

closed 2022-07-03 by MichaIng