MichaIng/DietPi #5597: Can't update to 8.6.1

andrew-jolley, 2022-07-04:
### Creating a bug report/issue

#### Required Information
- DietPi version:
` G_DIETPI_VERSION_CORE=8 `
` G_DIETPI_VERSION_SUB=5 `
` G_DIETPI_VERSION_RC=1 `
` G_GITBRANCH='master' `
` G_GITOWNER='MichaIng' `
- Distro version:
` bullseye 0 `
- Kernel version:
` Linux NextcloudPi 5.15.32-v8+ #1538 SMP PREEMPT Thu Mar 31 19:40:39 BST 2022 aarch64 GNU/Linux `
- SBC model:
` RPi 4 Model B (aarch64) `
- Power supply used:
5V, 3A
- SD card used:
SanDisk Ultra

#### Additional Information (if applicable)
- Software title:
Nextcloud
- Was the software title installed freshly or updated/migrated?
No
- Can this issue be replicated on a fresh installation of DietPi?
Haven't tried
- Bug report ID:
` 3bb6fa77-f8d0-4be5-b20a-cf7a80f459ea `

#### Steps to reproduce
1. Try to update from 8.5.1 to 8.6.1
2. DNS resolver failed

#### Expected behaviour
- The system should update

#### Actual behaviour
- The system can't connect to the server

#### Extra details
- Reboot doesn't help

Joulinar, 2022-07-04:
Usually DNS errors are indicating local network issue with name resolution. Can you share the error message pls. As well do you use some local DNS server inside your network?

andrew-jolley, 2022-07-04:
Yep, can ping 1.1.1.1 and 8.8.8.8. I can update now, so will close this issue

closed 2022-07-04 by andrew-jolley

Joulinar, 2022-07-04:
Did you changed anything?

andrew-jolley, 2022-07-04:
No, nothing.

Joulinar, 2022-07-04:
Maybe a temporary network/internet connection issues

andrew-jolley, 2022-07-04:
Highly possible