MichaIng/DietPi #5599: Sumologic on Dietpi

Alkhulaqi, 2022-07-04:
Hello 

I have a Raspberry Pi and an external hard drive, so I deseeded to create my cloud. I installed DietPi, and NextCloud and everything are working fine. Then, for security, I installed letsencrypt and  ClamAV. Lastly, I need to install a Sumologic collector to monitor the security logs of the DietPi. 

After installing the agent, I have to add its token. which is this code:
     ./collector.sh -q -Vsumo.token_and_url=[my.token]
It returns with this error
     ./collector.sh: 240: bin/unpack200: Exec format error
     Error unpacking jar files. The architecture or bitness (32/64)
     of the bundled JVM might not match your machine.

I did my research and found that unpack200 is not there. Is it possible to install unpack200 to dietpi?

MichaIng, 2022-07-05:
Are you sure the binary is not there? Seems like it is expected within the `bin` sub directory of where the `collector.sh` is located. How/where did you get this script from?

The error sounds like the binary is there but for the wrong architecture.

MichaIng, 2022-07-05:
Looks like it is generally not supported an RPi/ARM without using workarounds:
- https://support.sumologic.com/hc/en-us/articles/360034233293-Is-ARM64-architecture-supported-by-Sumo-Logic-collector-installer-
- https://support.sumologic.com/hc/en-us/community/posts/360025018314-Sumo-logic-collector-fails-to-start-on-rasspberry-PI

But it is hard to verify since I couldn't find downloads so far. But I think this is anyway better asked on Sumologic support.

Alkhulaqi, 2022-07-05:
> Are you sure the binary is not there? Seems like it is expected within the `bin` sub directory of where the `collector.sh` is located. How/where did you get this script from?
> 
> The error sounds like the binary is there but for the wrong architecture.

yes .. I am sure it is not there 
[image]

[image]

Alkhulaqi, 2022-07-05:
> Looks like it is generally not supported on RPi/ARM without using workarounds:
> 
> * https://support.sumologic.com/hc/en-us/articles/360034233293-Is-ARM64-architecture-supported-by-Sumo-Logic-collector-installer-
> * https://support.sumologic.com/hc/en-us/community/posts/360025018314-Sumo-logic-collector-fails-to-start-on-rasspberry-PI

will do. then I will post it here.
thank you

MichaIng, 2022-07-05:
You are looking at the wrong location (absolute vs relative path 😉). When you are in the directory of the `collector.sh`:
```sh
ls -l bin/unpack200
```

Alkhulaqi, 2022-07-28:
So the thing was that I am using 32bit OS and trying to install a 64bit agent... I changed the image to 64bit, but I had to configure the sumocollector manually. It worked fine at the beginning, then suddenly stopped.

Joulinar, 2022-07-28:
that's the related forum post I guess https://dietpi.com/forum/t/installing-sumologic-collector-in-dietpi/13696

Alkhulaqi, 2022-07-28:
this is my question there, I just finished all the configuration one more time and it is working fine

thank you

MichaIng, 2022-07-31:
Okay great, I'll mark this as closed then.

closed 2022-07-31 by MichaIng