MichaIng/DietPi #5601: Rclone will not install on raspberry pi b+

eddiermar, 2022-07-05:
### Creating a bug report/issue

#### Required 
- DietPi version | 8.6.1 
- SBC model | Raspberry pi b+

#### Additional Information (if applicable)
- Software title | Rclone

- Bug report ID | af2861c7-6caf-442a-b555-fbf26cc12607

#### Steps to reproduce

1. go to dietpi-software
2. try to install rclone
3. installation fails

#### Expected behaviour

- Rclone get installed

#### Actual behaviour

- I think the rclone installer is wrongly sermon because it tries to install the amd64 version.

#### Extra details

 
│ APT install for: ./rclone-v1.58.1-linux-amd64.deb
│  - Command: apt-get -y --allow-change-held-packages install
│ ./rclone-v1.58.1-linux-amd64.deb

MichaIng, 2022-07-05:
Many thanks for your report.

Indeed a bug our end with a false/missing condition statement. Sadly difficult to push a live patch for this.

For now:
```bash
cd /tmp
curl -fL 'https://github.com/rclone/rclone/releases/download/v1.58.1/rclone-v1.58.1-linux-arm.deb' -o rclone.deb
apt install ./rclone.deb
rm rclone.deb
```

MichaIng, 2022-07-07:
Fixed with: https://github.com/MichaIng/DietPi/commit/bacf34c

MichaIng, 2022-07-23:
Fix can be tested in beta: #5643

closed 2022-07-23 by MichaIng