MichaIng/DietPi #5602: Raspotify broken on ARMv6 devices (Pi 1 A/B/Zero 1.x)

dvelluto, 2022-07-05:
### Creating a bug report/issue

#### Required Information
- DietPi version | 
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=6
G_DIETPI_VERSION_RC=1
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='applied'

- Distro version | 
bullseye 1
- Kernel version |
Linux Cheetah 5.15.32+ #1538 Thu Mar 31 19:37:58 BST 2022 armv6l GNU/Linux
- SBC model |
RPi B+ (armv6l)

#### Additional Information (if applicable)
- Software title | Raspotify
- Was the software title installed freshly or updated/migrated?
updated
- Can this issue be replicated on a fresh installation of DietPi?
- Yes

#### Steps to reproduce
On Raspberry PI 1 (A/B) / Pi Zero 1.x /  ARMv6
1. update/install latest dietpi (since 8)
2. install/update raspotify package

#### Expected behaviour
Service should work

#### Actual behaviour
Service cannot start

#### Extra details
For these old images the version of Raspotify should not be updated as it has dropped support.
On Raspotify README it's stated:
https://github.com/dtcooper/raspotify#support-for-armv6-pi-v1-and-pi-zero-v1x-has-been-dropped

doing apt purge raspotify 
and following the manual steps from the Raspotify repo fix things.

Joulinar, 2022-07-05:
Thx for the report. We would need to adjust our install script to respect this.

@MichaIng 
I will do the change. One more that dropped support on ARMv6 😢

MichaIng, 2022-07-05:
Solved for v8.7, including a patch on DietPi update for ARMv6 to downgrade Raspotify to last supported version and have the repo removed.

MichaIng, 2022-07-23:
Fix can be tested in beta: #5643

closed 2022-07-23 by MichaIng