MichaIng/DietPi #5605: Multiple Issues with services after setting up Organizr (and PHP update to fix Organizr issue)

ZiggyPi, 2022-07-06:
@MichaIng @Joulinar 

Hey Guys,

I recently came back from a vacation and came to the conclusion that setting up a reverse proxy in lighttpd for sonarr, radarr etc. would be a good idea.
As i was setting this up and got it to work, i thought to myself "wouldn't it be nice the have Organizr running to have everything nicely accessable". Well that is when the problems started. Organizr website gave the error : Your Composer dependencies require a PHP version ">= 7.4.0". So i did some searching on the internet en installed PHP7.4. And after that everything went belly up.

So i have managed to get some stuff working again after some removes, reinstalls and installs.
I have made a matrix of the issues i still have now.

i am running dietpi 8.6.1 (buster), PHP7.3, PHP7.2-fpm, Lighttpd 1.4.53 (ssl)

Hope you guys can help me out with getting everything working again.
If you can help me with getting Organizr to work, that would be a great plus.

```
|	Service	 |IP:port Result				|IP/service Result												|
-----------------|----------------------------------------------|---------------------------------------------------------------------------------------------------------------|
|Sonarr		 |Perfect					|Perfect													|
|Radarr		 |Perfect					|Perfect													|
|Lidarr		 |Perfect					|Perfect													|
|Deluge		 |Perfect					|No Such Resource, No such child resource									|
|Jackett	 |Perfect					|Perfect													|
|Plex		 |Perfect					|401 Unauthorized												|
|Webmin ssl=1	 |Perfect					|ERROR - DOCUMENT FOLLOWS - This web server is running in SSL mode. Try the URL https://localhost:10000/ instead|
|Webmin ssl=0	 |This site can't provide a secure connection	|Basic text, no graphics, after Login : 404 Not Found								|
|HTPC Manager	 |Can't be reached				|Perfect													|
|Dietpi Dashboard|Perfect					|Blank Page													|
|Ubooquity	 |Perfect					|Basic text, no graphics											|
|Ubooquity Admin |Perfect					|HTTP ERROR 500 - Could not find matching provider (worng URL)					
|Nextcloud	 |-						|404 Not Found													|
```

Joulinar, 2022-07-06:
> And i rather would wait for a dietpi-update to bullseye

DietPi will never update your system automatically, not will it be a feature of `dietpi-update`. Something you need to do yourself. That's why we create the blog post nearly a year ago.

MichaIng, 2022-07-06:
> wait for a dietpi-update to bullseye

DietPi updates will never do Debian distro upgrades, that will always need to be done manually.

The guide includes some reinstalls, including PHP and webservers, which should fix the Nextcloud issue, if my assumption above regarding this is correct.

Everything else as said is a matter of configuring the backend software (e.g. a base path) and the proxy correctly. The distro upgrade won't fix this.

ZiggyPi, 2022-07-06:
@MichaIng @Joulinar Thanks for the input.
I will try and update to bullseye this week and do what i sad to switch from lighttpd to nginx. And see if that solves al my problems.

I will keep you guys posted

ZiggyPi, 2022-07-07:
@MichaIng @Joulinar 

I am currently running the guide to update buster to bullseye. I have got some error messages, but nothing that stopped the process untill "apt full-upgrade".

This gives me the following error message.
```
root@DietPi-HTPC:/# apt full-upgrade
Reading package lists... Done
Building dependency tree
Reading state information... Done
Calculating upgrade... Error!
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libc6-dev : Breaks: libgcc-8-dev (< 8.4.0-2~) but 8.3.0-6+rpi1 is to be installed
E: Error, pkgProblemResolver::Resolve generated breaks, this may be caused by held packages
```

Joulinar, 2022-07-07:
did you followed all steps in section 2.2.4? Did you have done following before, as it is quite important to detect the correct OS version Bullseye, e.g. PHP7.4

```
/boot/dietpi/func/dietpi-obtain_hw_model
. /boot/dietpi/func/dietpi-globals
```

For SAMBA you might need to install WSDD service https://github.com/MichaIng/DietPi/discussions/5244

Joulinar, 2022-07-07:
Could you run following

```
echo $G_DISTRO_NAME
rm /boot/dietpi/.hw_model
/boot/dietpi/func/dietpi-obtain_hw_model
. /boot/dietpi/func/dietpi-globals
cat /boot/dietpi/.hw_model
```

ZiggyPi, 2022-07-07:
echo $G_DISTRO_NAME >> buster

after all the commands

```
root@DietPi-HTPC:~# echo $G_DISTRO_NAME
buster
root@DietPi-HTPC:~# rm /boot/dietpi/.hw_model
root@DietPi-HTPC:~# /boot/dietpi/func/dietpi-obtain_hw_model
root@DietPi-HTPC:~# . /boot/dietpi/func/dietpi-globals
root@DietPi-HTPC:~# cat /boot/dietpi/.hw_model
G_HW_MODEL=4
G_HW_MODEL_NAME='RPi 4 Model B (armv7l)'
G_HW_ARCH=2
G_HW_ARCH_NAME='armv7l'
G_HW_CPUID=0
G_HW_CPU_CORES=4
G_DISTRO=5
G_DISTRO_NAME='buster'
G_ROOTFS_DEV='/dev/mmcblk0p2'
G_HW_UUID='64c1c99e-b114-42e0-b03b-b44edfcf0ea5'
G_RASPBIAN=1
G_HW_ONBOARD_WIFI=1
G_HW_REVISION='c03112'
G_HW_PCB_REVISION=2
G_HW_MEMORY_SIZE=4096
G_HW_MANUFACTURER='Sony UK'
```
so nothing changed

ZiggyPi, 2022-07-07:
@Joulinar reboot doesn't change a thing

Joulinar, 2022-07-07:
Looks like the update did not completed fully. @MichaIng any ideas?

ZiggyPi, 2022-07-07:
<details><summary>cat /etc/debian_version /etc/os-release</summary>
<p>

```
root@DietPi-HTPC:~# cat /etc/debian_version /etc/os-release
10.2
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

```
</p>
</details>

>Why did you remove /var/lib/mysql?

I found this on the internet to solve the problem i was having with ggc-8-base

>rm -R /var/lib/mysql # in case there was a directory recreated
ln -s /mnt/dietpi_userdata/mysql /var/lib/mysql

Okay did this, so that should be fine again

<details><summary>apt full-upgrade</summary>
<p>

```
root@DietPi-HTPC:~# apt full-upgrade
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Calculating upgrade... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

```
</p>
</details>

ZiggyPi, 2022-07-07:
<details><summary>apt install --reinstall base-files</summary>
<p>

```
root@DietPi-HTPC:~# apt install --reinstall base-files
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
0 upgraded, 0 newly installed, 1 reinstalled, 0 to remove and 0 not upgraded.
Need to get 70.3 kB of archives.
After this operation, 0 B of additional disk space will be used.
Get:1 https://mirror.netcologne.de/raspbian/raspbian bullseye/main armhf base-files armhf 11.1+rpi1+deb11u3 [70.3 kB]
Fetched 70.3 kB in 0s (221 kB/s)
(Reading database ... 135611 files and directories currently installed.)
Preparing to unpack .../base-files_11.1+rpi1+deb11u3_armhf.deb ...
Unpacking base-files (11.1+rpi1+deb11u3) over (11.1+rpi1+deb11u3) ...
Setting up base-files (11.1+rpi1+deb11u3) ...
Processing triggers for install-info (6.7.0.dfsg.2-6) ...

```
</p>
</details>

MichaIng, 2022-07-07:
Can you show:
```sh
ls -al /etc/debian_release.dpkg-{dist,new}
```

ZiggyPi, 2022-07-07:
@MichaIng 

>ls -al /etc/debian_release.dpkg-{dist,new}

root@DietPi-HTPC:~# ls -al /etc/debian_release.dpkg-{dist,new}
ls: cannot access '/etc/debian_release.dpkg-dist': No such file or directory
ls: cannot access '/etc/debian_release.dpkg-new': No such file or directory

MichaIng, 2022-07-07:
Okay, no idea why it is not updated, let's do it manually:
```sh
echo '11.3' > /etc/debian_release
```

ZiggyPi, 2022-07-07:
@MichaIng 

is this the outcome you expected and is this the outcome to pickup at 2.2.4?

```
root@DietPi-HTPC:~# echo '11.3' > /etc/debian_release
root@DietPi-HTPC:~# echo '11.3' > /etc/debian_release
root@DietPi-HTPC:~# /boot/dietpi/func/dietpi-obtain_hw_model
root@DietPi-HTPC:~# . /boot/dietpi/func/dietpi-globals
root@DietPi-HTPC:~# echo $G_DISTRO_NAME
buster
```

ZiggyPi, 2022-07-07:
@MichaIng 

Okay now it's returning "bullseye".
Gonna run the guide from 2.2.4 now.

Wish me luck

ZiggyPi, 2022-07-07:
@MichaIng 
Reinstall is still running, but now i can see PHP7.4 lines going down the terminal window, so that's a good sign :+1:

ZiggyPi, 2022-07-07:
@MichaIng 

Okay so theses issues are still unresolved : 

- Samba server (can see it now, but no access to folders)
- Deluge (This site can't be reached)
- Webmin ( This site can't provide a secure connection)
- HTPC Manager ( This site can't be reached)
- Nextcloud (Internal Server Error)

Organizr website is now not returning an error, but is keeps loading (turning loading wheel)

Joulinar, 2022-07-07:
can you share following

`dietpi-services status`

ZiggyPi, 2022-07-07:
@Joulinar 
<details><summary>dietpi-services status</summary>
<p>

```
root@DietPi-HTPC:~# dietpi-services status

 DietPi-Services
─────────────────────────────────────────────────────
 Mode: status

[  OK  ] DietPi-Services | nmbd                  active (running) since Thu 2022-07-07 23:05:27 CEST; 9min ago
[  OK  ] DietPi-Services | smbd                  active (running) since Thu 2022-07-07 23:05:27 CEST; 9min ago
[  OK  ] DietPi-Services | nfs-kernel-server     active (exited) since Thu 2022-07-07 23:05:29 CEST; 9min ago
[  OK  ] DietPi-Services | redis-server          active (running) since Thu 2022-07-07 23:05:29 CEST; 9min ago
[  OK  ] DietPi-Services | mariadb               active (running) since Thu 2022-07-07 23:05:30 CEST; 9min ago
[  OK  ] DietPi-Services | nginx                 active (running) since Thu 2022-07-07 23:05:31 CEST; 9min ago
[  OK  ] DietPi-Services | deluged               active (running) since Thu 2022-07-07 23:05:31 CEST; 9min ago
[  OK  ] DietPi-Services | plexmediaserver       active (running) since Thu 2022-07-07 23:05:31 CEST; 9min ago
[  OK  ] DietPi-Services | ubooquity             active (running) since Thu 2022-07-07 23:05:31 CEST; 9min ago
[  OK  ] DietPi-Services | jackett               active (running) since Thu 2022-07-07 23:05:31 CEST; 9min ago
[  OK  ] DietPi-Services | sonarr                active (running) since Thu 2022-07-07 23:05:31 CEST; 9min ago
[  OK  ] DietPi-Services | radarr                active (running) since Thu 2022-07-07 23:05:31 CEST; 9min ago
[  OK  ] DietPi-Services | lidarr                active (running) since Thu 2022-07-07 23:05:31 CEST; 9min ago
[ INFO ] DietPi-Services | deluge-web            inactive (dead)
[  OK  ] DietPi-Services | rpimonitor            active (running) since Thu 2022-07-07 23:05:34 CEST; 9min ago
[  OK  ] DietPi-Services | webmin                active (running) since Thu 2022-07-07 23:05:41 CEST; 9min ago
[FAILED] DietPi-Services | ● htpc-manager.service - HTPC Manager (DietPi)
     Loaded: loaded (/etc/systemd/system/htpc-manager.service; disabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Thu 2022-07-07 23:05:42 CEST; 9min ago
    Process: 961 ExecStart=/usr/bin/python3 -OO /mnt/dietpi_userdata/htpc-manager/Htpc.py (code=exited, status=1/FAILURE)
   Main PID: 961 (code=exited, status=1/FAILURE)
        CPU: 416ms

Jul 07 23:05:42 DietPi-HTPC HTPC Manager[961]: Traceback (most recent call last):
Jul 07 23:05:42 DietPi-HTPC HTPC Manager[961]:   File "/mnt/dietpi_userdata/htpc-manager/Htpc.py", line 254, in <module>
Jul 07 23:05:42 DietPi-HTPC HTPC Manager[961]:     main()
Jul 07 23:05:42 DietPi-HTPC HTPC Manager[961]:   File "/mnt/dietpi_userdata/htpc-manager/Htpc.py", line 172, in main
Jul 07 23:05:42 DietPi-HTPC HTPC Manager[961]:     from htpc.settings import Settings
Jul 07 23:05:42 DietPi-HTPC HTPC Manager[961]:   File "/mnt/dietpi_userdata/htpc-manager/htpc/settings.py", line 7, in <module>
Jul 07 23:05:42 DietPi-HTPC HTPC Manager[961]:     import cherrypy
Jul 07 23:05:42 DietPi-HTPC HTPC Manager[961]: ModuleNotFoundError: No module named 'cherrypy'
Jul 07 23:05:42 DietPi-HTPC systemd[1]: htpc-manager.service: Main process exited, code=exited, status=1/FAILURE
Jul 07 23:05:42 DietPi-HTPC systemd[1]: htpc-manager.service: Failed with result 'exit-code'.
[  OK  ] DietPi-Services | docker                active (running) since Thu 2022-07-07 23:05:41 CEST; 9min ago
[  OK  ] DietPi-Services | cron                  active (running) since Thu 2022-07-07 23:05:41 CEST; 9min ago
[  OK  ] DietPi-Services | dropbear              active (running) since Thu 2022-07-07 23:05:24 CEST; 9min ago
[ INFO ] DietPi-Services | dietpi-vpn            inactive (dead)
[ INFO ] DietPi-Services | dietpi-cloudshell     inactive (dead)
[  OK  ] DietPi-Services | dietpi-dashboard      active (running) since Thu 2022-07-07 23:05:24 CEST; 9min ago
[ INFO ] DietPi-Services | dietpi-ramlog         inactive (dead)
[  OK  ] DietPi-Services | dietpi-preboot        active (exited) since Thu 2022-07-07 23:05:24 CEST; 9min ago
[  OK  ] DietPi-Services | dietpi-postboot       active (exited) since Thu 2022-07-07 23:05:24 CEST; 9min ago
[ INFO ] DietPi-Services | dietpi-wifi-monitor   inactive (dead)

```
</p>
</details>

MichaIng, 2022-07-07:
Ah, HTPC-Manager needs to be reinstalled as well (like most Python software due to Python 3.7 => 3.9 upgrade):
```sh
dietpi-software reinstall 155
```

ZiggyPi, 2022-07-07:
<details><summary>dietpi-services status after reinstall</summary>
<p>

```
root@DietPi-HTPC:~# dietpi-services status

 DietPi-Services
─────────────────────────────────────────────────────
 Mode: status

[  OK  ] DietPi-Services | nmbd                  active (running) since Thu 2022-07-07 23:25:07 CEST; 1min 37s ago
[  OK  ] DietPi-Services | smbd                  active (running) since Thu 2022-07-07 23:25:07 CEST; 1min 36s ago
[  OK  ] DietPi-Services | nfs-kernel-server     active (exited) since Thu 2022-07-07 23:25:09 CEST; 1min 35s ago
[  OK  ] DietPi-Services | redis-server          active (running) since Thu 2022-07-07 23:25:09 CEST; 1min 35s ago
[  OK  ] DietPi-Services | mariadb               active (running) since Thu 2022-07-07 23:25:10 CEST; 1min 33s ago
[  OK  ] DietPi-Services | nginx                 active (running) since Thu 2022-07-07 23:25:11 CEST; 1min 33s ago
[  OK  ] DietPi-Services | deluged               active (running) since Thu 2022-07-07 23:25:11 CEST; 1min 33s ago
[  OK  ] DietPi-Services | plexmediaserver       active (running) since Thu 2022-07-07 23:25:11 CEST; 1min 33s ago
[  OK  ] DietPi-Services | ubooquity             active (running) since Thu 2022-07-07 23:25:11 CEST; 1min 33s ago
[  OK  ] DietPi-Services | jackett               active (running) since Thu 2022-07-07 23:25:11 CEST; 1min 33s ago
[  OK  ] DietPi-Services | sonarr                active (running) since Thu 2022-07-07 23:25:11 CEST; 1min 33s ago
[  OK  ] DietPi-Services | radarr                active (running) since Thu 2022-07-07 23:25:11 CEST; 1min 33s ago
[  OK  ] DietPi-Services | lidarr                active (running) since Thu 2022-07-07 23:25:11 CEST; 1min 33s ago
[ INFO ] DietPi-Services | deluge-web            inactive (dead)
[  OK  ] DietPi-Services | rpimonitor            active (running) since Thu 2022-07-07 23:25:13 CEST; 1min 31s ago
[  OK  ] DietPi-Services | webmin                active (running) since Thu 2022-07-07 23:25:18 CEST; 1min 26s ago
[  OK  ] DietPi-Services | htpc-manager          active (running) since Thu 2022-07-07 23:25:18 CEST; 1min 26s ago
[  OK  ] DietPi-Services | docker                active (running) since Thu 2022-07-07 23:25:18 CEST; 1min 26s ago
[  OK  ] DietPi-Services | cron                  active (running) since Thu 2022-07-07 23:25:18 CEST; 1min 26s ago
[  OK  ] DietPi-Services | dropbear              active (running) since Thu 2022-07-07 23:05:24 CEST; 21min ago
[ INFO ] DietPi-Services | dietpi-vpn            inactive (dead)
[ INFO ] DietPi-Services | dietpi-cloudshell     inactive (dead)
[  OK  ] DietPi-Services | dietpi-dashboard      active (running) since Thu 2022-07-07 23:05:24 CEST; 21min ago
[ INFO ] DietPi-Services | dietpi-ramlog         inactive (dead)
[  OK  ] DietPi-Services | dietpi-preboot        active (exited) since Thu 2022-07-07 23:05:24 CEST; 21min ago
[  OK  ] DietPi-Services | dietpi-postboot       active (exited) since Thu 2022-07-07 23:05:24 CEST; 21min ago
[ INFO ] DietPi-Services | dietpi-wifi-monitor   inactive (dead)

```
</p>
</details>

<details><summary>ss -tulpn | grep LISTEN</summary>
<p>

```

root@DietPi-HTPC:~# ss -tulpn | grep LISTEN
tcp   LISTEN 0      64                              0.0.0.0:2049       0.0.0.0:*                                                                                                      
tcp   LISTEN 0      4096                            0.0.0.0:33347      0.0.0.0:*    users:(("rpc.mountd",pid=5868,fd=17))                                                             
tcp   LISTEN 0      4096                            0.0.0.0:60195      0.0.0.0:*    users:(("rpc.mountd",pid=5868,fd=13))                                                             
tcp   LISTEN 0      128                             0.0.0.0:5252       0.0.0.0:*    users:(("dietpi-dashboar",pid=503,fd=6))                                                          
tcp   LISTEN 0      5                        10.0.0.30%eth0:54405      0.0.0.0:*    users:(("deluged",pid=6016,fd=13))                                                                
tcp   LISTEN 0      5                          127.0.0.1%lo:54405      0.0.0.0:*    users:(("deluged",pid=6016,fd=11))                                                                
tcp   LISTEN 0      80                            127.0.0.1:3306       0.0.0.0:*    users:(("mariadbd",pid=5953,fd=19))                                                               
tcp   LISTEN 0      511                           127.0.0.1:6379       0.0.0.0:*    users:(("redis-server",pid=5891,fd=7))                                                            
tcp   LISTEN 0      50                              0.0.0.0:139        0.0.0.0:*    users:(("smbd",pid=5852,fd=45))                                                                   
tcp   LISTEN 0      128                           127.0.0.1:39341      0.0.0.0:*    users:(("Plex Script Hos",pid=6389,fd=4))                                                         
tcp   LISTEN 0      128                           127.0.0.1:44077      0.0.0.0:*    users:(("Plex Script Hos",pid=6184,fd=7))                                                         
tcp   LISTEN 0      5                            172.17.0.1:5357       0.0.0.0:*    users:(("python3",pid=511,fd=18))                                                                 
[... cut ...]
tcp   LISTEN 0      512                                   *:9117             *:*    users:(("jackett",pid=6076,fd=206))                                                               
tcp   LISTEN 0      2                                     *:3389             *:*    users:(("xrdp",pid=530,fd=11))                                                                    
tcp   LISTEN 0      50                                 [::]:445           [::]:*    users:(("smbd",pid=5852,fd=42))  

```
</p>
</details>

ZiggyPi, 2022-07-07:
<details><summary>journalctl -u deluged</summary>
<p>

```
root@DietPi-HTPC:~# journalctl -u deluged
-- Journal begins at Thu 2022-07-07 23:17:02 CEST, ends at Thu 2022-07-07 23:35:01 CEST. --
Jul 07 23:20:20 DietPi-HTPC systemd[1]: Stopping Deluge Daemon (DietPi)...
Jul 07 23:20:20 DietPi-HTPC systemd[1]: deluged.service: Succeeded.
Jul 07 23:20:20 DietPi-HTPC systemd[1]: Stopped Deluge Daemon (DietPi).
Jul 07 23:20:20 DietPi-HTPC systemd[1]: deluged.service: Consumed 2.198s CPU time.
Jul 07 23:25:11 DietPi-HTPC systemd[1]: Started Deluge Daemon (DietPi).
root@DietPi-HTPC:~#

```
</p>
</details>

ZiggyPi, 2022-07-07:
>systemctl start deluge-web

This didn't help

<details><summary>journalctl -u deluged-web</summary>
<p>

```
root@DietPi-HTPC:~# journalctl -u deluge-web
-- Journal begins at Thu 2022-07-07 23:17:02 CEST, ends at Thu 2022-07-07 23:37:13 CEST. --
Jul 07 23:25:11 DietPi-HTPC systemd[1]: Started Deluge Web UI (DietPi).
Jul 07 23:25:15 DietPi-HTPC systemd[1]: deluge-web.service: Succeeded.
Jul 07 23:25:15 DietPi-HTPC systemd[1]: deluge-web.service: Consumed 1.643s CPU time.
Jul 07 23:37:12 DietPi-HTPC systemd[1]: Started Deluge Web UI (DietPi).
Jul 07 23:37:13 DietPi-HTPC systemd[1]: deluge-web.service: Succeeded.
Jul 07 23:37:13 DietPi-HTPC systemd[1]: deluge-web.service: Consumed 1.089s CPU time.
root@DietPi-HTPC:~#

```
</p>
</details>

MichaIng, 2022-07-07:
Ah, Deluge also needs to be reinstalled
```sh
sed -i '/^ExecStart=/deluge-web$/deluge-web -d/' /etc/systemd/system/deluge-web.service
systemctl daemon-reload
systemctl restart deluge-web
```

MichaIng, 2022-07-07:
sed -i '/^ExecStart=/s/deluge-web$/deluge-web -d/' /etc/systemd/system/deluge-web.service
systemctl daemon-reload
systemctl restart deluge-web

ZiggyPi, 2022-07-07:
@MichaIng  Okay deluge is working again :+1:

MichaIng, 2022-07-07:
There seems to be no PHP FPM server. Did you reinstall PHP to have PHP7.4?
```sh
dietpi-software reinstall 85 89
```
And check HTPC Manager logs:
```sh
journalctl -u htpc-manager
```

MichaIng, 2022-07-07:
Please update the cherrypy module:
```sh
pip3 install -U cherrypy
systemctl restart htpc-manager
```
I forgot that you setup a proxy, I hope the reinstall did not broke the Nginx proxy config somehow.

ZiggyPi, 2022-07-07:
>pip3 install -U cherrypy
systemctl restart htpc-manager

This didn't fix the problem with HTPC-Manager

And the proxy i had setup was with Lighttpd, but i uninstalled that, and switched to Nginx. Still have to setup the proxy and ssl if everything works again.

ZiggyPi, 2022-07-07:
Okay but i didnt changr anything in the htpc-manager itself for the proxy.

And i have reinstalled nextcloud after i switched to nginx

MichaIng, 2022-07-08:
Hmm it would have made much sense to change the HTPC host to 127.0.0.1 in case of a local proxy. The default is 0.0.0.0 and can be set via database only. I'll check how to change it manually with `sqlite` this evening.

About Nextcloud, is the database symlink now correct?
```sh
readlink -f /var/lib/mysql
```

ZiggyPi, 2022-07-08:
```
root@DietPi-HTPC:~# readlink -f /var/lib/mysql
/mnt/dietpi_userdata/mysql
```

MichaIng, 2022-07-08:
Do webserver or PHP logs say something?
```sh
journalctl -u nginx -u php7.4-fpm
cat /var/log/nginx/error.log /var/log/php7.4-fpm.log
```

ZiggyPi, 2022-07-08:
```
root@DietPi-HTPC:~# journalctl -u nginx -u php7.4-fpm
-- Journal begins at Fri 2022-07-08 07:17:02 CEST, ends at Fri 2022-07-08 07:54:10 CEST. --
-- No entries --
```
```
root@DietPi-HTPC:~# cat /var/log/nginx/error.log /var/log/php7.4-fpm.log
root@DietPi-HTPC:~#
```

ZiggyPi, 2022-07-08:
Got a new issue too

```
2022-07-08` 07:21:06.0|Warn|ImportApprovedEpisodes|Couldn't import episode /HTPC/Downloads/Complete/The.Boys.S03E08.1080p.WEB.H264-GLHF[rarbg]/the.boys.s03e08.1080p.web.h264-glhf.mkv

[v3.0.8.1507] System.UnauthorizedAccessException: Access to the path '/HTPC/Downloads/Complete/The.Boys.S03E08.1080p.WEB.H264-GLHF[rarbg]/the.boys.s03e08.1080p.web.h264-glhf.mkv' is denied. ---> System.IO.IOException: Permission denied
   --- End of inner exception stack trace ---
  at System.IO.FileSystem.DeleteFile (System.String fullPath) [0x0006d] in <de882a77e7c14f8ba5d298093dde82b2>:0 
  at System.IO.File.Delete (System.String path) [0x00014] in <de882a77e7c14f8ba5d298093dde82b2>:0 
  at NzbDrone.Mono.Disk.DiskProvider.TransferFilePatched (System.String source, System.String destination, System.Boolean overwrite, System.Boolean move) [0x00291] in <2da24ae1fe6e4b0189edfb4791d8f9a5>:0 
  at NzbDrone.Mono.Disk.DiskProvider.MoveFileInternal (System.String source, System.String destination) [0x00098] in <2da24ae1fe6e4b0189edfb4791d8f9a5>:0 
  at NzbDrone.Common.Disk.DiskProviderBase.MoveFile (System.String source, System.String destination, System.Boolean overwrite) [0x000e1] in <bb4bd9a0099f48de84455bcef8e9c0ae>:0 
  at NzbDrone.Common.Disk.DiskTransferService.TryMoveFileVerified (System.String sourcePath, System.String targetPath, System.Int64 originalSize) [0x00047] in <bb4bd9a0099f48de84455bcef8e9c0ae>:0 
  at NzbDrone.Common.Disk.DiskTransferService.TransferFile (System.String sourcePath, System.String targetPath, NzbDrone.Common.Disk.TransferMode mode, System.Boolean overwrite) [0x004b9] in <bb4bd9a0099f48de84455bcef8e9c0ae>:0 
  at NzbDrone.Core.MediaFiles.EpisodeFileMovingService.TransferFile (NzbDrone.Core.MediaFiles.EpisodeFile episodeFile, NzbDrone.Core.Tv.Series series, System.Collections.Generic.List`1[T] episodes, System.String destinationFilePath, NzbDrone.Common.Disk.TransferMode mode) [0x00129] in <f308258701c9443d9f77fbbaa937a672>:0 
  at NzbDrone.Core.MediaFiles.EpisodeFileMovingService.MoveEpisodeFile (NzbDrone.Core.MediaFiles.EpisodeFile episodeFile, NzbDrone.Core.Parser.Model.LocalEpisode localEpisode) [0x00054] in <f308258701c9443d9f77fbbaa937a672>:0 
  at NzbDrone.Core.MediaFiles.UpgradeMediaFileService.UpgradeEpisodeFile (NzbDrone.Core.MediaFiles.EpisodeFile episodeFile, NzbDrone.Core.Parser.Model.LocalEpisode localEpisode, System.Boolean copyOnly) [0x001b0] in <f308258701c9443d9f77fbbaa937a672>:0 
  at NzbDrone.Core.MediaFiles.EpisodeImport.ImportApprovedEpisodes.Import (System.Collections.Generic.List`1[T] decisions, System.Boolean newDownload, NzbDrone.Core.Download.DownloadClientItem downloadClientItem, NzbDrone.Core.MediaFiles.EpisodeImport.ImportMode importMode) [0x0029b] in <f308258701c9443d9f77fbbaa937a672>:0 

2022-07-08 07:29:45.9|Info|RssSyncService|Starting RSS Sync
2022-07-08 07:29:57.8|Info|DownloadDecisionMaker|Processing 380 releases
2022-07-08 07:30:03.9|Info|RssSyncService|RSS Sync Completed. Reports found: 380, Reports grabbed: 0
2022-07-08 07:31:55.9|Warn|ImportApprovedEpisodes|Couldn't import episode /HTPC/Downloads/Complete/The.Boys.S03E08.1080p.WEB.H264-GLHF[rarbg]/the.boys.s03e08.1080p.web.h264-glhf.mkv

[v3.0.8.1507] System.UnauthorizedAccessException: Access to the path '/HTPC/Downloads/Complete/The.Boys.S03E08.1080p.WEB.H264-GLHF[rarbg]/the.boys.s03e08.1080p.web.h264-glhf.mkv' is denied. ---> System.IO.IOException: Permission denied
   --- End of inner exception stack trace ---
  at System.IO.FileSystem.DeleteFile (System.String fullPath) [0x0006d] in <de882a77e7c14f8ba5d298093dde82b2>:0 
  at System.IO.File.Delete (System.String path) [0x00014] in <de882a77e7c14f8ba5d298093dde82b2>:0 
  at NzbDrone.Mono.Disk.DiskProvider.TransferFilePatched (System.String source, System.String destination, System.Boolean overwrite, System.Boolean move) [0x00291] in <2da24ae1fe6e4b0189edfb4791d8f9a5>:0 
  at NzbDrone.Mono.Disk.DiskProvider.MoveFileInternal (System.String source, System.String destination) [0x00098] in <2da24ae1fe6e4b0189edfb4791d8f9a5>:0 
  at NzbDrone.Common.Disk.DiskProviderBase.MoveFile (System.String source, System.String destination, System.Boolean overwrite) [0x000e1] in <bb4bd9a0099f48de84455bcef8e9c0ae>:0 
  at NzbDrone.Common.Disk.DiskTransferService.TryMoveFileVerified (System.String sourcePath, System.String targetPath, System.Int64 originalSize) [0x00047] in <bb4bd9a0099f48de84455bcef8e9c0ae>:0 
  at NzbDrone.Common.Disk.DiskTransferService.TransferFile (System.String sourcePath, System.String targetPath, NzbDrone.Common.Disk.TransferMode mode, System.Boolean overwrite) [0x004b9] in <bb4bd9a0099f48de84455bcef8e9c0ae>:0 
  at NzbDrone.Core.MediaFiles.EpisodeFileMovingService.TransferFile (NzbDrone.Core.MediaFiles.EpisodeFile episodeFile, NzbDrone.Core.Tv.Series series, System.Collections.Generic.List`1[T] episodes, System.String destinationFilePath, NzbDrone.Common.Disk.TransferMode mode) [0x00129] in <f308258701c9443d9f77fbbaa937a672>:0 
  at NzbDrone.Core.MediaFiles.EpisodeFileMovingService.MoveEpisodeFile (NzbDrone.Core.MediaFiles.EpisodeFile episodeFile, NzbDrone.Core.Parser.Model.LocalEpisode localEpisode) [0x00054] in <f308258701c9443d9f77fbbaa937a672>:0 
  at NzbDrone.Core.MediaFiles.UpgradeMediaFileService.UpgradeEpisodeFile (NzbDrone.Core.MediaFiles.EpisodeFile episodeFile, NzbDrone.Core.Parser.Model.LocalEpisode localEpisode, System.Boolean copyOnly) [0x001b0] in <f308258701c9443d9f77fbbaa937a672>:0 
[... cut ...]
  at NzbDrone.Core.MediaFiles.UpgradeMediaFileService.UpgradeEpisodeFile (NzbDrone.Core.MediaFiles.EpisodeFile episodeFile, NzbDrone.Core.Parser.Model.LocalEpisode localEpisode, System.Boolean copyOnly) [0x001b0] in <f308258701c9443d9f77fbbaa937a672>:0 
  at NzbDrone.Core.MediaFiles.EpisodeImport.ImportApprovedEpisodes.Import (System.Collections.Generic.List`1[T] decisions, System.Boolean newDownload, NzbDrone.Core.Download.DownloadClientItem downloadClientItem, NzbDrone.Core.MediaFiles.EpisodeImport.ImportMode importMode) [0x0029b] in <f308258701c9443d9f77fbbaa937a672>:0 
```

MichaIng, 2022-07-08:
That was directly after you tried to access Nextcloud (log files are cleared hourly with RAM log)?

What are the last Nextcloud logs?
```bash
tail -5 /mnt/dietpi_userdata/nextcloud_data/nextcloud.log
```

Sonarr has no write permissions for that directory. It's limited to /mnt and /media dirs. But check our docs about how to add access to a custom dir: https://dietpi.com/docs/software/bittorrent/#sonarr

ZiggyPi, 2022-07-08:
Ran it again after tried to open nextcloud

```
root@DietPi-HTPC:~# journalctl -u nginx -u php7.4-fpm
-- Journal begins at Fri 2022-07-08 07:17:02 CEST, ends at Fri 2022-07-08 08:09:08 CEST. --
-- No entries --
root@DietPi-HTPC:~# cat /var/log/nginx/error.log /var/log/php7.4-fpm.log
2022/07/08 08:08:58 [crit] 16064#16064: *70 connect() to unix:/run/php/php7.4-fpm.sock failed (2: No such file or directory) while connecting to upstream, client: 10.0.0.15, server: dietpi-htpc, request: "GET /nextcloud/ HTTP/1.1", upstream: "fastcgi://unix:/run/php/php7.4-fpm.sock:", host: "10.0.0.30"
```

How can sonarr not have access after the update? It was working before

MichaIng, 2022-07-08:
If you didn't edit the Sonarr service, the completed downloads dir and didn't move that download directory, it should never had access.

Is the PHP server running?
```bash
systemctl status php7.4-fpm
```

ZiggyPi, 2022-07-08:
It should start download in HTPC/downloads/incomplete/ then move to HTPC/downloads/complete where sonarr should copy it to HTPC/series. And that is how it always worked.

```
root@DietPi-HTPC:~# cat /var/log/nginx/error.log /var/log/php7.4-fpm.log
2022/07/08 08:08:58 [crit] 16064#16064: *70 connect() to unix:/run/php/php7.4-fpm.sock failed (2: No such file or directory) while connecting to upstream, client: 10.0.0.15, server: dietpi-htpc, request: "GET /nextcloud/ HTTP/1.1", upstream: "fastcgi://unix:/run/php/php7.4-fpm.sock:", host: "10.0.0.30"
root@DietPi-HTPC:~# systemctl status php7.4-fpm
● php7.4-fpm.service - The PHP 7.4 FastCGI Process Manager
     Loaded: loaded (/lib/systemd/system/php7.4-fpm.service; enabled; vendor preset: enabled)
    Drop-In: /etc/systemd/system/php7.4-fpm.service.d
             └─dietpi-process_tool.conf
     Active: inactive (dead) since Fri 2022-07-08 00:18:22 CEST; 7h ago
       Docs: man:php-fpm7.4(8)
   Main PID: 507 (code=exited, status=0/SUCCESS)
     Status: "Processes active: 0, idle: 4, Requests: 5, slow: 0, Traffic: 0req/sec"
        CPU: 1.573s

Warning: journal has been rotated since unit was started, output may be incomplete.
```

MichaIng, 2022-07-08:
Start PHP:
```bash
systemctl start php7.4-fpm
sleep 3
systemctl status php7.4-fpm
```
The Bullseye upgrade shouldn't touch any permissions. But let's check:
```bash
systemctl cat htpc-manager
ls -dl /HTPC/downloads/{in,}complete
```

ZiggyPi, 2022-07-08:
```
root@DietPi-HTPC:~# systemctl start php7.4-fpm
root@DietPi-HTPC:~# sleep 3
root@DietPi-HTPC:~# systemctl status php7.4-fpm
● php7.4-fpm.service - The PHP 7.4 FastCGI Process Manager
     Loaded: loaded (/lib/systemd/system/php7.4-fpm.service; enabled; vendor preset: enabled)
    Drop-In: /etc/systemd/system/php7.4-fpm.service.d
             └─dietpi-process_tool.conf
     Active: active (running) since Fri 2022-07-08 08:30:24 CEST; 5s ago
       Docs: man:php-fpm7.4(8)
    Process: 30715 ExecStartPost=/usr/lib/php/php-fpm-socket-helper install /run/php/php-fpm.sock /etc/php/7.4/fpm/pool.d/www.conf 74 (code=exited, status=0/SUCCESS)
   Main PID: 30710 (php-fpm7.4)
     Status: "Ready to handle connections"
      Tasks: 5 (limit: 4915)
        CPU: 247ms
     CGroup: /system.slice/php7.4-fpm.service
             ├─30710 php-fpm: master process (/etc/php/7.4/fpm/php-fpm.conf)
             ├─30711 php-fpm: pool www
             ├─30712 php-fpm: pool www
             ├─30713 php-fpm: pool www
             └─30714 php-fpm: pool www

Jul 08 08:30:23 DietPi-HTPC systemd[1]: Starting The PHP 7.4 FastCGI Process Manager...
Jul 08 08:30:24 DietPi-HTPC systemd[1]: Started The PHP 7.4 FastCGI Process Manager.
```
Things are getting weirder and weirder

```
root@DietPi-HTPC:~# ls -dl /HTPC/downloads/{in,}complete
ls: cannot access '/HTPC/downloads/incomplete': No such file or directory
ls: cannot access '/HTPC/downloads/complete': No such file or directory
```

MichaIng, 2022-07-08:
Then you need to check where those have gone. Are/were those local directories or an external drive?

ZiggyPi, 2022-07-08:
These were on an external drive.
I have noticed something weird. During our session yesterday night i checked sonarr and it status, and it was showing my external drive.... 245Gb free from 1.8TB. but now if i look.... it's no longer in the list.

<details><summary>systemctl cat sonarr</summary>
<p>

```

root@DietPi-HTPC:~# systemctl cat sonarr
# /etc/systemd/system/sonarr.service
[Unit]
Description=Sonarr Daemon (DietPi)
Wants=network-online.target
After=network-online.target dietpi-boot.service

[Service]
SyslogIdentifier=Sonarr
User=sonarr
UMask=002
ExecStart=/usr/bin/mono /usr/lib/sonarr/bin/Sonarr.exe -nobrowser -data=/mnt/dietpi_userdata/sonarr
TimeoutStopSec=20
KillMode=process
Restart=on-failure

# Hardening
ProtectSystem=strict
ProtectHome=true
PrivateDevices=true
ProtectKernelTunables=true
ProtectControlGroups=true
ReadWritePaths=/usr/lib/sonarr /mnt /media /var/log/sonarr /tmp

[Install]
WantedBy=multi-user.target

# /etc/systemd/system/sonarr.service.d/dietpi-services_edit.conf
[Unit]
#Description=Sonarr Daemon (DietPi)
#Wants=network-online.target
#After=network-online.target dietpi-boot.service

[Service]
#SyslogIdentifier=Sonarr
#User=sonarr
#Group=dietpi
#UMask=002
#ExecStart=/usr/bin/mono /usr/lib/sonarr/bin/Sonarr.exe -nobrowser -data=/mnt/dietpi_userdata/sonarr
#TimeoutStopSec=20
#KillMode=process
#Restart=on-failure

# Hardening
#ProtectSystem=strict
#ProtectHome=true
#PrivateDevices=true
#ProtectKernelTunables=true
#ProtectControlGroups=true
#ReadWritePaths=/usr/lib/sonarr /mnt /media /var/log/sonarr /tmp

[Install]
#WantedBy=multi-user.target

```
</p>
</details>

ZiggyPi, 2022-07-09:
@MichaIng 
A quick update.
Sonarr is working correctly again. Don't what happened here. The only thing i did was a reboot

ZiggyPi, 2022-07-10:
@MichaIng 
Another update.
Samba is working again after i hashed out "widelinks = yes" in the samba conf file. 

Apparently there is a samba-vfs-modules missing after the buster -> bullseye upgrade. And cant install it because i get an error that the file cant be found on the repo.

Joulinar, 2022-07-10:
can you share the exact error?

ZiggyPi, 2022-07-10:
@Joulinar 

```
root@DietPi-HTPC:~# sudo apt install samba-vfs-modules
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following additional packages will be installed:
  liburing1
Recommended packages:
  libcephfs2 libgfapi0
The following NEW packages will be installed:
  liburing1 samba-vfs-modules
0 upgraded, 2 newly installed, 0 to remove and 0 not upgraded.
Need to get 458 kB/466 kB of archives.
After this operation, 1,549 kB of additional disk space will be used.
Do you want to continue? [Y/n] y
Err:1 https://mirror.netcologne.de/raspbian/raspbian bullseye/main armhf samba-vfs-modules armhf 2:4.13.13+dfsg-1~deb11u3
  404  Not Found [IP: 194.8.197.22 443]
E: Failed to fetch https://mirror.netcologne.de/raspbian/raspbian/pool/main/s/samba/samba-vfs-modules_4.13.13%2bdfsg-1%7edeb11u3_armhf.deb  404  Not Found [IP: 194.8.197.22 443]
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
root@DietPi-HTPC:~#
```

ZiggyPi, 2022-07-10:
And i dont know if this is a problem.
But i get text messages running throught eachother

[image]

Joulinar, 2022-07-10:
maybe a temp issue of the apt package mirror? Because I see the file being available at https://mirror.netcologne.de/raspbian/raspbian/pool/main/s/samba/

Does following is working

```
cd /tmp
wget https://mirror.netcologne.de/raspbian/raspbian/pool/main/s/samba/samba-vfs-modules_4.13.13%2Bdfsg-1~deb11u4_armhf.deb
apt install ./samba-vfs-modules_4.13.13+dfsg-1~deb11u4_armhf.deb
rm samba-vfs-modules_4.13.13+dfsg-1~deb11u4_armhf.deb
```

Joulinar, 2022-07-10:
> But i get text messages running through eachother.
This is just cosmetic. Somehow text overlapping on the login page. Can be ignored.

`dietpi-banner 1`

ZiggyPi, 2022-07-10:
Get a new error on the samba-vfs-modules install
"broken packages"

```
root@DietPi-HTPC:~# cd /tmp
root@DietPi-HTPC:/tmp# wget https://mirror.netcologne.de/raspbian/raspbian/pool/main/s/samba/samba-vfs-modules_4.13.13%2Bdfsg-1~deb11u4_armhf.deb
--2022-07-10 13:52:58--  https://mirror.netcologne.de/raspbian/raspbian/pool/main/s/samba/samba-vfs-modules_4.13.13%2Bdfsg-1~deb11u4_armhf.deb
Resolving mirror.netcologne.de (mirror.netcologne.de)... 194.8.197.22, 2001:4dd0:1234:1::deb
Connecting to mirror.netcologne.de (mirror.netcologne.de)|194.8.197.22|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 459684 (449K) [application/octet-stream]
Saving to: ‘samba-vfs-modules_4.13.13+dfsg-1~deb11u4_armhf.deb’

samba-vfs-modules_ 100%[===============>] 448.91K  --.-KB/s    in 0.06s

2022-07-10 13:52:58 (7.88 MB/s) - ‘samba-vfs-modules_4.13.13+dfsg-1~deb11u4_armhf.deb’ saved [459684/459684]

root@DietPi-HTPC:/tmp# apt install ./samba-vfs-modules_4.13.13+dfsg-1~deb11u4_armhf.deb
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Note, selecting 'samba-vfs-modules' instead of './samba-vfs-modules_4.13.13+dfsg-1~deb11u4_armhf.deb'
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 samba-vfs-modules : Depends: samba-libs (= 2:4.13.13+dfsg-1~deb11u4) but 2:4.13.13+dfsg-1~deb11u3 is to be installed
                     Depends: libwbclient0 (= 2:4.13.13+dfsg-1~deb11u4) but 2:4.13.13+dfsg-1~deb11u3 is to be installed
E: Unable to correct problems, you have held broken packages.
root@DietPi-HTPC:/tmp# rm samba-vfs-modules_4.13.13+dfsg-1~deb11u4_armhf.deb
root@DietPi-HTPC:/tmp#
```

MichaIng, 2022-07-10:
Don't install it like that if your APT repo lists are not up-to-date (causing the dependency mismatch), always install via APT:
```sh
apt update
apt install samba-vfs-modules
```

Joulinar, 2022-07-10:
that's what he did @MichaIng https://github.com/MichaIng/DietPi/issues/5605#issuecomment-1179709304

However it seems the mirror has some issues. The packages got updated yesterday at 10pm. Maybe it has not been synced fully. At least all files are available.

ZiggyPi, 2022-07-10:
Ran it again as @MichaIng discribed, and that seems to work. Gonna check if unhashing widelinks will work

MichaIng, 2022-07-10:
Jep, retry with proceeding `apt update` until the mirror is in sync again. Probably do once before:
```sh
/boot/dietpi/func/dietpi-set_software apt clean
```

ZiggyPi, 2022-07-10:
Okay Samba is working as it should again with my original settings.

So only HTPC-Manager, Nextcloud and Organizr issues remain

MichaIng, 2022-07-10:
Can you show again the HTPC Manager logs:
```sh
journalctl -u htpc-manager
```

ZiggyPi, 2022-07-10:
```
root@DietPi-HTPC:~# journalctl -u htpc-manager
-- Journal begins at Sun 2022-07-10 13:17:02 CEST, ends at Sun 2022-07-10 14:12:43 CEST. --
-- No entries --
root@DietPi-HTPC:~#
```

Joulinar, 2022-07-10:
Did you setup your revers proxy already. Because HTPC was listen to localhost only

[81 more replies]

closed 2022-07-22 by MichaIng