MichaIng/DietPi #5610: error updating squeezelite conf  on new install

jsebright, 2022-07-08:
### Creating a bug report/issue

---------------------------------------------------------------------
#### Details:
- Date           | Fri Jul  8 19:23:25 BST 2022
- DietPi version | v8.6.1 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | from scratch
- Hardware       | RPi Zero 2 W (armv7l) (ID=3)
- Kernel version | `Linux BreweryPi 5.15.32-v7+ #1538 SMP Thu Mar 31 19:38:48 BST 2022 armv7l GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=0)
- Command        | `systemctl restart squeezelite`
- Exit code      | 1
- Software title | DietPi-Services
#### Steps to reproduce:

Only just installed DietPi, added PiHole and squeezeplayer.
Wanted to change the name of the player, so went into services, edited the file presented by uncommenting the line with the player name, changed the name and exited that page.
Something went wrong. The next option seemed to be to restart the service, but there was a message that it failed, with the info above.

#### Expected behaviour:

Squeezelite restarts and picks up the new name
.
#### Actual behaviour:

dietpi-launcher exits with error message as below, 

#### Extra details:

- ...
#### Additional logs:
```
Failed to restart squeezelite.service: Unit squeezelite.service has a bad unit file setting.
See system logs and 'systemctl status squeezelite.service' for details.
```

Random troubleshooting:
root@BreweryPi:~# systemctl status squeezelite
Warning: The unit file, source configuration file or drop-ins of squeezelite.service changed on disk. Run 'systemctl daemon-reload' to reload units.
● squeezelite.service - Squeezelite (DietPi)
     Loaded: bad-setting (Reason: Unit squeezelite.service has a bad unit file setting.)
    Drop-In: /etc/systemd/system/squeezelite.service.d
             └─dietpi-services_edit.conf
     Active: active (running) since Fri 2022-07-08 18:56:16 BST; 38min ago
       Docs: man:squeezelite(1)
             https://ralph-irving.github.io/squeezelite.html
   Main PID: 437 (squeezelite)
      Tasks: 4 (limit: 985)
        CPU: 2.994s
     CGroup: /system.slice/squeezelite.service
             └─437 /usr/bin/squeezelite -a 4096 1024 16 0 0 -C 5 -n DietPi-SqueezeLite

Jul 08 18:56:16 BreweryPi systemd[1]: Started Squeezelite (DietPi).
Jul 08 19:23:14 BreweryPi systemd[1]: squeezelite.service: Service has more than one ExecStart= setting, which is only allowed for Type=oneshot services. Refusing.

root@BreweryPi:~# cat /etc/systemd/system/squeezelite.service.d/dietpi-services_edit.conf
[Unit]
#Description=Squeezelite (DietPi)
#Documentation=man:squeezelite(1) https://ralph-irving.github.io/squeezelite.html
#After=sound.target

[Service]
#User=squeezelite
ExecStart=/usr/bin/squeezelite -a 4096:1024:16:0:0 -C 5 -n 'Brewery'

[Install]
#WantedBy=multi-user.target
root@BreweryPi:~#

Joulinar, 2022-07-08:
Can you share following

`systemctl cat squeezelite.service`

I guess you would have need to set an empty `ExecStart` to overwrite the original command before you added the new line `ExecStart=/usr/bin/squeezelite -a 4096:1024:16:0:0 -C 5 -n 'Brewery'`

jsebright, 2022-07-08:
root@BreweryPi:/etc# systemctl cat squeezelite.service
# /lib/systemd/system/squeezelite.service
[Unit]
Description=Squeezelite (DietPi)
Documentation=man:squeezelite(1) https://ralph-irving.github.io/squeezelite.html
After=sound.target

[Service]
User=squeezelite
ExecStart=/usr/bin/squeezelite -a 4096:1024:16:0:0 -C 5 -n 'DietPi-SqueezeLite'

[Install]
WantedBy=multi-user.target

# /etc/systemd/system/squeezelite.service.d/dietpi-services_edit.conf
[Unit]
#Description=Squeezelite (DietPi)
#Documentation=man:squeezelite(1) https://ralph-irving.github.io/squeezelite.html
#After=sound.target

[Service]
#User=squeezelite
ExecStart=/usr/bin/squeezelite -a 4096:1024:16:0:0 -C 5 -n 'Brewery'

[Install]
#WantedBy=multi-user.target

Joulinar, 2022-07-08:
Yes you would need to add an additional `ExecStart=` right before the changed command

jsebright, 2022-07-08:
just made the edit and it restarted without error.
Squeezelite has the correct name now.

MichaIng, 2022-07-08:
Great, I'll mark this as closed then.

I'm thinking of adding an `/etc/default/squeezelite` to allow changing the command-line arguments with. This service editing is not exactly comfortable 😄.

jsebright, 2022-07-08:
If it's possible to make this clearer that would be great -  the extra ExecStart= in the template might make it more obvious.
Even better, if those startup options were in a separate config that could be edited.

Thanks for your help - impressed at the speed of your response!

jsebright, 2022-07-08:
Agree re the   /etc/default/squeezelite   - it would make it easier if standard editing worked.

Regards.

closed 2022-07-08 by jsebright

MichaIng, 2022-07-08:
Well, closely reading the docs should make it clear:
> - Unset and re-set the ExecStart entry:
> ```
> ExecStart=
> ExecStart=/usr/bin/squeezelite [<your custom arguments>]
> ```
> The first `ExecStart=` is required to replace the existing ExecStart entry instead of adding a second one.

But yeah, it's not intuitive and more work than just editing a single config file with clear `key=value` lines.

jsebright, 2022-07-08:
Thought I could make the change just by following the prompts from dietpi-launcher. That normally works OK.
I found the documentation it would have been good to have checked it in advance...

MichaIng, 2022-07-10:
Ah okay, makes sense then. Adding that line to the template is not so easy since with other service types like `oneshot`, multiple `ExecStart` entries are allowed and will run one after another. And of course `ExecStart` may be not what one wants to edit.

Using `/etc/default/<service_name>` for all non-mandatory command-line arguments seems best and matches the way it is commonly handled among Linux software. I wanted to keep it simple without environment file in the past, but that isn't handy.