MichaIng/DietPi #5612: Error when adding pihole via diet-software

smittyj, 2022-07-09:
### Creating a bug report/issue

#### Details:
- Date           | Sat Jul  9 16:39:22 ADT 2022
- DietPi version | v8.6.1 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | from scratch
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | `Linux DietPi-RPi8G 5.15.32-v8+ #1538 SMP PREEMPT Thu Mar 31 19:40:39 BST 2022 aarch64 GNU/Linux`
- Distro         | bookworm (ID=7,RASPBIAN=0)
- Command        | `systemctl enable --now unbound`
- Exit code      | 1
- Software title | DietPi-Software
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Synchronizing state of unbound.service with SysV service script with /lib/systemd/systemd-sysv-install.
Executing: /lib/systemd/systemd-sysv-install enable unbound
Job for unbound.service failed because the control process exited with error code.
See "systemctl status unbound.service" and "journalctl -xeu unbound.service" for details.

Jul 09 16:38:56 DietPi-RPi8G systemd[1]: Starting Unbound DNS server...
░░ Subject: A start job for unit unbound.service has begun execution
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░
░░ A start job for unit unbound.service has begun execution.
░░
░░ The job identifier is 1076.
Jul 09 16:38:56 DietPi-RPi8G unbound[28149]: [28149:0] error: unable to open /var/lib/unbound/root.key for reading: No such file or directory
Jul 09 16:38:56 DietPi-RPi8G unbound[28149]: [28149:0] error: error reading auto-trust-anchor-file: /var/lib/unbound/root.key
Jul 09 16:38:56 DietPi-RPi8G unbound[28149]: [28149:0] error: validator: error in trustanchors config
Jul 09 16:38:56 DietPi-RPi8G unbound[28149]: [28149:0] error: validator: could not apply configuration settings.
Jul 09 16:38:56 DietPi-RPi8G unbound[28149]: [28149:0] error: module init for module validator failed
Jul 09 16:38:56 DietPi-RPi8G unbound[28149]: [28149:0] fatal error: failed to setup modules
Jul 09 16:38:56 DietPi-RPi8G systemd[1]: unbound.service: Main process exited, code=exited, status=1/FAILURE
░░ Subject: Unit process exited
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░
░░ An ExecStart= process belonging to unit unbound.service has exited.
░░
░░ The process' exit code is 'exited' and its exit status is 1.
Jul 09 16:38:56 DietPi-RPi8G systemd[1]: unbound.service: Failed with result 'exit-code'.
░░ Subject: Unit failed
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░
░░ The unit unbound.service has entered the 'failed' state with result 'exit-code'.
Jul 09 16:38:56 DietPi-RPi8G systemd[1]: Failed to start Unbound DNS server.
░░ Subject: A start job for unit unbound.service has failed
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░
░░ A start job for unit unbound.service has finished with a failure.
░░
░░ The job identifier is 1076 and the job result is failed.
Jul 09 16:38:56 DietPi-RPi8G systemd[1]: unbound.service: Scheduled restart job, restart counter is at 1.
[... 1035 lines cut ...]
- SD card used | (EG: SanDisk ultra)

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?
<!-- If you sent a "dietpi-bugreport", please paste the ID here -->
- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce
<!-- Explain how to reproduce the issue -->
1. ...
2. ...

#### Expected behaviour
<!-- What SHOULD be happening? -->
- ...

#### Actual behaviour
<!-- What IS happening? -->
- ...

#### Extra details
<!-- Please post any extra details that might help solve the issue -->
- ...

Joulinar, 2022-07-09:
Could you check why unbound is 

`journalctl -u unbound`

smittyj, 2022-07-09:
***@***.***:/home/js# journalctl -u unbound
Jul 09 16:49:45 DietPi-RPi8G systemd[1]: Starting Unbound DNS server...
Jul 09 16:49:46 DietPi-RPi8G unbound[478]: [478:0] error: unable to open /var/lib/unbound/root.key for reading: No such file or directory
Jul 09 16:49:46 DietPi-RPi8G unbound[478]: [478:0] error: error reading auto-trust-anchor-file: /var/lib/unbound/root.key
Jul 09 16:49:46 DietPi-RPi8G unbound[478]: [478:0] error: validator: error in trustanchors config
Jul 09 16:49:46 DietPi-RPi8G unbound[478]: [478:0] error: validator: could not apply configuration settings.
Jul 09 16:49:46 DietPi-RPi8G unbound[478]: [478:0] error: module init for module validator failed
Jul 09 16:49:46 DietPi-RPi8G unbound[478]: [478:0] fatal error: failed to setup modules
Jul 09 16:49:46 DietPi-RPi8G systemd[1]: unbound.service: Main process exited, code=exited, status=1/FAILURE
Jul 09 16:49:46 DietPi-RPi8G systemd[1]: unbound.service: Failed with result 'exit-code'.
Jul 09 16:49:46 DietPi-RPi8G systemd[1]: Failed to start Unbound DNS server.
Jul 09 16:49:46 DietPi-RPi8G systemd[1]: unbound.service: Scheduled restart job, restart counter is at 1.
Jul 09 16:49:46 DietPi-RPi8G systemd[1]: Stopped Unbound DNS server.
Jul 09 16:49:46 DietPi-RPi8G systemd[1]: Starting Unbound DNS server...
Jul 09 16:49:47 DietPi-RPi8G unbound[526]: [526:0] error: unable to open /var/lib/unbound/root.key for reading: No such file or directory
Jul 09 16:49:47 DietPi-RPi8G unbound[526]: [526:0] error: error reading auto-trust-anchor-file: /var/lib/unbound/root.key
Jul 09 16:49:47 DietPi-RPi8G unbound[526]: [526:0] error: validator: error in trustanchors config
Jul 09 16:49:47 DietPi-RPi8G unbound[526]: [526:0] error: validator: could not apply configuration settings.
Jul 09 16:49:47 DietPi-RPi8G unbound[526]: [526:0] error: module init for module validator failed
Jul 09 16:49:47 DietPi-RPi8G unbound[526]: [526:0] fatal error: failed to setup modules
Jul 09 16:49:47 DietPi-RPi8G systemd[1]: unbound.service: Main process exited, code=exited, status=1/FAILURE
Jul 09 16:49:47 DietPi-RPi8G systemd[1]: unbound.service: Failed with result 'exit-code'.
Jul 09 16:49:47 DietPi-RPi8G systemd[1]: Failed to start Unbound DNS server.
Jul 09 16:49:47 DietPi-RPi8G systemd[1]: unbound.service: Scheduled restart job, restart counter is at 2.
Jul 09 16:49:47 DietPi-RPi8G systemd[1]: Stopped Unbound DNS server.
Jul 09 16:49:47 DietPi-RPi8G systemd[1]: Starting Unbound DNS server...
Jul 09 16:49:47 DietPi-RPi8G unbound[558]: [558:0] error: unable to open /var/lib/unbound/root.key for reading: No such file or directory
Jul 09 16:49:47 DietPi-RPi8G unbound[558]: [558:0] error: error reading auto-trust-anchor-file: /var/lib/unbound/root.key
Jul 09 16:49:47 DietPi-RPi8G unbound[558]: [558:0] error: validator: error in trustanchors config
Jul 09 16:49:47 DietPi-RPi8G unbound[558]: [558:0] error: validator: could not apply configuration settings.
Jul 09 16:49:47 DietPi-RPi8G unbound[558]: [558:0] error: module init for module validator failed
Jul 09 16:49:47 DietPi-RPi8G unbound[558]: [558:0] fatal error: failed to setup modules
Jul 09 16:49:47 DietPi-RPi8G systemd[1]: unbound.service: Main process exited, code=exited, status=1/FAILURE
Jul 09 16:49:47 DietPi-RPi8G systemd[1]: unbound.service: Failed with result 'exit-code'.
Jul 09 16:49:47 DietPi-RPi8G systemd[1]: Failed to start Unbound DNS server.
Jul 09 16:49:47 DietPi-RPi8G systemd[1]: unbound.service: Scheduled restart job, restart counter is at 3.
Jul 09 16:49:47 DietPi-RPi8G systemd[1]: Stopped Unbound DNS server.
Jul 09 16:49:47 DietPi-RPi8G systemd[1]: Starting Unbound DNS server...
Jul 09 16:49:47 DietPi-RPi8G unbound[587]: [587:0] error: unable to open /var/lib/unbound/root.key for reading: No such file or directory
Jul 09 16:49:47 DietPi-RPi8G unbound[587]: [587:0] error: error reading auto-trust-anchor-file: /var/lib/unbound/root.key
Jul 09 16:49:47 DietPi-RPi8G unbound[587]: [587:0] error: validator: error in trustanchors config
Jul 09 16:49:47 DietPi-RPi8G unbound[587]: [587:0] error: validator: could not apply configuration settings.
Jul 09 16:49:47 DietPi-RPi8G unbound[587]: [587:0] error: module init for module validator failed
Jul 09 16:49:47 DietPi-RPi8G unbound[587]: [587:0] fatal error: failed to setup modules
Jul 09 16:49:47 DietPi-RPi8G systemd[1]: unbound.service: Main process exited, code=exited, status=1/FAILURE
Jul 09 16:49:47 DietPi-RPi8G systemd[1]: unbound.service: Failed with result 'exit-code'.
Jul 09 16:49:47 DietPi-RPi8G systemd[1]: Failed to start Unbound DNS server.
Jul 09 16:49:48 DietPi-RPi8G systemd[1]: unbound.service: Scheduled restart job, restart counter is at 4.
Jul 09 16:49:48 DietPi-RPi8G systemd[1]: Stopped Unbound DNS server.
Jul 09 16:49:48 DietPi-RPi8G systemd[1]: Starting Unbound DNS server...
Jul 09 16:49:48 DietPi-RPi8G unbound[601]: [601:0] error: unable to open /var/lib/unbound/root.key for reading: No such file or directory
Jul 09 16:49:48 DietPi-RPi8G unbound[601]: [601:0] error: error reading auto-trust-anchor-file: /var/lib/unbound/root.key
Jul 09 16:49:48 DietPi-RPi8G unbound[601]: [601:0] error: validator: error in trustanchors config
Jul 09 16:49:48 DietPi-RPi8G unbound[601]: [601:0] error: validator: could not apply configuration settings.
Jul 09 16:49:48 DietPi-RPi8G unbound[601]: [601:0] error: module init for module validator failed
Jul 09 16:49:48 DietPi-RPi8G unbound[601]: [601:0] fatal error: failed to setup modules
[... cut ...]
Jul 09 16:50:32 DietPi-RPi8G systemd[1]: Failed to start Unbound DNS server.
Jul 09 16:50:32 DietPi-RPi8G systemd[1]: unbound.service: Scheduled restart job, restart counter is at 5.
Jul 09 16:50:32 DietPi-RPi8G systemd[1]: Stopped Unbound DNS server.
Jul 09 16:50:32 DietPi-RPi8G systemd[1]: unbound.service: Start request repeated too quickly.
Jul 09 16:50:32 DietPi-RPi8G systemd[1]: unbound.service: Failed with result 'exit-code'.
Jul 09 16:50:32 DietPi-RPi8G systemd[1]: Failed to start Unbound DNS server.

Joulinar, 2022-07-09:
Did you selected Unbound to be installed together with PiHole or was this a mistake?

smittyj, 2022-07-09:
I selected to install pihole from diet-software but checking the box from the browse option, I went back a level and selected install.  As part of the install process I was prompted (asked) to install unbound as well.  Both packages appears on the list of apps to install.

Joulinar, 2022-07-09:
It's an optional software title that is not necessarily needed. The question is, do you need the Unbound package?

smittyj, 2022-07-09:
My understanding is that if I want pihole to be a rescursive resolver and DHCP server I would need unbound to be installed.

Joulinar, 2022-07-09:
that's not correct. Purpose of Unbound is to be able to ask DNS queries towards DNS root server and not to use a public upstream DNS. Or to do DoT. But Unbound is not needed for PiHole. As said above, it's just an add-on.

MichaIng, 2022-07-09:
Verified, it's a Bookworm issue, tested on `x86_64`, working well on Bullseye.

smittyj, 2022-07-09:
Ah — thanks for checking.  I just tried a manual install had the same issue.  I tried installing dns-root-data (it was recommend) and had the same issue.  Out of curiosity, I installed the dns-root-date and then installed unbound.  Same results in all scenarios.

Thanks for your efforts — much appreciate and you do a great job.

MichaIng, 2022-07-09:
`/usr/share/dns/root.key` is missing, which is the original file that is copied `/var/lib/unbound/root.key` during service start, if it exists. Strange since without it Unbound is doomed to fail, of course, not sure how such a bug could have slipped through Sid testing and made it into Bookworm repo.

MichaIng, 2022-07-09:
Found it in Debian changelog:
>   * use dns root.key stored in /usr/share/dns/ (as provided by dns-root-data
    package) instead of the unbound-owned /var/lib/unbound/root.key (which is
    managed by an untrusted user). This changes defaults for unbound-host and
    unbound-anchor.  Add Recommends: dns-root-data for unbound-host so it can
    find this root.key in the default install. Closes: #641704

Doesn't make much sense to have package as "recommendation" only which is strictly required for the service start with default config to succeed. However, easy to solve:
```sh
apt install dns-root-data
```

MichaIng, 2022-07-09:
Fixed with: https://github.com/MichaIng/DietPi/commit/42c92c1

MichaIng, 2022-07-23:
Fix can be tested in beta: #5643

closed 2022-07-23 by MichaIng

smittyj, 2022-07-24:
Confirmed to be fixed on my install!  Thanks, much appreciated.