MichaIng/DietPi #5613: Unable to update from 7.2.3 to 7.9.4

betyar, 2022-07-10:
Tried to run dietpi-update but receive an error. Here's the required info:

#### Details:
- Date           | Sun Jul 10 13:01:48 CEST 2022
- DietPi version | v7.2.3 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi Zero W (armv6l) (ID=1)
- Kernel version | Linux DietPi 4.19.66+ #1253 Thu Aug 15 11:37:30 BST 2019 armv6l GNU/Linux
- Distro         | stretch (ID=4,RASPBIAN=1)
- Command        | apt-get -o Acquire::AllowReleaseInfoChange::Suite=true -q update
- Exit code      | 100
- Software title | DietPi-Pre-patches
#### Steps to reproduce:

1. Run dietpi-update
#### Expected behaviour:

- DietPi should be updated to the latest version
#### Actual behaviour:

- Error occurs soon after launching dietpi-update
#### Extra details:

- None
#### Additional logs:
```
Get:1 http://raspbian.raspberrypi.org/raspbian stretch InRelease [15.0 kB]
Ign:2 http://archive.raspbian.org/raspbian jessie InRelease
Err:3 http://archive.raspbian.org/raspbian jessie Release
  404  Not Found
Get:4 https://archive.raspberrypi.org/debian stretch InRelease [25.3 kB]
Ign:5 https://repo.mosquitto.org/debian stretch InRelease
Get:6 http://raspbian.raspberrypi.org/raspbian stretch/main armhf Packages [11.7 MB]
Err:7 https://repo.mosquitto.org/debian stretch Release
  server certificate verification failed. CAfile: /etc/ssl/certs/ca-certificates.crt CRLfile: none
Get:8 https://archive.raspberrypi.org/debian stretch/main armhf Packages [192 kB]
Reading package lists...
E: The repository 'http://archive.raspbian.org/raspbian jessie Release' does no longer have a Release file.
E: The repository 'https://repo.mosquitto.org/debian stretch Release' does no longer have a Release file.
```

Joulinar, 2022-07-10:
uhhh you are running a very old system. There is see you still have Debian `jessie` configured. For mosquitto we might need to set the source to HTTP instead of HTTPS. Can you share following

```
for i in /etc/apt/sources.list{,.d/*.list}; do echo "$i:"; cat "$i"; done
```

MichaIng, 2022-07-10:
Hum, DietPi v7 actually doesn't support Jessie anymore. There was a last v6 branch a while ago but in the meantime where even the next Debian version "Stretch" is not supported, We removed that one + migration code. And as you can see, the APT repository for Jessie isn't online anymore either, as well not for 3rd party repos which is why you already started to mix with Stretch repos. Missing release files/repos + unknown (by Jessie OpenSSL/ca-certificates) certificates make it impossible to update with these.

There are no security updates by the LTS team of Debian anymore, which means security vulnerabilities in kernel and software packages are not patched anymore. This is even true for Debian Stretch since end of June.

I strongly recommend, if you find the time to setup, to flash a fresh DietPi image, which is 3 Debian generations newer.

Joulinar, 2022-07-10:
Jussie is just a left over within the apt source list. I guess it's actually a Stretch system

betyar, 2022-07-10:
Thanks for the help. I'll flash a new image as recommended.

closed 2022-07-10 by betyar

MichaIng, 2022-07-10:
`archive.raspbian.org` is added with Jessie suite only, so the core is Jessie, I suppose. But indeed that the banner/bug report template says "stretch" doesn't fit, it should show "Unsupported" then, although not sure how it was with v7.2.3 code.

betyar, 2022-10-11:
Strange, I was able to update to 7.2.3, thus I thought an update to a later
version of v7 would be no problem. I guess it is.