MichaIng/DietPi #5617: Invoice Ninja

blackscooby, 2022-07-11:
### Creating a software request

#### Formal software information
- Software title | Invoice Ninja
- Short description | Invoicing solution 
- Official URL (if available) |[https://www.turnkeylinux.org/invoice-ninja}
- GitHub URL (if available) | 
- (Official) install documentation | 

#### Are there similar/alternative software titles available with DietPi-Software?
Nothing that I've found.
Very good and configurable turnkey solution, lightweight and suitable for DietPi deployment.
- ...

#### What makes your requested software better than the above solutions, if available?
None avail.
- ...

#### How can DietPi make the installation easier or compatible, than following the install instructions or do APT installation, if available?

- ...

#### Can you provide the installation steps that you would suggest DietPi-Software to do?

1. ...

#### Are you willing to help maintaining the software installation, e.g. in case of needed setup changes due to updates etc.? This is not needed, but could speed up our decision to implement it, as man power is always a topic :wink:.

- ...

Joulinar, 2022-07-11:
is h´that really a software that can be installed on an operating system? It seems more some kind of appliance running as own VM image or Live CD https://github.com/turnkeylinux-apps/core

blackscooby, 2022-07-15:
> is that really a software that can be installed on an operating system? It seems more some kind of appliance running as own VM image or Live CD https://github.com/turnkeylinux-apps/core

Yes it can be self hosted in Linux with a LAMP server

MichaIng, 2022-07-16:
I guess these are better resources:
- https://www.invoiceninja.org/
- https://github.com/invoiceninja/invoiceninja
- https://github.com/invoiceninja/invoiceninja/wiki/v5-Installation
- https://forum.invoiceninja.com/t/install-invoice-ninja-v5-on-ubuntu-20-04/4588

Install instructions from GitHub are way too short, things like [required PHP modules](https://github.com/invoiceninja/invoiceninja/blob/master/composer.json) and `.env` setup are missing.
Install instructions on their forum are very messy and would lead to failures, but the required steps can be extracted.