MichaIng/DietPi #5621: dietpi 8.6 not booting on quartz64 model A : Failed to get kernel dtb, ret=-19

rmadrona, 2022-07-13:
### Creating a bug report/issue

#### Required Information
- DietPi version | 8.6
- Distro version | not relevant
- Kernel version | not relevant
- SBC model | quartz64 model A 8GB
- Power supply used | 12V 3A JX microtech
- SD card used | SanDisk 8GB

#### Additional Information (if applicable)
BOOT LOG
DDR Version V1.13 20220218  
ln                                                                                                               
ddrconfig:7                                                                                                      
LPDDR4X, 324MHz                                                                                                  
BW=32 Col=10 Bk=8 CS0 Row=17 CS1 Row=17 CS=2 Die BW=8 Size=8192MB                                                
tdqss: cs0 dqs0: 72ps, dqs1: -48ps, dqs2: -24ps, dqs3: -96ps,                                                    
tdqss: cs1 dqs0: 72ps, dqs1: -72ps, dqs2: -24ps, dqs3: -96ps,                                                    
                                                                                                                 
change to: 324MHz                                                                                                
PHY drv:clk:36,ca:36,DQ:29,odt:0                                                                                 
vrefinner:24%, vrefout:41%                                                                                       
dram drv:40,odt:0                                                                                                
clk skew:0x60                                                                                                    
                                                                                                                 
change to: 528MHz                                                                                                
PHY drv:clk:36,ca:36,DQ:29,odt:0                                                                                 
vrefinner:24%, vrefout:41%                                                                                       
dram drv:40,odt:0                                                                                                
clk skew:0x58                                                                                                    
                                                                                                                 
change to: 780MHz                                                                                                
PHY drv:clk:36,ca:36,DQ:29,odt:0                                                                                 
vrefinner:24%, vrefout:41%                                                                                       
dram drv:40,odt:0                                                                                                
clk skew:0x58                                                                                                    
                                                                                                                 
change to: 1056MHz(final freq)                                                                                   
PHY drv:clk:36,ca:36,DQ:29,odt:60                                                                                
vrefinner:16%, vrefout:22%                                                                                       
dram drv:40,odt:80                                                                                               
vref_ca:00000071                                                                                                 
clk skew:0x2f                                                                                                    
cs 0:                                                                                                            
the read training result:                                                                                        
DQS0:0x3b, DQS1:0x39, DQS2:0x41, DQS3:0x3b,                                                                      
min  : 0xc  0xb  0xd  0xb  0x1  0x5  0x7  0x3 , 0x8  0x8  0x3  0x1  0xb  0x9  0xc  0x8 ,                         
       0xf 0x10  0xe  0xc  0x5  0x2  0x2  0x5 , 0xd  0x9  0x9  0x2  0xd  0xd  0xa  0xe ,                         
mid  :0x29 0x2a 0x2b 0x29 0x1f 0x22 0x23 0x20 ,0x23 0x24 0x1f 0x1e 0x27 0x26 0x29 0x25 ,                         
      0x2e 0x2e 0x2b 0x29 0x24 0x20 0x21 0x22 ,0x2b 0x27 0x25 0x20 0x2b 0x2a 0x29 0x2c ,                         
max  :0x46 0x49 0x4a 0x48 0x3d 0x40 0x40 0x3e ,0x3f 0x40 0x3c 0x3b 0x44 0x44 0x46 0x43 ,                         
      0x4e 0x4c 0x49 0x47 0x43 0x3e 0x40 0x40 ,0x49 0x45 0x42 0x3f 0x4a 0x48 0x49 0x4b ,                         
range:0x3a 0x3e 0x3d 0x3d 0x3c 0x3b 0x39 0x3b ,0x37 0x38 0x39 0x3a 0x39 0x3b 0x3a 0x3b ,                         
      0x3f 0x3c 0x3b 0x3b 0x3e 0x3c 0x3e 0x3b ,0x3c 0x3c 0x39 0x3d 0x3d 0x3b 0x3f 0x3d ,                         
the write training result:                                                                                       
DQS0:0x38, DQS1:0x29, DQS2:0x2c, DQS3:0x23,                                                                      
min  :0x5e 0x5f 0x62 0x5f 0x54 0x59 0x5a 0x5b 0x5a ,0x4d 0x4f 0x49 0x49 0x53 0x52 0x54 0x52 0x4e ,               
      0x52 0x51 0x4d 0x4d 0x49 0x45 0x47 0x4c 0x4d ,0x4c 0x4b 0x49 0x45 0x4f 0x4e 0x4c 0x50 0x4a ,               
mid  :0x79 0x7b 0x7d 0x7a 0x6e 0x71 0x75 0x75 0x73 ,0x69 0x69 0x63 0x63 0x6c 0x6b 0x6d 0x6c 0x67 ,               
      0x6e 0x6d 0x68 0x69 0x61 0x5c 0x5e 0x64 0x67 ,0x68 0x63 0x61 0x5e 0x69 0x6a 0x64 0x6c 0x62 ,               
max  :0x95 0x98 0x99 0x95 0x89 0x8a 0x91 0x8f 0x8c ,0x86 0x83 0x7d 0x7d 0x86 0x84 0x86 0x86 0x80 ,               
      0x8a 0x89 0x83 0x85 0x7a 0x74 0x75 0x7c 0x81 ,0x85 0x7b 0x7a 0x78 0x83 0x86 0x7d 0x89 0x7a ,               
range:0x37 0x39 0x37 0x36 0x35 0x31 0x37 0x34 0x32 ,0x39 0x34 0x34 0x34 0x33 0x32 0x32 0x34 0x32 ,               
      0x38 0x38 0x36 0x38 0x31 0x2f 0x2e 0x30 0x34 ,0x39 0x30 0x31 0x33 0x34 0x38 0x31 0x39 0x30 ,               
cs 1:                                                                                                            
the read training result:                                                                                        
DQS0:0x3b, DQS1:0x3b, DQS2:0x40, DQS3:0x3b,                                                                      
min  : 0xc  0xb  0xc  0xa  0x2  0x4  0x7  0x2 , 0x8  0x9  0x4  0x1  0xc  0xb  0xe  0x9 ,                         
       0xe  0xe  0xd  0xb  0x5  0x2  0x2  0x6 , 0xb  0x8  0x8  0x2  0xd  0xd  0xa  0xd ,                         
[... 155 lines cut ...]
Retrieving file: pxelinux.cfg/defaee(master_bus) failed: -22eqos_prfailed: -22eqos_probe_resources(failed: -22eqe
Retrieving file: pxelinux.cfg/default-arm                                                                        
clk_get_by_name(master_bus) failed: -22eqos_probe_resourceqos_probe_resources() failed: -2qos_probe_resources() e
Retrieving file: blelinux.cfg/default                                                                            
clk_get_by_name(master_bus) failed: -22eqos_probe_resources() failed: -22cces() failed: -22clk_get_by_nameces() .
clk_get_by_name(master_busbs() failed: -22clk_get_by_name(m2clk_get_by_name(master_bus) failed: -22eqos_probe_re.
clk_get_by_name(master_bus) failed: -22eqos_probe_resources() failed: -22=>

#### Steps to reproduce

1. flash DietPi_Quartz64A-ARMv8-Bullseye.img
2. Use pine64 recommanded serial adapter as described https://forum.pine64.org/showthread.php?tid=5029 

#### Expected behaviour

- boot

#### Actual behaviour

- the board does not boot

#### Extra details

- fresh flash (no boot attempt before capturing these logs)
- not using any other interface :  network,  USB,  video, PCIe, ...

MichaIng, 2022-07-13:
Many thanks for your report.

I'll try to replicate. Last successfully tested on eMMC.

rmadrona, 2022-07-13:
I just tried with an eMMC 64GB, and I got the same KO result.

MichaIng, 2022-07-31:
A new image is up with Linux v5.19-rc1: https://dietpi.com/downloads/images/DietPi_Quartz64A-ARMv8-Bullseye.7z

Probably it works better. I remember when I tested SD cards, it failed with a Samsung EVO 128 MB while it worked (and does work now) with a Samsung PRO Endurance 128 MB, as if it was a bit picky. I'll test again with the EVO when I find time which currently boots a NanoPi R5S very well.

rmadrona, 2022-07-31:
boot ok for quertz64a on emmc with dietpi version 8.7.1 and Linux v5.19-rc1
https://dietpi.com/downloads/images/DietPi_Quartz64A-ARMv8-Bullseye.7z
I will continue validation with various devices.
My final goal is to run pimox on this SBC.

root@DietPi:~# uname -ar
Linux DietPi 5.19.0-rc1 #1 SMP PREEMPT Fri Jul 15 09:39:18 UTC 2022 aarch64 GNU/Linux
thx @MichaIng  for your support

MichaIng, 2022-08-02:
That is great. I'll test again on the Samsung EVO SD card.

MichaIng, 2022-09-17:
Works fine here, I'll merk this as closed now.

closed 2022-09-17 by MichaIng